espressif/arduino-esp32 #5088: WiFiScanClass::BSSIDstr() doesn't reserve enough space for the null termination of the MAC char array

mbratch, 2021-04-19:
### Hardware:
ESP32-WROOM-32D Dev Board (ESP-32S)

### Description:
Using the f`ramework-arduinoespressif` library package for WiFi.
I am using a ESP32-WROOM-32D dev board running a simple WiFi test program.

I noticed that, frequently, when I call `WiFi.BSSIDstr(i)` I will ultimately get some spurious results (strings printing with unknown characters or board reset). I'm also getting occasional WDT resets on some runs of the code.

The subject of this issue is one specific observation: I looked at `WiFiScanClass::BSSIDstr()` and noticed it doesn't allocate an extra byte for the null character written by `sprintf`:
```cpp
String WiFiScanClass::BSSIDstr(uint8_t i)
{
    char mac[18] = { 0 };
    wifi_ap_record_t* it = reinterpret_cast<wifi_ap_record_t*>(_getScanInfoByIndex(i));
    if(!it) {
        return String();
    }
    sprintf(mac, "%02X:%02X:%02X:%02X:%02X:%02X", it->bssid[0], it->bssid[1], it->bssid[2], it->bssid[3], it->bssid[4], it->bssid[5]);
    return String(mac);
}
```
The declaration of `mac` should be `char mac[19] = { 0 };` because the `sprintf` format string, `"%02X:%02X:%02X:%02X:%02X:%02X"` requires 19 characters including the null terminator.

I don't have useful data captured yet for the WDT resets.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

String translateEncryptionType(wifi_auth_mode_t encryptionType) {
  switch (encryptionType) {
    case (WIFI_AUTH_OPEN):
      return "Open";
    case (WIFI_AUTH_WEP):
      return "WEP";
    case (WIFI_AUTH_WPA_PSK):
      return "WPA_PSK";
    case (WIFI_AUTH_WPA2_PSK):
      return "WPA2_PSK";
    case (WIFI_AUTH_WPA_WPA2_PSK):
      return "WPA_WPA2_PSK";
    case (WIFI_AUTH_WPA2_ENTERPRISE):
      return "WPA2_ENTERPRISE";
    default:
      return "Unknown";
  }
}

void scanNetworks() {
  int numberOfNetworks = WiFi.scanNetworks();
 
  Serial.print("Number of networks found: ");
  Serial.println(numberOfNetworks);
 
  for (int i = 0; i < numberOfNetworks; i++) {
 
    Serial.print("Network name: ");
    Serial.println(WiFi.SSID(i));
 
    Serial.print("Signal strength: ");
    Serial.print(WiFi.RSSI(i));
    Serial.println(" dBm");
 
    Serial.print("MAC address: ");
    Serial.println(WiFi.BSSIDstr(i));
 
    Serial.print("Encryption type: ");
    String encryptionTypeDescription = translateEncryptionType(WiFi.encryptionType(i));
    Serial.println(encryptionTypeDescription);
    Serial.println("-----------------------");
  }
}

void connectToNetwork() {
  static int status = WL_IDLE_STATUS;     // the Wifi radio's status
  const char* ssid = "MySSID";
  const char* password = "xxxxxxx";

  while (status != WL_CONNECTED) {
    Serial.printf("Trying to connect to %s\n", ssid);
    status = WiFi.begin(ssid, password);
    delay(10000);
  }
 
  Serial.println("Connected to network");
}

void setup() {
  scanNetworks();
  connectToNetwork();
}

void loop() {
}
```

dwzg, 2021-04-21:
I think you might have counted the chars wrong.

`"%02X"` prints a hex number that is at least 2 characters wide. If it has only one digit, it is padded with zeros on the left. If the number has more than 2 digits, it will be wider than 2 characters, because `sprintf()` does not truncate the number. However, as the `bssid` member of the `it` struct is an `uint8_t` array, this is not possible.

We have six of those 2 digit hex numbers and additionally 5 colons in between those.
This adds up to 17 characters. Adding the zero-terminator gives us a length of 18 for the `mac` array.

mbratch, 2021-04-21:
Hi

Thanks and my apologies. You're absolutely right. For some reason in my mind I included a redundant colon which brought me to 18 instead of 17. Silly error on my part. I must have been tired. :-/

Again, my apologies for the error.

Mark

⁣Sent from TypeApp ​

closed 2021-04-21 by VojtechBartoska