espressif/arduino-esp32 #5089: PWM ledCWrite(255) or ledcWrite(1023) is not 100% 1

hallard, 2021-04-20:
### Hardware:
Board: ESP32

### Description:

Unable to light off PWM common anode RGB Led when setting value to latest value of range.

Old known issue of ESP8266 [#5957](https://github.com/esp8266/Arduino/issues/5957) and now seen is [Tasmota](https://github.com/arendst/Tasmota/pull/11786) and my sample test program

The fix is described in comments of official sample [code](https://github.com/espressif/arduino-esp32/blob/404a31f4455d2459dad0b6533b7bfa70347e0044/libraries/ESP32/examples/AnalogOut/ledcWrite_RGB/ledcWrite_RGB.ino#L22)

But from logical working, having value going from 0 to 255 should means that 255 is full scale so always 1 and not having a little pulse. This does now works on ESP8266 with analogWrite(255) and comments have been updated to code according in this [PR](https://github.com/esp8266/Arduino/pull/7456/files)
```
// Per the Arduino docs at https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/
// val: the duty cycle: between 0 (always off) and 255 (always on).
// So if val = 0 we have digitalWrite(LOW), if we have val==range we have digitalWrite(HIGH)
```
I know it's not arduino but I think if PWM could work as is `0 (always off) and max range value (always on)` would be less confusing

my `ledOff()`sample function with 256 does works, but should works with 255, much easier if we use an `uint8_t` as scale range.

```
void ledOff() {
  ledcWrite(LED_RED_CHAN, 256);
  ledcWrite(LED_GRN_CHAN, 256);
  ledcWrite(LED_BLU_CHAN, 256);
}
```

lbernstone, 2021-04-20:
esp32 does not have a fixed bit width like atmel and esp8266 boards, so max is pow(2,bit_res)-1.  You have not included your ledcSetup call, so your issue is not clear.

hallard, 2021-04-20:
Yeah sorry about that, what I mean is that the bug is there if I call `ledcWrite()` with the max value of resolution

```
  ledcAttachPin(LED_RED_PIN, LED_RED_CHAN); // assign RGB led pins to channels
  ledcAttachPin(LED_GRN_PIN, LED_GRN_CHAN);
  ledcAttachPin(LED_BLU_PIN, LED_BLU_CHAN);
  
  // Initialize channels 
  // channels 0-15, resolution 1-16 bits, freq limits depend on resolution
  // ledcSetup(uint8_t channel, uint32_t freq, uint8_t resolution_bits);
  ledcSetup(LED_RED_CHAN, 12000, 8); // 12 kHz PWM, 8-bit resolution
  ledcSetup(LED_GRN_CHAN, 12000, 8);
  ledcSetup(LED_BLU_CHAN, 12000, 8);
```

So 8 bits max value 255 does not light off the Led, 10bits max value 1023 does not light off led, I need to set to max value + 1 to get it working

me-no-dev, 2021-04-22:
255 is max light and 0 is off

hallard, 2021-04-22:
yeah that is the issue (255) is not full 100% HIGH level, if you connect the LED as follow (3V3 not 5V of course)

[image]

You won't be able to light off the LED because 255 is not fully 100% 

Detailled explanation, same issue than this one
https://github.com/esp8266/Arduino/issues/5957

Of course if you connect the LED other side 

[image]

your eye is unable to see light difference to full light and missing 1/255 pulse.

PR #6045 (2021-12-20): Implement LEDC based on ESP-IDF API
## Summary
This PR is refactoring of LEDC HAL in order to use IDF instead of current Register manipulation approach.
Fixing duty -> if all bits in resolution are set -> FULL ON

## Impact
None

## Related links
Fixed #5089
closed 2021-07-08 by stale[bot]