espressif/arduino-esp32 #5090: Using Secure and unsecure mqtt in single code for esp32

sanketa-54, 2021-04-20:
How to use Secure and unsecure mqtt in single code for esp32

### Hardware:

Board: ESP32 D2WD +STM
Core Installation version: 1.0.5 (and later)
IDE name: Arduino IDE and VSCode
Flash Frequency: 40Mhz
PSRAM enabled: No
Upload Speed: 115200
Computer OS: Windows 10

### Description:
I wanted to use both secure and non-secure mqtt clients in single code , At a time only one will be used .
What is the best approach for this ? If anyone I tried this Please post the solution.

TheRealDJ, 2021-04-30:
You will need two client instances
```
// unsecure mqtt
WiFiClient net;
MQTTClient mqtt(1024);

// secure mqtt
WiFiClientSecure nets;
MQTTClient mqtts(1024);

const char* cacert = "...";
const char* ssl_key = "...";
const char* ssl_cert = "...";

nets.setCACert(cacert);
nets.setPrivateKey(ssl_key);
nets.setCertificate(ssl_cert);

mqtt.begin("host_name_or_ip", 1883, net);
mqtts.begin("host_name_or_ip", 8883, nets);
```

sanketa-54, 2021-05-03:
@TheRealDJ 
Thanks for the solution.
I'll try and update.
Also I tried the same logic with only one client instance ...and both secured and non-secured MQTT are connected at same time.

closed 2021-07-20 by stale[bot]