espressif/arduino-esp32 #5096: 'make menuconfig' not working when using esp32-arduino-lib-builder to compile custom libraries

kristofmulier, 2021-04-21:
<h3>1. Hardware</h3>

Board:							ESP32 PICO KIT
ESP-IDF Installation version:			ESP-IDF v4.4-dev-744-g1cb31e509
IDE name:						IDF component
Computer OS:                                         Ubuntu 20.04.2 LTS

<h3>2. Problem description</h3>

I'm following the guide at https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md to build the static Arduino libraries. I'm referring to the precompiled archive files that Arduino stores at `~/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/`:

[image]

Instead of simply using these precompiled archive files, I want to know how they are built and be able to do it myself.

I follow the steps meticulously from the guide at https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md , but I get stuck at nr. 5:

```sh
$ make menuconfig

make: *** No rule to make target 'menuconfig'.  Stop.
```

I'm not sure if this is relevant, but in step nr. 4 I get the following output:

```sh
$ ./tools/install-esp-idf.sh
ESP-IDF is already installed at: /home/kristof/esp/esp-idf
Detecting the Python interpreter
Checking "python" ...
Python 3.8.5
"python" has been detected
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2020r3-8.4.0 (already installed)
Skipping riscv32-esp-elf@1.24.0.123_64eb9ff-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Python 3.8.5
pip 21.0.1 from /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/pip (python 3.8)
Installing Python packages from /home/kristof/esp/esp-idf/requirements.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring python-socketio: markers 'python_version <= "2.7"' don't match your environment
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 4)) (54.1.2)
Requirement already satisfied: click>=7.0 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 11)) (3.4.7)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 13)) (0.27)
Requirement already satisfied: gdbgui==0.13.2.0 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 15)) (0.13.2.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 18)) (0.9.0.2)
Requirement already satisfied: kconfiglib==13.7.1 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 22)) (13.7.1)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 25)) (1.5.4)
Requirement already satisfied: bitstring>=3.1.6 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 26)) (3.1.7)
Requirement already satisfied: ecdsa>=0.16.0 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 27)) (0.16.1)
Requirement already satisfied: construct==2.10.54 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/kristof/esp/esp-idf/requirements.txt (line 31)) (2.10.54)
Requirement already satisfied: Flask<1.0,>=0.12.2 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/kristof/esp/esp-idf/requirements.txt (line 15)) (0.12.5)
Requirement already satisfied: Pygments<3.0,>=2.2.0 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/kristof/esp/esp-idf/requirements.txt (line 15)) (2.8.1)
Requirement already satisfied: Flask-Compress<2.0,>=1.4.0 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/kristof/esp/esp-idf/requirements.txt (line 15)) (1.9.0)
Requirement already satisfied: gevent<2.0,>=1.2.2 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/kristof/esp/esp-idf/requirements.txt (line 15)) (1.5.0)
Requirement already satisfied: Flask-SocketIO<3.0,>=2.9 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/kristof/esp/esp-idf/requirements.txt (line 15)) (2.9.6)
Requirement already satisfied: cffi>=1.12 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from cryptography>=2.1.4->-r /home/kristof/esp/esp-idf/requirements.txt (line 11)) (1.14.5)
Requirement already satisfied: six>=1.9.0 in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from ecdsa>=0.16.0->-r /home/kristof/esp/esp-idf/requirements.txt (line 27)) (1.15.0)
Requirement already satisfied: pycparser in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from cffi>=1.12->cryptography>=2.1.4->-r /home/kristof/esp/esp-idf/requirements.txt (line 11)) (2.20)
[... 12 lines cut ...]

Detecting the Python interpreter
Checking "python" ...
Python 3.8.5
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /home/kristof/.espressif/python_env/idf4.4_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /home/kristof/esp/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/kristof/esp/esp-idf/components/esptool_py/esptool
  /home/kristof/esp/esp-idf/components/espcoredump
  /home/kristof/esp/esp-idf/components/partition_table
  /home/kristof/esp/esp-idf/components/app_update
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```

<h3>3. Context</h3>

I'm part of a team developing a new IDE for microcontrollers: Embeetle IDE. We're currently trying to support the ESP32 microcontroller family in our IDE. For convenience, we're using the Arduino precompiled archive files in the sample projects. However, we don't want to just *use* them without having a thorough understanding on how they were built in the first place. You can find our project here:

https://embeetle.com

Jason2866, 2021-04-21:
I do find building with https://github.com/espressif/esp32-arduino-lib-builder easier and you see in detail how it is done.

lbernstone, 2021-04-21:
The documentation for arduino as a component is based on the v1 code, which is the current stable release.  You are trying to build development code.

I have docker containers available for release versions, and some developmental versions.  The containers can be pulled from https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder.  The scripts to build them are at https://github.com/lbernstone/arduino-esp32-dockerfiles. I take over all the downloading in my scripts in order to keep the container images as small as possible, but otherwise use lib-builder.

lib-builder is a series of scripts (the stuff in [tools](https://github.com/espressif/esp32-arduino-lib-builder/tree/master/tools) along with build.sh), sdkconfig files and a simple project, in order to build all the artifacts provided in the releases.

closed 2021-07-08 by stale[bot]