espressif/arduino-esp32 #5100: not possible to comipile mcpwm_isr_register

Pepito-design, 2021-04-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.06
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Tumbleweed / Suse

### Description:
trying to use the capture unit out of the MCPWM unit.
Everything works, up to the moment I want to register my interrupt service routine.
My code ist based on the example from espressiv [https://github.com/espressif/esp-idf/blob/cf457d412a7748139b77c94e7debe72ee86199af/examples/peripherals/mcpwm/mcpwm_basic_config/main/mcpwm_basic_config_example.c](url)
Line 280 mcpwm_isr_register(MCPWM_UNIT_0, isr_handler, NULL, ESP_INTR_FLAG_IRAM, NULL);  //Set ISR Handler

My code is not finished, still under development, but it  compiles without errors as long mcpwm_isr_register is not used.
Remove comment in line 105, then is no longer possible to compile 
### Sketch:  
#include <Arduino.h>
#include "esp_attr.h"
#include "soc/mcpwm_struct.h"
#include "driver/mcpwm.h"

#define ALPHA 45.0           // Winkel Pickup <-- OT  anpassen!

#define MCPWM_EN_CAPTURE 1
#define CAP0_INT_EN BIT(27)  //Capture 0 interrupt bit
#define CAP1_INT_EN BIT(28)  //Capture 1 interrupt bit
#define GPIO_CAP0_IN   23    //Set GPIO 23 as  CAP0  Pick-UP
#define GPIO_CAP1_IN   25    //Set GPIO 25 as  CAP1  Ignition

typedef struct {
    uint32_t capture_signal_P;
    uint32_t capture_signal_Z;
    mcpwm_capture_signal_t sel_cap_signal;
} capture;

uint32_t *current_cap_value  = NULL;
uint32_t *previous_cap_value = NULL;
static mcpwm_dev_t *MCPWM[1] = {&MCPWM0};

xQueueHandle cap_queue;
float Zuendwinkel ;

static void disp_captured_signal(void *arg)
{
 capture evt;
 float   Drehzahl;
 int32_t OT;
   xQueueReceive(cap_queue, &evt, portMAX_DELAY);
   if (evt.sel_cap_signal == MCPWM_SELECT_CAP0)
    {
     // OT = P + P x (ALPHA/360°)   ALPHA = Offset Pick-UP   redundant, wird nicht benoetigt
     // Drehzahl = 60sec / ( P x 25nsec )
     // OT = evt.capture_signal_P + (( ALPHA / 360.0 ) * evt.capture_signal_P));
     Drehzahl = 60.0 / ((float)evt.capture_signal_P * 0.000000025);
     Serial.print("       P = ");
     Serial.print(evt.capture_signal_P, DEC);  // eingefangener Wert vom Timer
     Serial.print("Drehzahl = ");
     Serial.println(Drehzahl,3);                 // 3 Stellen hinterm Komma
     // Serial.print("      OT = ");
     // Serial.println(OT,DEC);
    }
   if ((evt.sel_cap_signal == MCPWM_SELECT_CAP1))
    { 
     // gamma = ( ALPHA - ( Z * 360 / P )
     Zuendwinkel = ALPHA - ( (float) evt.capture_signal_Z * 360.0 / (float) previous_cap_value[0] );
     Serial.print("    Z = ");
     Serial.print(evt.capture_signal_Z);
     Serial.print("    Zuendwinkel = ");
     Serial.println(Zuendwinkel,3);
    }
}

static void IRAM_ATTR isr_handler(void)
// static void isr_handler(void)
{
    uint32_t mcpwm_intr_status;
    capture evt;
    mcpwm_intr_status = MCPWM[MCPWM_UNIT_0]->int_st.val; //Read interrupt status
    //calculate the interval in the ISR,
    //so that the interval will be always correct even when cap_queue is not handled in time and overflow.
    if (mcpwm_intr_status & CAP0_INT_EN) { //Check for interrupt on rising edge on CAP0 signal
        // mcpwm_start(MCPWM_UNIT_0, MCPWM_SELECT_CAP1);  // Auf Zuendimpuls warten
        current_cap_value[0]  = mcpwm_capture_signal_get_value(MCPWM_UNIT_0, MCPWM_SELECT_CAP0); //get capture signal counter value
        evt.capture_signal_P  = (current_cap_value[0] - previous_cap_value[0]) ;
        previous_cap_value[0] = current_cap_value[0];
        evt.sel_cap_signal    = MCPWM_SELECT_CAP0;
        xQueueSendFromISR(cap_queue, &evt, NULL);
    }
    if (mcpwm_intr_status & CAP1_INT_EN) { //Check for interrupt on rising edge on CAP1 signal   Zuendimpuls gekommen
        // mcpwm_stop(MCPWM_UNIT_0, MCPWM_SELECT_CAP1);
        current_cap_value[1]  = mcpwm_capture_signal_get_value(MCPWM_UNIT_0, MCPWM_SELECT_CAP1); //get capture signal counter value
        evt.capture_signal_Z  = abs((current_cap_value[1] - previous_cap_value[0])) ;  // Differenz Z - P
        evt.sel_cap_signal    = MCPWM_SELECT_CAP1;
        xQueueSendFromISR(cap_queue, &evt, NULL);
    }
    MCPWM[MCPWM_UNIT_0]->int_clr.val = mcpwm_intr_status;
}

static void mcpwm_example_gpio_initialize(void)
   {
    printf("initializing mcpwm gpio...\n");
    mcpwm_pin_config_t pin_config;
          pin_config.mcpwm_cap0_in_num   = GPIO_CAP0_IN;
          pin_config.mcpwm_cap1_in_num   = GPIO_CAP1_IN;

    mcpwm_set_pin(MCPWM_UNIT_0, &pin_config);

    gpio_pulldown_en((gpio_num_t) GPIO_CAP0_IN);    //Enable pull down on CAP0   signal
    gpio_pulldown_en((gpio_num_t) GPIO_CAP1_IN);    //Enable pull down on CAP1   signal
    
    if ( ESP_OK != mcpwm_capture_enable(MCPWM_UNIT_0, MCPWM_SELECT_CAP0, MCPWM_POS_EDGE, 0))  //capture signal on rising edge, prescale = 0 
     {
      Serial.println(" ERROR: MCPWM_UNIT_0, MCPWM_SELECT_CAP0 enable geht nicht");
     }
    if ( ESP_OK != mcpwm_capture_enable(MCPWM_UNIT_0, MCPWM_SELECT_CAP1, MCPWM_POS_EDGE, 0))  //capture signal on rising edge, prescale = 0 
     {
      Serial.println(" ERROR: MCPWM_UNIT_0, MCPWM_SELECT_CAP1 enable geht nicht");
     }
     //enable interrupt, so each this a rising edge occurs interrupt is triggered
    MCPWM[MCPWM_UNIT_0]->int_ena.val = CAP0_INT_EN | CAP1_INT_EN;  //Enable interrupt on  CAP0, CAP1 signal
//    mcpwm_isr_register(MCPWM_UNIT_0, isr_handler, NULL, ESP_INTR_FLAG_IRAM, NULL);  //Set ISR Handler
   }

void setup() 
{
 Serial.begin(115200);
  mcpwm_example_gpio_initialize();
  Serial.println("Configuring Initial Parameters of mcpwm...");
  mcpwm_config_t pwm_config;
  pwm_config.frequency    = 40000000;       //frequenz 40MHz
  pwm_config.cmpr_a       = 50.0;           //duty cycle of PWMxA = 50.0%
  pwm_config.counter_mode = MCPWM_UP_COUNTER;
  pwm_config.duty_mode    = MCPWM_DUTY_MODE_0;
  if ( ESP_OK != mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config))   //Configure PWM0A with above settings
   {
    Serial.println(" ERROR: MCPWM_INIT Timer 0 geht nicht");
   }
 // attachInterrupt(MCPWM_UNIT_0, isr_handler, RISING);
//  mcpwm_start(bla bla);
}

void loop() {
 //unsigned int number = mcpwm_capture_signal_get_value(MCPWM_UNIT_0,MCPWM_SELECT_CAP0);  // get value 
}
```

### Debug Messages:
/home/peter/Arduino/ESP32/capture_test1/capture_test1.ino: In function 'void mcpwm_example_gpio_initialize()':
capture_test1:105:81: error: invalid conversion from 'void (*)()' to 'void (*)(void*)' [-fpermissive]
     mcpwm_isr_register(MCPWM_UNIT_0, isr_handler, NULL, ESP_INTR_FLAG_IRAM, NULL);  //Set ISR Handler
                                                                                 ^
In file included from /home/peter/Arduino/ESP32/capture_test1/capture_test1.ino:4:0:
/home/peter/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/driver/driver/mcpwm.h:703:11: note:   initializing argument 2 of 'esp_err_t mcpwm_isr_register(mcpwm_unit_t, void (*)(void*), void*, int, intr_handle_data_t**)'
 esp_err_t mcpwm_isr_register(mcpwm_unit_t mcpwm_num, void (*fn)(void *), void *arg, int intr_alloc_flags, intr_handle_t *handle);
           ^
exit status 1
invalid conversion from 'void (*)()' to 'void (*)(void*)' [-fpermissive]

lbernstone, 2021-04-21:
isr_handler should be (void *arg) even if you do nothing with arg.
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/mcpwm/mcpwm_bldc_control/main/mcpwm_bldc_control_hall_sensor_example.c#L165

Pepito-design, 2021-04-22:
thank you very much, it compiles fine now

Pepito-design, 2021-04-22:
I'm very astonished and happy about the very quick and useful response
Pepito

closed 2021-04-22 by Pepito-design