espressif/arduino-esp32 #5102: how to run timer / capture with max possibe speed ( 40MHz or 20MHz )?

Pepito-design, 2021-04-22:
I want to measure the distance between 2 signals with very high precision. Idea is to use the capture functionality of MCPWM.
First I'm trying to set the clock prescaler to 2. Expected is half of the crystal frequency. 
I add debug info ( Serial output ) .... it seems to be correct .... better as expected.
Then I start the pwm config stuff.
And here it sucks in my program.
`    if ( ESP_OK != mcpwm_set_pin(MCPWM_UNIT_0, &pin_config))
     {
      Serial.println(" ERROR: mcpwm_set_pin problem");
     }`
I expect, in case something is going wrong, the error message .... but the chip is hanging .... no message, red led is on .... 
no reaction at all.
If I comment the line with mcpwm_set_pin ... it works as expected ( but pin mode isn't set ).
Can see all debug messages.
The program is not finished, it's only the setup to do the final work, to debug and very the settings.
mcpwm_get_frequency tells me ... is running with 15Hz .... first assumption was data type error, but it isn't.
Up to 1MHz it works fine ....
Duty cycle should be 50.0% to make the max speed possible. But mcpwm_get_duty tells me it's 0.0%.
I expect I made some mistakes ... or one really big one.
I'm giving up at the moment .... need help.
Or is a better, more straight forward way possible to measure the distance between 2 signals in number of clocks?
The frequency of the signals is up to some 10 kHz, the distance between 0 and huge value ( < 0xFF FF FF FF ) .
Think ESP32 is a good choice for that. Next time I have to control a 180V 5A DC motor ( PWM based ). Hope I learned
all the tricky things around the mcpwm units up front ..... ;-)
Thanks for help.
Program is compiling, possible to upload,
watch the Serial output. 
It's hanging here ` if ( ESP_OK != mcpwm_set_pin(MCPWM_UNIT_0, &pin_config))`
as  long this line is active.

Pepito

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Tumbleweed / Suse

### Description:
1.) hanging at: if ( ESP_OK != mcpwm_set_pin(MCPWM_UNIT_0, &pin_config))
2.) frequency is 15Hz .... expected is 20MHz
3.) duty cycle is 0%  .... expected is 50%  

### Sketch:  
#include <Arduino.h> 
#include <string.h>
#include "esp_attr.h"
#include "soc/mcpwm_struct.h"
#include "driver/mcpwm.h"
#include "driver/timer.h"

#define ALPHA 52.0           // Winkel Pickup <-- OT  anpassen!

#define MCPWM_EN_CAPTURE 1
#define CAP0_INT_EN BIT(27)  //Capture 0 interrupt bit
#define CAP1_INT_EN BIT(28)  //Capture 1 interrupt bit
#define GPIO_CAP0_IN   23    //Set GPIO 23 as  CAP0  Pick-UP
#define GPIO_CAP1_IN   25    //Set GPIO 25 as  CAP1  Ignition
#define TIMER_DIVIDER   2    //  Hardware timer clock divider
#define TIMER_SCALE    (TIMER_BASE_CLK / TIMER_DIVIDER)  // convert counter value to seconds

typedef struct {
    timer_group_t timer_group;
    timer_idx_t   timer_idx;
    int alarm_interval;
    bool auto_reload;
} example_timer_info_t;

[... 250 lines cut ...]
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
ets Jun  8 2016 00:22:57
00:51:29.340 -> 
00:51:29.340 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:51:29.340 -> configsip: 0, SPIWP:0xee
00:51:29.340 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:51:29.340 -> mode:DIO, clock div:1
00:51:29.340 -> load:0x3fff0018,len:4
00:51:29.340 -> load:0x3fff001c,len:1216
00:51:29.340 -> ho 0 tail 12 room 4
00:51:29.340 -> load:0x40078000,len:10944
00:51:29.340 -> load:0x40080400,len:6388
00:51:29.340 -> entry 0x400806b4
00:51:29.439 -> E (132) psram: PSRAM ID read error: 0xffffffff
00:51:29.473 -> Configuring Timer 0
00:51:29.473 -> Timer_0 configuration:
00:51:29.473 ->     Divider = 2
00:51:29.473 -> auto reload = 1
00:51:29.473 -> counter_dir = 1
00:51:29.473 ->   intr_type = 0
00:51:29.473 ->  counter_en = 1
00:51:29.473 ->    alarm_en = 0
00:51:29.572 -> initializing mcpwm gpio...
00:51:29.572 -> E (241) MCPWM: /home/runner/work/e
Hanging forever .... reset works .... end up always here

lbernstone, 2021-04-23:
This forum is for issues with the code here, not issues with your code.
MCPWM is bound by the same restrictions as LED PWM. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#supported-range-of-frequency-and-duty-resolutions

Pepito-design, 2021-04-23:
Understand.  But why is it hanging here? If it fails, the function generates specific return value.
if ( ESP_OK != mcpwm_set_pin(MCPWM_UNIT_0, &pin_config))
 {
  Serial.println(" ERROR: mcpwm_set_pin problem");
 }
So the error message should appear. It's hanging inside mcpwm_set_pin ... and that's not my code ;-)
Thanks for help and have a good day
Pepito

closed 2021-07-08 by stale[bot]