espressif/arduino-esp32 #5104: ISR delayed

ericlangel, 2021-04-23:
I try to measure the time of a Servo pulse (1000us to 2000us with 50Hz) with a ESP32 Wroom
The measured time of a constant signal (1000us) is instable -> between 970us and 1030us

Code:

Setup(){
attachInterrupt(SERVO_IN, GetServo, CHANGE);
}

static void IRAM_ATTR GetServo() 
{
    uint32_t temp = ESP.getCycleCount();
    //if the pin has gone HIGH, record the microseconds since the Arduino started up 
    if(digitalRead(SERVO_IN) == HIGH) 
    {   
        //servo_start = micros();
        servo_start = temp;
        digitalWrite(LED_RED, HIGH);
    } 
    //otherwise, the pin has gone LOW 
    else
    {
      digitalWrite(LED_RED, LOW);
      //servo_tmp = ((volatile int)micros() - servo_start);
      servo_tmp = (temp - servo_start);
      servo_tmp = servo_tmp / 240;
      raw_analog_input = true;
      servo_start = 0;
    }
} 

With 1.0.4 its working with a measured time from 998us to 1002us
with 1.0.5 its measured time from 970us to 1030us -> tried it with micros() and getcyclecount()

ericlangel, 2021-04-23:
my Code had problems to compile with 1.0.4 because analogReadMilliVolts() was unknown, so i commented it out for test.

after the isr was working i copied the esp32-hal-adc.h and esp32-hal-adc.c from 1.0.5 to 1.0.4 and used analogReadMilliVolts() again.

V1.0.4 with analogReadMilliVolts() in loop() also has problems with ISR delay
V1.0.5 without analogReadMilliVolts() in loop() has no problems with ISR delay

me-no-dev, 2021-04-23:
that would point to ADC ISR having higher priority than the IO ISR, so you get IO ISR later when the newer ADC API is used. What are the results without ADC in your code? Also, you should look into RMT IN, which uses hardware to measure the pulses and will provide a much more adequate data back.

closed 2021-04-23 by me-no-dev

ericlangel, 2021-04-23:
is it possible to change ADC ISR priority?
Or IO ISR Priority?

ericlangel, 2021-04-26:
IO ISR should have more priority than ADC

this issue will cause many problems.

Or the User should be able to select priority.

why analogread() starts ADC ISR? -> better ADC performance?