espressif/arduino-esp32 #5106: Enable precompiled libraries in 1.0.4

cuttingedge1109, 2021-04-23:
https://github.com/espressif/arduino-esp32/pull/4209#issue-457033365
It is already fixed from 1.0.5.
I am using 1.0.4 and want to this function. Can you please make same changes on 1.0.4?
I already made changes on 1.0.4 of my forked repo using this way https://gist.github.com/danielestevez/2044589.
Regards.

Jason2866, 2021-04-26:
Is there a specific reason why you want to use a old outdated (buggy) version?

cuttingedge1109, 2021-04-26:
Just have several projects using that version yet.

Jason2866, 2021-04-26:
Do not expect that old outdated version will be active maintained. This is usual for sofware.
Time to update your projects ;-)

cuttingedge1109, 2021-04-26:
Yes, we are planning soon. Could you provide some guides or links how to upgrade from 1.0.4 to 1.0.6?
I am facing errors with 1.0.6.
But if backport didn't require much effort, can you do it?

lbernstone, 2021-04-26:
That would be "revisionist history".  The release is complete, and the assets provided are not going to be changed.  If you have issues migrating your application, you can ask about those issues.  Otherwise, you are welcome to fork the repo and modify it, as you have already done.

cuttingedge1109, 2021-04-26:
Ok, thank you, guys. 
I will close this issue, seems not valuable.

closed 2021-04-26 by cuttingedge1109