espressif/arduino-esp32 #5112: Using Serial.begin() more than one time crashes

VisualMicro, 2021-04-26:
It is a mistake to begin() the Serial multiple times but users often do but it would not normally crash the mcu.

This will crash

```
void setup()
{
  Serial.begin(115200);
  Serial.print("Hello World");
  Serial.begin(115200);
}

void loop() {}

```

lbernstone, 2021-04-26:
Please follow the issue template and indicate which device and version has the issue.  2.0.0alpha seems to have some problems in HardwareSerial.  Even initializing two uarts causes a crash.

VisualMicro, 2021-04-26:
> Please follow the issue template and indicate which device and version has the issue. 2.0.0alpha seems to have some problems in HardwareSerial. Even initializing two uarts causes a crash. Does not occur in esp32-s2 (@me-no-dev)

Sorry, I have added a link to the latest arduino-esp32 json, the crash info from the Serial Monitor and made a clearer statement of the issue.

Do you need anything else?

me-no-dev, 2021-04-27:
confirmed

pgrawehr, 2021-05-05:
Just observed a very similar issue. It does not crash, but fails to work after the second `Serial.begin()`. Try this simple sketch:

```cpp

void setup() {
  // initialize serial:
  Serial.begin(115200);
  // Serial.begin(115200); // If this line is uncommented, Serial.available() will always return 0. 
}

void loop() {
  // if there's any serial available, read it:
  while (Serial.available() > 0) {
    Serial.print("There's some data: ");
    // look for the next valid integer in the incoming serial stream:
    int red = Serial.read();
    Serial.println(red, HEX);
  }
  // Serial.print("I'm still alive"); // Enable this to see that the board does not crash
}
```
If the program is run with the second Serial.begin() commented out, everything works as expected and the sketch prints the hex values of whatever it gets over the serial console. If the line is enabled, nothing is ever printed. But the board is not dead, as can be shown by uncommenting the last line, which keeps being printed.

In case it matters, this is with the released version 1.0.6.

sansillusion, 2021-05-05:
Why would anyone run serial.begin() twice ?

pgrawehr, 2021-05-05:
@sansillusion In my particular case, I have a quite complex library, and I haven't found out where exactly the second call comes from, but it seems to happen while executing a software reset command (initiated over the serial communication). The library compiles and runs flawlessly on other boards. A more common case would be if you want to dynamically change the baudrate.

pgrawehr, 2021-05-07:
Update: While the above finding is still true (it reproduces easily), my problem was actually caused by setting the state of GPIO 1 and 3 to output during a software reset. Apparently, these two logical lines are used for the USB Uart and should not be touched. So I found a bug that wasn't even the cause of my problems...

VojtechBartoska, 2021-07-13:
keeping this open

closed 2021-07-14 by me-no-dev

SuGlider, 2021-07-16:
Issue fixed on PR https://github.com/espressif/arduino-esp32/pull/5385

commit f4f1c895 (2021-07-14): pinMatrixInDetach() has wrong parameter (#5385)

Fixes #5112
Call to pinMatrixInDetach() was changed from version 1.0.6 in version 2.0.0 injecting a bug as seen in cores/esp32/esp32-hal-uart.c
https://github.com/espressif/arduino-esp32/commit/80418fadcfb91c75d5100a8fddeb9318a8ef7d42
PR #5549 (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF
## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
commit b1d072df (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF (#5549)

## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.