espressif/arduino-esp32 #5116: Fix BLEAdvertising::stop()

marcidelux, 2021-04-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE with Platform.io
Flash Frequency:					40Mhz
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
Even after this fix: https://github.com/espressif/arduino-esp32/pull/3034
It still dosnt stop. I found out a misstype caused the issue.
At BLEAdvertising.cpp line 248 :
`esp_err_t errRc = ::esp_ble_gap_stop_advertising();`
Remove the two semicolons, and it will work afterwards.

lbernstone, 2021-04-27:
That is not a mistype.  The colons there indicate that there are two options for that function name, one in the local namespace, and one at the top level namespace. :: indicates to use the function at the top level namespace.

marcidelux, 2021-04-27:
Sorry even, my solution wasnt good, I reacted faster than I tested it properly.
So sadly the stop() function still dosn't work. I'm gona try some other things too.

chegewara, 2021-04-27:
Its weird, because i never saw the problem to stop advertising. You can always call this in you code, without using library:
```
esp_err_t errRc = esp_ble_gap_stop_advertising();
```

marcidelux, 2021-04-28:
Thanks for the help, it wasnt the situation in my case.
Now I truly can say that I have found the line what caused Advertising every time after I called stop Advertising, or I disconnected from the device.
At BLEServer.cpp --> Line 209: `startAdvertising(); //- do this with some delay from the loop()`
Afte I commented it out it works as I wanted to.
What is your opinion about this?

chegewara, 2021-04-28:
I believe you have some old library:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEServer.cpp#L209

marcidelux, 2021-04-28:
Yep, it was the issue.
I used framework-arduinoespressif32 3.10004.200129 (1.0.4)
After update to 1.0.6 it works fine.
Thanks for your fast help, thumbs up!

closed 2021-04-28 by marcidelux