espressif/arduino-esp32 #5117: How to configure DHCP server to provide DNS to AP to use esp32 as a repeater using LWIP NAT feature

paclema, 2021-04-27:
### Hardware:
Board:							mhetesp32minikit
Core Installation version:			1.0.6
IDE name:						Arduino IDE with Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:

I am trying to create an example for esp32 to use it as WiFi repeater using the recent features of LWIP NAT.

The example is posted in this repo: [esp32_lwip_nat_example](https://github.com/paclema/esp32_lwip_nat_example)

As you can see described in steps within the readme, to approach this I had to re-compile the esp-idf sdk to enable LWIP NAT options in order to be able to access to [lwip_napt.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/lwip_napt.h) methods. 

It looks that the ESP32 can be connected as a station and provide the AP, but the devices that I connect to the ESP32 do not get internet and I am not sure how I should configure the DHCP server to provide the DNSs received on the station.

I tried to call [dhcps_set_option_info](https://github.com/paclema/esp32_lwip_nat_example/blob/main/src/main.cpp#L69) and [dhcps_dns_setserver](https://github.com/paclema/esp32_lwip_nat_example/blob/main/src/main.cpp#L76) to configure it like in this esp-idf example [esp-idf-nat-example](https://github.com/jonask1337/esp-idf-nat-example/blob/master/main/main.c) but i have the next errors while linking:

```
Compiling .pio\build\mhetesp32minikit\src\main.cpp.o
Linking .pio\build\mhetesp32minikit\firmware.elf
c:/users/pacle/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\mhetesp32minikit\src\main.cpp.o:(.literal._Z5setupv+0x48): undefined 
reference to `dhcps_set_option_info(unsigned char, void*, unsigned int)'
c:/users/pacle/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\mhetesp32minikit\src\main.cpp.o:(.literal._Z5setupv+0x4c): undefined 
reference to `dhcps_dns_setserver(ip_addr const*)'
```

How should I configure the dhcp server to provide correctly the DNSs to the AP?
It looks that the [dhcpserver.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/lwip/include/apps/dhcpserver/dhcpserver.h) it is available but I am not sure how can I call it and configure it.

Thank you!

chegewara, 2021-04-27:
Have you seen this?
https://github.com/martin-ger/esp_wifi_repeater

paclema, 2021-04-27:
Yes, thank you @chegewara. I also have seen this other example: [RangeExtender-NAPT.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino) which I am actually trying to approach the same thing but using an esp32 instead of an esp8266. Both examples are made for esp8266 :(

chegewara, 2021-04-27:
Sorry, i mean this:
https://github.com/martin-ger/esp32_nat_router

paclema, 2021-04-27:
Yes, also this example is useful. It is made for esp32 in this case but using esp-idf framework and I look forward to create an example using this arduino-esp32 framework with platformio. 

The example that you shared also use the [dhcps_set_option_info](https://github.com/martin-ger/esp32_nat_router/blob/master/main/esp32_nat_router.c#L426) and [dhcps_dns_setserver](https://github.com/martin-ger/esp32_nat_router/blob/master/main/esp32_nat_router.c#L431) on those lines, and those are the functions that I am trying to use as I mentioned in the description but I have errors while linking and I don't know how to add that part

lbernstone, 2021-04-27:
You need to include the library where that function is defined.  In this case, at least <dhcpserver/dhcpserver.h>

paclema, 2021-04-27:
Thank you @lbernstone. The header is already included in the example that I am doing at this line https://github.com/paclema/esp32_lwip_nat_example/blob/main/src/main.cpp#L26. With VScode I can also check that it is imported correctly and I can navigate successfully to this file. However, appears the error that I mention at the end of the building process when It is linking.

lbernstone, 2021-04-27:
You have some incompatibility between your compiled libraries and your code.  You are running gcc 8.4, but the code you are trying to run is for IDF 3, which should use gcc 5.2.  If you don't understand what I am talking about, this is going to be very difficult for you to implement.

chegewara, 2021-04-27:
Which arduino-esp32 version are you using?  From description it is v1.0.6, but you are using code that is not enabled in that version. That header is from master branch:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/lwip/include/apps/dhcpserver/dhcpserver.h

paclema, 2021-04-27:
I am using a fork from this repo arduino-esp32 adding the "Enable IP forwarding" and "Enable NAT" feature from the esp-idf. I followed the steps explained in the [esp32_lwip_nat_example readme](https://github.com/paclema/esp32_lwip_nat_example) to rebuild the sdk for the esp32 using the [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder). 

This fork recompiling the sdk was the only method I could find to get the IP_NAPT defined, if not I am not able to use the [lwip/lwip_napt.h](https://github.com/paclema/esp32_lwip_nat_example/blob/main/src/main.cpp#L21) functions because these sdk functions are not enabled by default.

I am going to check if there is some mismatch with the IDF version

chegewara, 2021-04-27:
> Board: mhetesp32minikit
Core Installation version: 1.0.6 forked: https://github.com/paclema/arduino-esp32/commits/lwip_enabled
IDE name: Arduino IDE with Platform.io

It is really hard to help if you are providing inconsistent information about your setup.
So, is it v1.0.6 or v2.0.0-alpha1?

paclema, 2021-04-27:
Sorry, this was not so clear for me. Since my fork is at the top of the current master branch, I suppose that the version is v2.0.0-alpha1 + LWIP nat lib https://github.com/paclema/arduino-esp32/commits/lwip_enabled

chegewara, 2021-04-27:
You enabled few options in sdkconfig/menuconfig. Best way to test if changes are aplied is to test in code by checking with `#ifdef ...`. That may give you a clue.

paclema, 2021-04-27:
Yes, I made that test with `lwip/lwip_napt.h`. I check it [in this line](https://github.com/paclema/esp32_lwip_nat_example/blob/main/src/main.cpp#L18) for example. 
Configuring platformio.ini using platform_package = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master tells me that this sdkconfig/menuconfig config is not enabled and I can not use lwip_napt.h. But if I use my fork instead [check this platformio.ini](https://github.com/paclema/esp32_lwip_nat_example/blob/main/platformio.ini#L23) I don't have this problem because I have enabled the necessary sdkconfig/menuconfig.

```
#ifdef __DHCPS_H__
Serial.println("all should works");
#endif
```
I have added that and "all should works". Also as I said, I can `#include "dhcpserver/dhcpserver.h"` without any problem. The problem is when linking :/

chegewara, 2021-04-27:
What i meant, try it in empty sketch, without functions that prevent you from building it. Just add headers.

chegewara, 2021-04-27:
Ok, i see what is wrong. That header file is nor C++ compatible:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/lwip/include/apps/dhcpserver/dhcpserver.h

You have to add header with:
```
extern "C {
   "dhcpserver/dhcpserver.h"
}
```

paclema, 2021-04-27:
I tried adding:

```
extern "C" {
  #include "dhcpserver/dhcpserver.h"
  #include "dhcpserver/dhcpserver_options.h"
}
```
And it also includes the libraries correctly but if I uncomment the functions `dhcps_set_option_info` and `dhcps_dns_setserver` I still get the linking errors.

I am not sure if this [lwip/apps/dhcpserver/dhcpserver.c](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/lwip/apps/dhcpserver/dhcpserver.c) is perhaps not added correctly into the sdk libraries?

chegewara, 2021-04-27:
You seems to not have `ESP_DHCP` enabled/defined somewhere.
https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/lwip/apps/dhcpserver/dhcpserver.c#L28

paclema, 2021-04-27:
Also tried to defined within platformio.ini build_flags or main.cpp code and the result was not successful. Although I am not sure if it should also be defined before compiling the sdk.

chegewara, 2021-04-27:
Yes, most likely before compiling sdk with builder.

paclema, 2021-04-27:
I tried to build the sdk enabling this `ESP_DHCP` define but did not work neither

chegewara, 2021-04-27:
https://github.com/espressif/esp-idf/blob/9d34a1cd42f6f63b3c699c3fe8ec7216dd56f36a/components/lwip/port/esp32/include/lwipopts.h#L973

paclema, 2021-04-30:
I have found the solution based on [this post](https://esp32.com/viewtopic.php?t=3761#p56131):

* Instead to use `dhcps_set_option_info()` I have use `tcpip_adapter_dhcps_option()` to supply dns server via dhcps.
* Instead to use `dhcps_dns_setserver()` I have use `tcpip_adapter_set_dns_info()` to provide the DNS obtained from the esp32 as station (you can get them with `tcpip_adapter_get_dns_info()` )

You can check how I could configure properly the example in this part: https://github.com/paclema/esp32_lwip_nat_example/blob/main/src/main.cpp#L120

Using those functions the linking errors do not appear (because I don't use those functions anymore :P)  and I have tested that stations connected to the ESP32 AP can get internet access too.

I wonder if in the future @me-no-dev IP forwarding and NAT will be enabled by default for the LWIP component (as described in [the readme fo the example](https://github.com/paclema/esp32_lwip_nat_example)). In that case, I could add this example to this repo if you would consider it useful. Until then, as it is written in the description, I am using a for of this framework with the esp32 sdk recompiled enabling these features.

pablofr918, 2021-06-23:
Hello @paclema. I have been looking for this for weeks. The unique difference is that I would need to have this implemented in Arduino and I wonder if you know the way to make this up.

a1exande, 2021-08-30:
Hello @paclema.  Add the following directives to the **"dhcpserver.h"** to solve the link issues:

after initial includes:
**#ifdef __cplusplus
extern "C" {
#endif**

before final #endif:
**#ifdef __cplusplus
}
#endif**

johnnytolengo, 2021-09-21:
Is that issue already solved? I am interested to enable the IP_NAPT in my platformio esp32
if yes, could you share how to proceed?

mtnbkr88, 2022-01-21:
I am also interested in using IP_NAPT in my arduino-ESP32 development environment.

VojtechBartoska, 2022-04-07:
Hello folks, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? 

Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-21:
Hello,

as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable.

If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-21 by VojtechBartoska

as-iotex, 2024-02-19:
@VojtechBartoska Hello, could you let us know what the state of this is?
> Hello folks, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid?

You were asking this, but I don't see that NAPT has been enabled for 2.0.3-RC
Has NAPT been enabled and can you tell us for which release?