espressif/arduino-esp32 #5119: ESp32-C3 Arduino IDE Compilation on Any Sketch.

salmanfarisvp, 2021-04-28:
### Hardware:
Board:							ESP32-C3-DevKitM-1
Core Installation version:			2.0.0-alpha1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I was trying to upload an example application on ESP32-C3 with Arduino IDE, but getting a compilation error, tried a different inbuilt example, but the same error. 

### Sketch:  

```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```

### Debug Messages:

```
Arduino: 1.8.13 (Windows 10), Board: "ESP32C3 Dev Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Faris\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Faris\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Faris\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32c3:PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\Faris\AppData\Local\Temp\arduino_build_558238 -warnings=none -build-cache C:\Users\Faris\AppData\Local\Temp\arduino_cache_205364 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2020r3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2020r3.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2020r3 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-crosstool-ng-1.24.0-123-g64eb9ff -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-crosstool-ng-1.24.0-123-g64eb9ff.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-crosstool-ng-1.24.0-123-g64eb9ff -prefs=runtime.tools.mkspiffs.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2020r3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2020r3.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2020r3 -prefs=runtime.tools.mklittlefs.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.esptool_py.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi\examples\WiFiScan\WiFiScan.ino
[... 59 lines cut ...]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version,get_security_info}

               ...

esptool: error: argument --chip/-c: invalid choice: 'esp32c3' (choose from 'auto', 'esp8266', 'esp32', 'esp32s2')

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

 Not used: C:\Users\Faris\Documents\Arduino\libraries\Seeed_Arduino_atWiFi-master

 Not used: C:\Users\Faris\Documents\Arduino\libraries\Seeed_Arduino_rpcWiFi-master

Using library WiFi at version 1.0 in folder: C:\Users\Faris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFi 

exit status 2

C:\Program Files (x86)\Arduino\arduino-builder returned 2

Error compiling for board ESP32C3 Dev Module.

```

lbernstone, 2021-04-28:
esptool has not yet been updated in the alpha.  It can be added to the list. @VojtechBartoska

Jason2866, 2021-04-28:
I use PlatformIO. Did a custom setup since the C3 is not yet official supported from PlatformIO.
Blink example is working...
```
[env:test_env]
board = esp32c3
framework = arduino
platform                    = https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3
platform_packages           = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
                              tasmota/toolchain-riscv32
```

rinscr3003, 2021-06-02:
> esptool.exe has not yet been updated in the alpha. It can be added to the list. @VojtechBartoska

Had this been fixed in the current version?

Jason2866, 2021-06-02:
> > esptool.exe has not yet been updated in the alpha. It can be added to the list. @VojtechBartoska
> 
> Had this been fixed in the current version?

It is with commit https://github.com/espressif/arduino-esp32/commit/0db9e2f45b50a261dad997701af5c892bf198855

SuGlider, 2021-07-19:
I've tested some examples with the latest version 2.0.0Alpha and confirm that the example posted by @salmanfarisvp compiles and runs using Arduino IDE 1.8.15 and Win10, using as target ESP32-C3.

Therefore, I'm closing this issue. Please feel free to reopen it if necessary.

closed 2021-07-19 by SuGlider