espressif/arduino-esp32 #5125: About EEPROM  storage

hzhh110, 2021-04-29:
[image]

What is the maximum number of bytes that can be stored in EEPROM? I found that when I store more than 3,400, it will crash, but in the actual project, I will crash in more than 2,000. After trying many ways, I still can’t solve it. Is there any solution? What about the problem?

DeviceDetails loadDetailsConfig()
{
  DeviceDetails config;
  int size = sizeof(config);
  memset(&config, 0, size);
  EEPROM.begin(size);
  uint8_t *p = (uint8_t *)(&config);
  for (int i = 0; i < size; i++)
  {
    *(p + i) = EEPROM.read(i);
  }
  EEPROM.commit();
  EEPROM.end();
  return config;
}
void saveDetailsConfig(DeviceDetails details)
{
  int size = sizeof(details);
  EEPROM.begin(size);
  uint8_t *p = (uint8_t *)(&details);
  for (int i = 0; i < size; i++)
  {
    EEPROM.write(i, *(p + i));
  }
  EEPROM.commit();
  EEPROM.end();
}

hzhh110, 2021-04-29:
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x400882dc  PS      : 0x00060336  A0      : 0x40083870  A1      : 0x3ffc1f50  
A2      : 0x3ff000e0  A3      : 0x00000001  A4      : 0x3ffbde6c  A5      : 0x00000001  
A6      : 0x00060323  A7      : 0x00000000  A8      : 0x80082c5e  A9      : 0x3ffc1ff0  
A10     : 0x00000000  A11     : 0x3ffba5fc  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060320  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

hzhh110, 2021-04-29:
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (BTC_TASK) 
Core 0 register dump:
PC      : 0x4008e2c4  PS      : 0x00060736  A0      : 0x3ffe0540  A1      : 0x3ffe0480  
A2      : 0x3ffde364  A3      : 0x3ffe0880  A4      : 0x00000001  A5      : 0x00000010  
A6      : 0x00000002  A7      : 0x3ffe07a4  A8      : 0x00000000  A9      : 0x00000002  
A10     : 0x3ffe09fc  A11     : 0x3ffe0762  A12     : 0x3ffe09fe  A13     : 0x3fff0ec0  
A14     : 0x0000003e  A15     : 0x6e776f6e  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

hzhh110, 2021-04-29:
ELF file SHA256: 0000000000000000

Backtrace: 0x400882dc:0x3ffc1f50 0x4008386d:0x3ffc2030 0x40085f6d:0x3ffc2050 0x400e2bd6:0x3ffc2090 0x400e2c9e:0x3ffc20b0 0x40083c19:0x3ffc20d0 0x4008452b:0x3ffc20f0 0x400e7d24:0x3ffc2110 0x400eb242:0x3ffc2130 0x400e7cb0:0x3ffc2150 0x40083ef2:0x3ffc2170 0x400e79bd:0x3ffc21c0 0x400e5e22:0x3ffc21e0 0x400e618d:0x3ffc2200 0x400e51c0:0x3ffc2280 0x400e5752:0x3ffc2300 0x400e4b9b:0x3ffc2370 0x400d11c4:0x3ffc23b0 0x400d0ea7:0x3ffc23e0 0x400d0f40:0x3ffc2400 0x400d2632:0x3ffc3ec0 0x40086125:0x3ffc3ee0

hzhh110, 2021-04-29:
platform = espressif32

board = nodemcu-32s

lbernstone, 2021-04-29:
This has nothing to do with EEPROM, which gets its size limits from your nvs partition size.  You are using all your stack memory.  String variables will use heap memory.  EEPROM is provided for compatibility with the Arduino spec, but you should use the Preferences library if your code will only be used on ESP32.

hzhh110, 2021-04-30:
How to clear nvs? It is found that nvs_erase_all(handle); after the chip cannot read and write, it has been ESP_ERR_NVS_NOT_FOUND, version: 3.2.0
state
used_entries:
350
free_entries:
280
total_entries:
630
namespace_count:
12

closed 2021-07-14 by stale[bot]