espressif/arduino-esp32 #5137: ESP32 fails to boot when a SPI master is sending it data

coriolan-v, 2021-04-30:
Hi everyone,
I've got a ESP32-WROOM-32UE working quite well.
It is a SPI slave to another device.
The pinout is as follow:
- MOSI: GPIO 23
- MISO GPIO 19
- SCK: GPIO 18
- /CS: GPIO 5
- CS (the same /CS version but inverted, so that the rising edge can wake up the MCU when it's sleeping): GPIO 13

If my SPI master is active and sending data to the ESP32 when it is not powered, and I turn on the ESP32, it sometimes fails to boot. This happens maybe 5% of the time, and difficult to reproduce. The only way is to pull the plug and power it again.

I have looked closely at the USB serial data coming from the ESP32 when it fails to boot and I see nothing instead of the usual date, time, and "rst:0xxx" thing. That indicates (I think) that the bootloader doesn't even start.

On the oscilloscope the EN and GPIO0 signals are 3.3V, as normal.

Is there a specific thing that I am doing that would prevent the MCU from booting in this scenario?

Thank you!

chegewara, 2021-04-30:
That `"rst:0xxx" thing` is important, so if you could post some logs, also which esp-idf version/commit you are using and chip revision.

lbernstone, 2021-04-30:
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message can help you interpret why it won't boot, as per @chegewara.

coriolan-v, 2021-05-01:
Hello,

This is the message I get when the MCU **successfully boots**:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) 
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac`

However, when the MCU does not boot, I get nothing, no data on the serial port:
` `

Is there another way to diagnose if this message precisely is not coming up?

Thank you

lbernstone, 2021-05-01:
The reasons why there is no info at all:
1) EN pin is grounded (you would probably have checked if this were possible)
2) gpio2 is pulled up while gpio0 is pulled down (you would probably sometimes see the downloading message)
3) gpio12 is pulled up.  This causes the vdd_sd to be 1.8V, so your flash is inaccessible.
4) gpio15 is pulled down.  This disables logging on uart0.

If you are using gpio12, I would recommend [burning the efuse](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12-esp32-only).

coriolan-v, 2021-05-01:
Hello, I have looked at all the signals using the oscilloscope:
1. EN is 3.3V
2. GPIO2 is pulled down trough a 5.1kOhm resistor.
3. GPIO12 is pulled down trough a 10kOhm resistor.

lbernstone, 2021-05-01:
gpio12 is the most likely suspect.  It is pretty safe to burn the efuse unless you plan on changing to internal 1.8V memory at some time :smile:

coriolan-v, 2021-05-24:
It looks like [ burning the efuse](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12-esp32-only) (`components/esptool_py/esptool/espefuse.py set_flash_voltage 3.3V -p COMXXX`) solved the issue, I could not recreate it... 

Thank you community!

closed 2021-05-24 by coriolan-v