espressif/arduino-esp32 #5139: ESP32-S2 CDC doesn't show up after flashing in download mode

UnexpectedMaker, 2021-04-30:
While in the beta branch, any S2 boards that were set to use CDC, that had a VID/PID, would appear in the ports list after initial; flashing in download mode (and reset) as the board name (eg: `/dev/cu.usbmodem12345 (UM FeatherS2)` - but this no longer works.

Once reset, the board doesn't appear at all in the post list and requires going back into download mode again to been seen and flashed again.

Tested on WIn 10 and macOS Catalina with both FeatherS2 and TinyS2 boards.  I cant test using `ESP32S2 Dev Board` as that has no menu for CDC.

pedrominatel, 2021-05-03:
Hi @UnexpectedMaker,

Is something similar you are trying to do? https://youtu.be/_I-MI8_gGM8?t=4189 (It's in Portuguese, sorry :( )
I was able to use it directly with CDC. I've tested with Ubuntu 20.04.

UnexpectedMaker, 2021-05-06:
> Hi @UnexpectedMaker,
> 
> Is something similar you are trying to do? https://youtu.be/_I-MI8_gGM8?t=4189 (It's in Portuguese, sorry :( )
> I was able to use it directly with CDC. I've tested with Ubuntu 20.04.

I'm not really sure what that video is showing, sorry.

UnexpectedMaker, 2021-05-06:
@me-no-dev @chegewara I've found the issue and how to reproduce it and why others are not seeing it:
https://www.dropbox.com/s/3as58ok0qzckri3/ESP32-S2-CDC-Issue-repo.mov?dl=0

If Serial.begin() is present, it works. If Serial.begin() is not present, then the CDC doesn't enumerate and teh user has to go into download mode to flash the board every time.

So I guess this is still linked to the previous issue about CDC and Serial, but for use cases where serial is not needed, the CDC should still enumerate as a device to flash to.

Thanks!

chegewara, 2021-05-06:
Yes, it is exactly what i described in the other issue. `Serial.begin()` is mandatory to start USB, without it USB just send descriptors and thats it.

UnexpectedMaker, 2021-05-06:
> Yes, it is exactly what i described in the other issue. `Serial.begin()` is mandatory to start USB, without it USB just send descriptors and thats it.

Ok, sorry I created extra noise about it ;)

My focus was on no CDC enumerating and not that Serial wasn't working, so I was just focused on the enumeration/flashing side of the issue.

Do we have an idea about how to fix this? It used to work fine in the S2 beta - and as I mentioned in the video, there used to be another CDC option in the menu, one for `which USB` was being used, and one for `which UART serial would output to`.

chegewara, 2021-05-06:
When you select `Serial connected to: USB` you have to call `Serial.begin()` to flash over USB next time. It is how it is.
It always was like that. That line of code is required to fully initialize USB CDC.

UnexpectedMaker, 2021-05-06:
> When you select `Serial connected to: USB` you have to call `Serial.begin()` to flash over USB next time. It is how it is.
> It always was like that. That line of code is required to fully initialize USB CDC.

I don't agree - this is not how it used to work while in Beta. I was the one working with @me-no-dev on how this initial worked, and there wasn't even a "serial output goes to...." menu option when CDC support was first implemented.

It was possible back then to flash to the CDC and have it enumerate before we could even direct serial output to the CDC - that came after.

This is a BUG. This should 100% NOT work this way. As I said, there used to be 2 menu options. One has been removed. the one that allowed the user to decide if they were using the CDC or external Serial2UART.

If this is how it's intended to work, then it's a terrible implementation. It's not obvious or user friendly in any way.

Having the CDC enumerate has nothing to do with having serial output enabled or not. Not every project uses serial out.

This workflow has to change.

UnexpectedMaker, 2021-05-07:
Just to be clear... I should be able to connect a FeatherS2 to my computer, select I am using the CDC, flash an empty Arduino project onto it..  just an empty setup(){} and loop(){} and after I flash it and it restarts, it should appear in my ports list so I can flash it again and again, no matter what it in my firmware.

A user should not have to put any "required" code in their own firmware for CDC to kick in.

UnexpectedMaker, 2021-05-18:
Here is a support request I had on my discord server today, one of many I have to deal with showing this exact issue and why this behaviour needs to be fixed.  I have permission from the OP to post this....
[image]

chegewara, 2021-05-18:
@me-no-dev

me-no-dev, 2021-06-17:
what that user describes is expected. You do use a different serial for your boards so the enumerated serial will change.

1. puts manually the board into download mode
2. flashes the board, but since this is USB flash with manual enter into download mode, the chip can not reboot into the app on it's own
3. reboots manually into the app, but now the app starts the CDC with different serial (your feather s2 one) so the port address changes to some other /dev/cu.usbmodemXXXXXX

persistence or not, this will always be the case :)

UnexpectedMaker, 2021-06-17:
> what that user describes is expected. You do use a different serial for your boards so the enumerated serial will change.
>
> 1. puts manually the board into download mode
> 2. flashes the board, but [...]

Yup, I understand that (Though the ERROR:... is going to be changed to something less dramatic it seems - https://github.com/espressif/esptool/issues/631 )

But none of this addresses my issue and what I believe is a bug. The FeatherS2 in this case never presents itself as a new port `unless` the user has put `Serial.begin()` in their code.

If I choose the CDC, then it should enumerate and appear as a port no matter if I put Serial.begin() or not in my code. I've explicitly stated I am using the CDC, and therefor expect that the device appears for me after it restarts.

igrr, 2021-06-18:
I guess we can call USBCDC::begin after USB::begin in https://github.com/espressif/arduino-esp32/blob/21947ebe76cf6380fe06aa885390b86cc477709a/cores/esp32/main.cpp#L50-L52, maybe with a smaller default RX buffer size. It doesn't seem like a lot more code would be linked into the application if this is done, so the cost of starting CDC early is not too high.

Alternatively if the global change is not desirable, this can be made a board-specific thing, in one of two ways:
1. in main.cpp, add `Serial.begin()` guarded by `ARDUINO_SERIAL_PORT == 1 && ARDUINO_AUTO_START_CDC`, then define ARDUINO_AUTO_START_CDC macro for a specific board via boards.txt
2. provide board-specific `initVariant` function in https://github.com/espressif/arduino-esp32/tree/master/variants/um_feathers2, which would call Serial.begin if `ARDUINO_SERIAL_PORT == 1`.

UnexpectedMaker, 2021-06-20:
> I guess we can call USBCDC::begin after USB::begin in
> 
> https://github.com/espressif/arduino-esp32/blob/21947ebe76cf6380fe06aa885390b86cc477709a/cores/esp32/main.cpp#L50-L52
> 
> , maybe with a smaller default RX buffer size. It doesn't seem like a lot more code would be linked into the application if this is done, so the cost of starting CDC early is not too high.
> Alternatively if the global change is not desirable, this can be made a board-specific thing, in one of two ways:
> 
> 1. in main.cpp, add `Serial.begin()` guarded by `ARDUINO_SERIAL_PORT == 1 && ARDUINO_AUTO_START_CDC`, then define ARDUINO_AUTO_START_CDC macro for a specific board via boards.txt
> 2. provide board-specific `initVariant` function in https://github.com/espressif/arduino-esp32/tree/master/variants/um_feathers2, which would call Serial.begin if `ARDUINO_SERIAL_PORT == 1`. `initVariant` is called before `setup`, and any custom logic can be added there.

Hey Ivan :)

I can't think of any use case or situation where if I chose CDC from the menu that I would't want the USB to enumerate, so I think this should 100% be a global setting when using CDC, and not something at the board level.

Can you give me an example of a situation where a user might choose CDC but not want the CDC to actually enumerate to be used?

me-no-dev, 2021-07-20:
Can you please retry with [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino) sketch and current master?

UnexpectedMaker, 2021-07-20:
> Can you please retry with [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino) sketch and current master?

Hiya!!!!

Not really sure what that sketch is supposed to do.. I only have CDC USB on my boards, so I have no way of seeing the output I don't think.

What I will say is this is now (kinda) working - if I flash blink on my FeatherS2 without any Serial.begin() my CDC enumerates after reset - yay!

But...  when flashing via the CDC, the board doesn't auto reset after flash, but also shows no reset error or notification that the user has to reset the board to make the code start running.

So flashing blink to CDC does nothing until after pressing reset. Hopefully just a bug :)

Thanks for working on this!

igrr, 2021-07-20:
> So flashing blink to CDC does nothing until after pressing reset. Hopefully just a bug :)

Is this in the case when you initially pull GPIO0 low to put the board into download mode? Or you already have some CDC-capable program loaded, and flashing is done without forcing GPIO0 low?

UnexpectedMaker, 2021-07-20:
> > So flashing blink to CDC does nothing until after pressing reset. Hopefully just a bug :)
> 
> Is this in the case when you initially pull GPIO0 low to put the board into download mode? Or you already have some CDC-capable program loaded, and flashing is done without forcing GPIO0 low?

Hey Ivan :)

This is after I have CDC capable code running. I can re-flash blink again and again and it will never reset and start running  unless I press reset manually.

me-no-dev, 2021-07-20:
@UnexpectedMaker this sketch works on all USB boards. It will be very helpful if you have another USB-UART connected to UART0 to see the logs. I have recently added changes to the USB code and with current mater can upload fine with all boards but one, which has it's D+and D- lines swapped. I even tried with selecting FeatherS2 as board. (Note: tested on both Windows10 and OS X Big Sur)

UnexpectedMaker, 2021-07-20:
@me-no-dev my uploading is working fine now with your changes - just no auto reset happening after upload, well, not on this FeatherS2 anyway. I can try some other boards tomorrow and I'll get back to you :)

me-no-dev, 2021-07-20:
wait... what do you mean by auto-reset? how did you enter download mode? hit upload while the sketch is running or manually with buttons?

UnexpectedMaker, 2021-07-20:
No, this is nothing to do with download mode. This is after the very first flash using download mode, then switch to CDC and flash again. Flashing via CDC doesn't reset the board after flashing it. I still have to manually hit reset.

it gets to the end of the flashing....
```Hash of data verified.
Compressed 3072 bytes to 125...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (125 compressed) at 0x00008000 in 0.0 seconds (effective 590.7 kbit/s)...
Hash of data verified.

Leaving...```

But doesn't reset the board for me like any other board does. So no blinking LED until after I hit reset.

This happens with or without Serial.begin()

me-no-dev, 2021-07-20:
huh? this is really strange. Please attach UART to RX0/TX0 so we can see what is going on (and use the sketch I pointed to)

me-no-dev, 2021-07-20:
You did update all tools with get.exe, right?

VojtechBartoska, 2021-07-29:
@UnexpectedMaker Can you please take a look on proposed questions? We will appreciate it!

VojtechBartoska, 2021-09-22:
@me-no-dev Can we closed this?

me-no-dev, 2021-09-22:
I guess. Surely @UnexpectedMaker can ping us if here are more issues

closed 2021-09-22 by me-no-dev