espressif/arduino-esp32 #5143: FreeRTOS V8.2 to V10.2 for arduino IDE

Rp777, 2021-05-02:
Hi All,

Are we going to get an update for freeRTOS from V8.2 to V10.2 for Arduino IDE?
When i install the ESP32 package on Arduino IDE V1.8.13, I see in the list that it still uses FreeRTOS V8.2
How can we upgrade to V10.2 in Arduino IDE?
Any pointers on this would be much helpful.

I know that we can use Arduino code to compile under ESP-IDF but as of now I do not want to use ESP-IDF .

I see that ESP-IDF now uses V10.2

lbernstone, 2021-05-02:
Already included in the [development release](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md).

Rp777, 2021-05-03:
> Already included in the [development release](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md).
@lbernstone Thank you.!

closed 2021-05-03 by Rp777