espressif/arduino-esp32 #5146: WiFi.RSSI(); method breaks MQTTS connection

cardtap, 2021-05-03:
### Hardware:
Board:							ESP32-WROOM-32UE 16MB
IDE name:						Platform.io
Flash Frequency:					240Mhz
Computer OS:                                         Windows 10
Latest version of frameworks.

### Description:
**Goal:**
Periodically read signal strength (rssi) of connection between ESP32 and an access point that the ESP32 is connected to.

**System:**
ESP32-WROOM-32 16MB, Windows 10, PlatformIO on VSCode, latest version of frameworks.

**Failure point:**
MQTTS connection (MQTT using WiFiClientSecure) works great and receives all messages normally up until WiFi.RSSI(); is called, which somehow breaks the SSL connection and makes the MQTT connection unstable or completely fail and lockout.

I am currently using PubSubClient for MQTT communications but I've spent the past week and 12+ hours trying to find this bug (finally realized it was WiFi.RSSI) and so I have tested two other MQTT libraries as well.

**Code to cause error:** (single line of code called anywhere causes this)
`int8_t rssi = WiFi.RSSI();`

**Error message:**
```
[LOG]DISP : Updated RSSI 1 : -49
Must reconnect
Attempting MQTT connection
[ 44456][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():184]: (-8576) X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
[ 44459][E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -8576
failed, rc=-2 try again in 5 seconds
Attempting MQTT connection
[ 49503][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():184]: (-8576) X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
[ 49505][E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -8576
failed, rc=-2 try again in 5 seconds
```

After that loops around 10 times...
```
[463677][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():138]: (-10368) X509 - Allocation of memory failed
[463677][E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -10368
failed, rc=-2 try again in 5 seconds
Attempting MQTT connection
[468688][E][ssl_client.cpp:98] start_ssl_client(): Connect to Server failed!
[468688][E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
failed, rc=-2 try again in 5 seconds
Attempting MQTT connection
```

amarotica, 2021-06-20:
Has anyone figured out a fix for this? I am experiencing the exact same issue where WiFi.RSSI() breaks the TLS connection.

amarotica, 2021-08-21:
> [STALE_SET] This issue has been automatically marked as stale because it has not had recent activity. It will be closed in 14 days if no further activity occurs. Thank you for your contributions.

Not stale, still have this issue!

cardtap, 2021-10-29:
Has anyone else ran into this issue of WiFi.RSSI(); breaking a secure MQTT connection? I've temporarily disabled RSSI measurements but it would be awesome if this might work again in the future.

Thanks for any ideas!

VojtechBartoska, 2022-04-07:
Hello, can you test this with v2.0.3-RC1? Thanks

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska