espressif/arduino-esp32 #5147: How the get time that spent on deep sleep

dogancignakli, 2021-05-03:
Hi,
Can i get how much time the esp32 spent while deep sleep mode active? On my project, I am using deep sleep feature. My esp sleeping 12 hour a day. If  we use gpio while esp sleeping, my device's next sleep time will be shifted. So I have to calculate how much time that i spent in deep sleep time until gpio wake up occurs.
Is there any ways like millis function while esp sleeping?

lbernstone, 2021-05-03:
RTC time will be running while the cpu is asleep, if it is waking on a timer.
This can be read with rtc_time_get() (ticks) or just time(NULL) (seconds).
https://github.com/lbernstone/NTP_sleep

dogancignakli, 2021-05-03:
Can i convert this ticks into hours or minutes? I need this duration only. Not actual time. Only seconds or minutes that elapsed. For instance my device will awake per 12 hour a day via timer. If some one, push wake up button while device sleeping( after 30 min sleep for example) i substract this 30 mins from my 12*60 min interval.So I will wake up 11 hour 30 min later again.

lbernstone, 2021-05-03:
You can use time(NULL) if you don't need micros precision.

closed 2021-07-23 by stale[bot]