espressif/arduino-esp32 #5148: WiFi code causing loop

DuckyIsHere, 2021-05-03:
### Description:
Whenever I try to use wifi code it breaks and ends up in some kind of loop (I keep hearing my computer beeping like a device is being connected)

This code works fine until I uncomment 'WiFi.mode(WIFI_STA). No example sketches work either. 'Test2' is output first time but never again, including when I press en button.

### Sketch:  
```
#include "WiFi.h"
// the setup function runs once when you press reset or power the board
int led = 13;
const char* ssid = "WiFi-1E78";
const char* password =  "67558322";
void setup() {
  Serial.begin(9600);
  pinMode(led, OUTPUT);
   Serial.println("Test1");
  delay(1000);
   Serial.println("Test2");
  //WiFi.mode(WIFI_STA);
  //WiFi.begin(ssid, password);
  Serial.println("Test4");
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("Test3");
  Serial.println(WiFi.status());
  delay(1000);                       // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

lbernstone, 2021-05-03:
Your usb port is not providing enough power.  Try a different cable or port.

sansillusion, 2021-05-03:
It could be that your computer or usb cable does not provide enough
amperage to run the wifi. Have you tried with an external power source ?

YordanYanakiev, 2021-05-04:
Wonder if this hapens on battery 3.7V ?

VojtechBartoska, 2021-05-06:
@DuckyIsHere Did you solve your problem with different power source?

closed 2021-08-04 by stale[bot]

YordanYanakiev, 2021-08-05:
I am reopening it, since I don't believe there is any sort of "gone by itself" type of bug here.
This is absurdly problematic bug.

me-no-dev, 2021-08-10:
@YordanYanakiev this is some sort of hardware bug (LDO, cable, bad solder, etc.). The software is all the same and 99.99% do not have issues. Arduino is software. No reason to reopen