espressif/arduino-esp32 #5149: every Task running 2 times

nimaltd, 2021-05-04:
I am using platformio and esp-wroom32 . core 5.1.1
i create a new task. but my task run 2 times 
[image]

smarq8, 2021-05-04:
please share full code or prepare simplified example, screenshot is not suefficient to determine what happen.

nimaltd, 2021-05-04:
in setup

nimaltd, 2021-05-04:
TaskHandle_t                        hTaskASK;

void TaskASK(void * pvParameters);

void setup()
{
  xTaskCreatePinnedToCore(
        TaskASK,   /* Task function. */
        "TaskASK", /* name of task. */
        10000,     /* Stack size of task */
        NULL,      /* parameter of the task */
        4,         /* priority of the task */
        &hTaskASK, /* Task handle to keep track of created task */
        0);        /* pin task to core 0 */
}

and my new task 

void TaskASK(void * pvParameters)
{
    int x;
 
   while (1)
    {
   
        x++;
        Serial.println(x);
        delay(1000);
    }
}

nimaltd, 2021-05-04:
I found it. I created a init() function. I think in arduino platform it called again.

closed 2021-05-04 by nimaltd