espressif/arduino-esp32 #5150: Disconnect of WiFiClient When Buffer Read

Xylopyrographer, 2021-05-04:
### Hardware:
Board:							M5Stack ATOM Matrix (esp32 PICO-D4)
Core Installation version:			1.0.6
IDE name:						Arduino IDE 1.8.13
Flash Frequency:					40MHz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                       macOS 10.15.7

### Description:
Problem: The WiFiClient receives data from a host. The call to `client.available()` is > 0. Reading the first character in the buffer causes an error in the `WiFiClient.cpp` library which then disconnects the client. 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
Client Disconect Error code snippet

    char reply = '\0';
    int maxloops = 0;
    tally.tState = "nojoy";
    tally.tConnect = false;
    tally.tTimeout = false;
    tally.tNoReply = true;
    
    if (!stClient.connected()) {

        Serial.println("\r\nAbout to call stClient.connect");
        
        if (stClient.connect(stIP, stPort)) {
            
            Serial.println("Return from call to stClient.connect --> SUCCESS\r\n");
            tally.tConnect = true;
        }
        else {
            Serial.println("Return from call to stClient.connect --> FAIL\r\n");
            return tally;
        }
    }
    
    stClient.print("GET /tally/6/status\r\n\r\n");      // send the status request to the server

   delay(1000); 

    while (stClient.available() == 0 && maxloops < 1000) {    // Wait a maximum of 1 second for the ST server's reply to become available
        maxloops++;
        delay(1);                                             // wait a tick before checking again if there is a response from the ST Server.
    }
    if (maxloops == 1000) {         // response from the ST Server timed out
        tally.tTimeout = true;
        return tally;
    }

    Serial.print("stClient.available() = ");
    Serial.println(stClient.available());
    Serial.println("\r\nAbout to enter the while read loop\r\n");
    
    while (stClient.available() > 0) {
        
        Serial.println("About to call stClient.read()");
        
        reply = stClient.read();     // *all* read methods cause a client disconnect at line 514 of WiFiClient.cpp
        
        Serial.println("Returned from call to stClient.read()");
        Serial.print("reply char = ");
        Serial.println(reply);
        Serial.print("stClient.connected() = ");
        Serial.println(stClient.connected());
        Serial.print("stClient.available() = ");
        Serial.println(stClient.available());
 
    }
    
    Serial.println("\r\nJust left the while read loop\r\n");
    
    while(true);        // suspend execution
```

### Debug Messages:
Error occurs at timestamp 10:00:21.540:
`[D][WiFiClient.cpp:514] connected(): Disconnected: RES: 0, ERR: 128`

 From ~l/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/newlib/sys/errno.h

ERR:128 is: `ENOTCONN 128		/* Socket is not connected */`
```
10:00:16.511 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
10:00:16.511 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
10:00:18.587 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
10:00:18.587 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
10:00:18.621 -> [D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.2.61, MASK: 255.255.255.0, GW: 192.168.2.1
10:00:20.510 -> 
10:00:20.510 -> About to call stClient.connect
10:00:20.510 -> Return from call to stClient.connect --> SUCCESS
10:00:20.510 -> 
10:00:21.507 -> stClient.available() = 5
10:00:21.507 -> 
10:00:21.507 -> About to enter the while read loop
10:00:21.540 -> 
10:00:21.540 -> About to call stClient.read()
10:00:21.540 -> Returned from call to stClient.read()
10:00:21.540 -> reply char = o
10:00:21.540 -> stClient.connected() = [D][WiFiClient.cpp:514] connected(): Disconnected: RES: 0, ERR: 128
10:00:21.540 -> 
10:00:21.540 -> About to call stClient.read()
10:00:21.540 -> Returned from call to stClient.read()
10:00:21.540 -> reply char = n
10:00:21.540 -> stClient.connected() = 0
10:00:21.540 -> stClient.available() = 3
10:00:21.540 -> About to call stClient.read()
10:00:21.540 -> Returned from call to stClient.read()
10:00:21.574 -> reply char = a
10:00:21.574 -> stClient.connected() = 0
10:00:21.574 -> stClient.available() = 2
10:00:21.574 -> About to call stClient.read()
10:00:21.574 -> Returned from call to stClient.read()
10:00:21.574 -> reply char = i
10:00:21.574 -> stClient.connected() = 0
10:00:21.574 -> stClient.available() = 1
10:00:21.574 -> About to call stClient.read()
10:00:21.574 -> Returned from call to stClient.read()
10:00:21.574 -> reply char = r
10:00:21.574 -> stClient.connected() = 0
10:00:21.574 -> stClient.available() = 0
10:00:21.574 -> 
10:00:21.574 -> Just left the while read loop
```

lbernstone, 2021-05-04:
The connection is disconnected.  This is not necessarily an error, and it is marked as a debug message.  If one side decides the communication is complete, it will close the session.

Xylopyrographer, 2021-05-05:
@lbernstone
Thanks for the reply.

Did some more testing. The crash of the client connection is not caused by a read of the buffer, but by calling `client.connect()` any time after the buffer has been "touched" via _any_ method (any `.read()` method or a `.flush()`).

In practice this means establishing a persistent connection to a server is not possible using a WiFi client.

I'll close this issue and open a new one with revised test code to show the bug.

closed 2021-05-05 by Xylopyrographer