espressif/arduino-esp32 #5155: WiFiClient.connected() Crashes the Client Connection to the Server

Xylopyrographer, 2021-05-06:
Calling `client.connected()` after data has been received by the client and the receive buffer is "touched" crashes the client connection to the server. The crash is in `WiFiClient.cpp`

### Hardware:
Board:M5Stack ATOM Matrix (esp32 PICO-D4)
Core Installation version: 1.0.6
IDE name: Arduino IDE 1.8.13
Flash Frequency: 40MHz
PSRAM enabled: N/A
Upload Speed: 115200
Computer OS: macOS 10.15.7

### Description:

1. Connect a wiFiClient to a server.
2. Send a request to the server.
3. Check that the server returns a response (there are characters in the receive buffer).
4. Call any method that touches the the receive buffer. Any `.read()` method or `.flush()`
5. Call `WiFiClient.connected()`
6. `WiFiClient.cpp` at `line 514` will erroneously disconnect the client from the server throwing:
`ERR: 128 - ENOTCONN 128 /* Socket is not connected */`

With this bug, it is impossible in practice for the client to maintain a persistent connection to the server.

### Sketch:
```cpp
//  Code sample to verify client.connected() bug
   
    char replyC;
    int maxloops = 0;
    tally.tState = "nojoy";
    tally.tConnect = false;
    tally.tTimeout = false;
    tally.tNoReply = true;

    Serial.println("\r\n@@@@@@@@ Start of test. At top of the test client function()\r\n");
    Serial.print("    stClient.connected() = ");
    Serial.println(stClient.connected());
    
    if (!stClient.connected()) {

        Serial.println("\r\nAbout to call stClient.connect()");
        
        if (stClient.connect(stIP, stPort)) {
            
            Serial.println("Return from call to stClient.connect() --> SUCCESS");
            Serial.print("    stClient.connected() = ");
            Serial.println(stClient.connected());
            
            tally.tConnect = true;
        }
        else {
            Serial.println("Return from call to stClient.connect() --> FAIL\r\n");
            return tally;
        }
    }
    
    stClient.print("GET /tally/");      // send the status request to the ST server
    stClient.print(tally.tChannel);
    stClient.print("/status\r\n\r\n");

    while (!stClient.available() && maxloops < 1000) {    // Wait a maximum of 1 second for the ST server's reply to become available
        maxloops++;
        delay(1);                                             // wait a tick before checking again if there is a response from the ST Server.
    }
    if (maxloops == 1000) {         // response from the ST Server timed out
        tally.tTimeout = true;
        return tally;
    }
    
    Serial.println("\r\nResponse received from the server...");
    Serial.print("    stClient.connected() = ");
    Serial.println(stClient.connected());

    int stcBuff = stClient.available();
    
    Serial.print("    stClient.available() = ");
    Serial.println(stcBuff);
    Serial.print("    stClient.connected() = ");
    Serial.println(stClient.connected());

    Serial.println("\r\n======== About to enter the for... read loop.");
    
    for (int i = 1; i <= stcBuff; i++) {
        
        replyC = stClient.read();       // pull a char out  of the rx buffer
        Serial.print(replyC);           // dump it to the Serial port
        
    }
    
    Serial.println("\r\n======== Just left the for... read loop");   
    Serial.print("    stClient.available() = ");
    Serial.println(stClient.available());
    
    Serial.println("\r\nAbout to call stClient.connected()...\r\n");
    
    bool connectState = stClient.connected();   // !!!--> AT THIS POINT *ANY* CALL TO stClient.connected() CRASHES THE CLIENT CONNECTION <--!!!
    
    delay(250);     // without this delay, subsequent calls Serial.println() fail. log_d() has priority on the serial port?
    Serial.println("\r\nBack from call to stClient.connected().");
    Serial.print("    stClient.connected() = ");
    Serial.println(connectState);

    Serial.println("\r\n@@@@@@@@ End of test. Parking the bus...");
    while(true);                        // hold here
```

### Debug Messages:
Here's a sample output from the above code. Bug is seen at timestamp 16:24:18.660
```
16:24:18.627 -> @@@@@@@@ Start of test. At top of the test client function()
16:24:18.627 -> 
16:24:18.627 ->     stClient.connected() = 0
16:24:18.627 -> 
16:24:18.627 -> About to call stClient.connect()
16:24:18.627 -> Return from call to stClient.connect() --> SUCCESS
16:24:18.627 ->     stClient.connected() = 1
16:24:18.627 -> 
16:24:18.627 -> Response received from the server...
16:24:18.660 ->     stClient.connected() = 1
16:24:18.660 ->     stClient.available() = 8
16:24:18.660 ->     stClient.connected() = 1
16:24:18.660 -> 
16:24:18.660 -> ======== About to enter the for... read loop.
16:24:18.660 -> selected
16:24:18.660 -> ======== Just left the for... read loop
16:24:18.660 ->     stClient.available() = 0
16:24:18.660 -> 
16:24:18.660 -> About to call stClient.connected()...
16:24:18.660 -> 
16:24:18.660 -> [D][WiFiClient.cpp:514] connected(): Disconnected: RES: 0, ERR: 128
16:24:18.905 -> 
16:24:18.905 -> Back from call to stClient.connected().
16:24:18.905 ->     stClient.connected() = 0
16:24:18.905 -> 
16:24:18.905 -> @@@@@@@@ End of test. Parking the bus...
```

joko2020, 2021-06-22:
I have same problem..

1. Board        :ESP32 Dev Board.
2. IDE            : Arduino  1.8.12
3. Library      : Adafruit MQTT ,Adafruit MQTT client & WiFi.h

Problem a rise when : 
 Put manual ClientID in the connection and the broker applied JWT authentication method (**token base**)
Problem disappear when:
- Unset ClientID manualy and the broker  applied JWT authentication method or
- Set ClientID manualy and the broker use Username & password only authentication method.
googling & searching to get ride from this problem .. still unsolved

PilnyTomas, 2022-01-27:
Hi, could you please provide a complete code/sketch that can be flashed right away?
You can use [pastebin.com](https://pastebin.com/) if the code should be too long for a comment.

PilnyTomas, 2022-02-11:
@Xylopyrographer, @joko2020 please if any of you have a complete code that would help me to replicate your issue, please provide it.

Xylopyrographer, 2022-02-11:
Hi. Let me blow the dust off that one & see if I can make a generalized sketch 😄.

VojtechBartoska, 2022-03-23:
Hi @Xylopyrographer, did you have a chance to take a look on this one? :)

Xylopyrographer, 2022-03-23:
Hi Vojtěch:

Thanks for the nudge and for staying on this. I’ll put something together and post it real soon.

Cheers,
    Xylo
========================================

Xylopyrographer, 2022-03-23:
@VojtechBartoska
Test sketch below. And below that is the python script also needed to talk to the sketch.
``` c++
//  Test sketch to verify the client.connected() bug in WiFiClient.cpp

/*
 * To use:
 * This sketch is designed to run on a "standard" ESP32 DEV board. Tested on one with an ESP32 WROOM with 4MB flash & 520K RAM.
 * You'll need a WiFi connection to the computer and python 2 (yeah, I know...) installed.
 * Copy the python script "stemulator.py" to your computer as well.
 * 
 *  1. Change the networkSSID[] and networkPass[] below to match your WiFi network. Lines 27 & 28.
 *  2. Change the IPAddress at line 30 to match that of your computer.
 *  3. The test is set to run once and then stop. To keep looping the test, comment out the "while (true);"
 *       statement at the bottom of loop() at line 160.
 *     Time between runs is changed by the POLL_INTERVAL at line 33.
 *  4. In your IDE, set the Core debug level to "Debug".
 *  5. Run the "stemulator.py" python script.
 *  6. Connect your board to the computer & open the Serial Monitor.
 *  7. Compile and download this sketch to your board.
 *  
 * Results are sent to the serial monitor.
*/

#include <WiFi.h>               // this is the esp32 version of this library

#define WIFI_CONNECT_TIMEOUT 60000      // # of ms to wait for a WiFi connection before aborting.

char networkSSID[] = "SSID";     // change to match the SSID...
char networkPass[] = "password";   //  and password of the WiFi network.

IPAddress stIP(0, 0, 0, 0);    // change to match the IP of the computer running stemulator.py
uint16_t stPort = 80;               // shouldn't have to change this, but it needs to match the port used by stemulator.py

#define POLL_INTERVAL 5000          // if the test is set to loop, # of ms between sending requests to stemulator.py
unsigned long nextPollTime;

WiFiClient stClient;

void setup() {

    // introduce ourselves
    Serial.begin(115200);
    delay(50);
    Serial.println("\r\nHey there..."); Serial.print("Running sketch: "); Serial.println(__FILE__); Serial.println("");

    WiFi.disconnect(true);
    WiFi.mode(WIFI_STA);        // set WiFi station mode
    WiFi.setSleep(false);

    int wfStatus = WiFi.status();
    unsigned long wfTimeout;

    wfTimeout = millis() + WIFI_CONNECT_TIMEOUT;

    Serial.println( "\r\nAttempting to connect to WiFi...\r\n" );
    
    WiFi.begin(networkSSID, networkPass);

    while ( ( wfStatus != WL_CONNECTED ) && ( wfTimeout >= millis() ) ) {
        delay(250);            // pause as it takes some time for the WiFi routines to do their thing in the background
        wfStatus = WiFi.status();
    }
    
    if (wfStatus == WL_CONNECTED) {
        Serial.println("\r\nConnected to WiFi.");
    }
    else {
        WiFi.disconnect();
        Serial.println("Unable to connect to WiFi. Stopping test.");
[... 109 lines cut ...]
		self.wfile.write('HTTP-1.0 200 Okay\r\n\r\n')
		self.wfile.write(modes(self.path))

def run(server_class=BaseHTTPServer.HTTPServer,
	handler_class=SimpleRequestHandler):
	server_address = ('', 80)
	httpd = server_class(server_address, handler_class)
	httpd.serve_forever()

def modes(argument):
	# returns value based on time of day to simulate tally data
	# /tally/[address]/status
	curr_time = datetime.now()
	seconds = int(curr_time.strftime('%S'))
	tally_state = (seconds)//10	# do some math so the state changes every 10 seconds

	if tally_state == 0 or tally_state == 3 :
		return "unselected"
	elif tally_state == 1 or tally_state == 4:
		return "selected"
	else:
		return "onair"

run ( )
```

PilnyTomas, 2022-03-24:
Hi @Xylopyrographer, just a quick question - did you try it with the latest core version (2.0.2) ? Or is there any reason you need to use 1.0.6?

Xylopyrographer, 2022-03-24:
@PilnyTomas: I haven't, for a few reasons, tried under the 2.x series of the core.

Question on that though. Using v1.8.19 of the Arduino IDE, if I use the embedded Boards Manager to upgrade to 2.x—to revert to core 1.0.6, is it as simple as using the Boards Manager again to downgrade? Any gotcha's along the way?

Seems it wouldn't be worth it anyway until 2.0.3 is released.

PilnyTomas, 2022-03-30:
Hi, there was a recent update of Arduino-ESP32 core to version 2.0.3-RC1 and with this, it seems to me that the issue is gone. Could you please update a try it out?

VojtechBartoska, 2022-04-11:
@Xylopyrographer Thanks for nice test case, for us it works on latest 2.0.3-rc1. As @PilnyTomas, you can give it a try and confirm this issue is solved.

If you are not able to do so, we are going to close this. Thank you once again for your help.

Xylopyrographer, 2022-04-13:
@VojtechBartoska Thanks for the follow up. When you say it works for you, are you seeing the debug log message?

Something on my end is still not right. The production code (from which the test case was derived) connects 100% (after trapping and resetting the client connection) when using core 1.0.6, but fails 100% using core 2.0.3-RC1 (it never connects to the client).

The test case is structured a bit different than the production code so I'll have to dig into that a bit.

TD-er, 2022-04-13:
When it doesn't connect, please have a look at what is set as DNS server.
I've seen with Ethernet that the first DHCP reply does not set the DNS server.
A next call does set the DNS server to a value other than `0.0.0.0`

Xylopyrographer, 2022-04-13:
@TD-er Thanks for the input. In this use, no DNS server is involved. Hard coded IP addresses all round.

PilnyTomas, 2022-04-20:
Hi, the stemulator script wasn't really working for me, so I tried connecting to `const char* stIP = "google.com";` instead and I'm reading a response quite ok:
```
About to call stClient.connect() with IP google.com
Return from call to stClient.connect() --> SUCCESS
    stClient.connected() = 1

Response received from the server...
    stClient.connected() = 1
    stClient.available() = 1730
    stClient.connected() = 1

======== About to dump the response
HTTP/1.0 404 Not Found
```
... and so on...

VojtechBartoska, 2022-05-04:
Hi @Xylopyrographer, up to the last comment. Can this be considered as solved? We cannot prove the issue.

VojtechBartoska, 2022-05-04:
Thanks @Xylopyrographer for feedback and your contribution! Hope all your project will run smoothly in the future. Closing.

closed 2022-05-04 by VojtechBartoska