espressif/arduino-esp32 #5157: WIDTH ERR: ESP32S2 support 13 bit width Voltage= 0 still not resolved

rahmanshaber, 2021-05-06:
I am using latest git master branch, and esp32s2 nodemcu-esp32-s2. 

I am trying to get ADC working on pin 18, but getting this error 
```
E (437048) ADC: adc2_get_raw(526): WIDTH ERR: ESP32S2 support 13 bit width
Voltage= 0
```
Here is my code
```
static float read_voltage(void)
{
  // Get Battery voltage
  value = analogRead(BATTERY_VOLTAGE);
//  voltage = value * 3.3/1023;
  Serial.print("Voltage= ");
  Serial.println(value);
  delay(200);
  return voltage;
}

void setup(void)
{  

  // Serial setup
  Serial.begin(115200);
  Serial.println("start");

  pinMode(BATTERY_VOLTAGE, INPUT_PULLUP);
}

void loop(void)
{
  read_voltage();
}
```

mMerlin, 2021-05-08:
My test board is on usb power, not battery, so I can not test this, but try without `pinMode(BATTERY_VOLTAGE, INPUT_PULLUP);` ADC input does not use or want a pinMode setting. That is for digital I/O. Not analog or PWM.

You can also try explicitly setting the bit width using `analogReadResolution(12)`. Once you get reasonable values, that divide by 1023 likely needs to change. At 12 bit resolution, the maximum ADC result is 4095. Also, if memory is right, that connection to the battery is through a divide by 2 voltage divider, so the expected value is half the battery voltage. Combined, that probably means divided by 2047.

If that does not get better results, you may need to use the *lower* level API. See [ADC API reference](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#analog-to-digital-converter) documentation

rahmanshaber, 2021-05-09:
Tested with pin 2 and i did not got the error, with the your fix. But still i am getting the width error with the pin 18. 
I think it's the ADC_channel 2. 
```

// For Battery Voltage
uint32_t  value = 0;
float voltage;
#define BATTERY_VOLTAGE 2
static float read_voltage(void)
{
  // Get Battery voltage
  value = analogRead(BATTERY_VOLTAGE);
//  voltage = value * 3.3/1023;
  Serial.print("Voltage= ");
  Serial.println(value);
  delay(200);
  return voltage;
}

void setup(void)
{  

  // Serial setup
  Serial.begin(115200);
  Serial.println("start");
  analogReadResolution(13);

//  pinMode(BATTERY_VOLTAGE, INPUT_PULLUP);
}

void loop(void)
{
  read_voltage();
}
```

rahmanshaber, 2021-05-09:
@mMerlin I am testing the ADC using the 3v power pin in the broad, You can also test that way.

mMerlin, 2021-05-09:
I am using an Adafruit HUZZAH32 Feather board (ESP32-WROOM-32). It has the internal battery measurement on A13 (gpio 35), which seems to be on ADC1. Testing using A5, which is on ADC2 works for me as well. Is your posted sample code complete? I see that in the original post, value and voltage were not declared. If that was an extract, there is a documented limitation that you can not use ADC2 if WiFi is enabled.  On the HUZZAH32, gpio 18 is a digital only pin. Using that actually works, just returning '0' values.

My code that works for both A13 and A5 on the HUZZAH32, using the Arduino IDE.

```cpp
//#define BATTERY_VOLTAGE A13
#define BATTERY_VOLTAGE A5
static float read_voltage(void)
{
  uint32_t value = analogRead(BATTERY_VOLTAGE);
  Serial.printf("reading= %d\n", value);
  delay(200);
  return 0;
}

void setup(void)
{
  Serial.begin(115200);
  Serial.println("start");
}

void loop(void)
{
  read_voltage();
}
```

rahmanshaber, 2021-05-10:
This issues in related to S2 not S1. you are using a ESP32-S1, so that's why it's working fine.
Also i am not using Wifi at all. As i read in the docs if you don't call the wifi then it's not turned on. 
Yes that original post is missing those variables, i forgot to add them here. This is a small part of the code, here is the full code https://gitlab.com/mutantC/mutantc-v4/firmware/firmware.ino

closed 2021-07-24 by stale[bot]