espressif/arduino-esp32 #5159: Writing to EEPROM on esp32s2

xtrinch, 2021-05-06:
I've updated an "old" project that was using the `https://github.com/espressif/arduino-esp32.git#idf-release/v4.2` to the new alpha version (using the latest master). The project does some EEPROM writes, which are now automagically not persisted between reboots. `EEPROM.commit()` returns true and everything looks fine. If I read them back during the same boot cycle, the values are read back fine. Then I restart and all is lost (But yet it's always the same garbage values being read back).

Here's a link to my project (which was working just fine under the idf-release/v4.2 branch which is now gone): https://github.com/xtrinch/esp32s2-bme280-webusb

Has something changed with the EEPROM handling between the branch I was using and the latest alpha version? Couldn't find any info.

lbernstone, 2021-05-07:
There's only been one minor change in readString in the last 2 years, and I can't see how that would present with these symptoms.  There's a lot of logging in EEPROM, try setting core debug level to verbose and see if it throws any messages that might indicate an error.

xtrinch, 2021-05-07:
I've given arduino preferences a try, to see if I have any success there. 
`preferences.putString("ssid", ssid)` returns for example 10 (indicating that it saved successfully),
then I run `preferences.getString("ssid", "Unknown")` right after it, and it returns Unknown.

lbernstone, 2021-05-07:
Do the examples work for you?
https://github.com/espressif/arduino-esp32/blob/master/libraries/EEPROM/examples/eeprom_class/eeprom_class.ino

xtrinch, 2021-05-07:
Yes. But I've modified the sketch to demonstrate that the values do not persist across reboots:
```
  double height = 0;
  uint32_t age = 0;

  // Read: Variables <--- EEPROM stores
  HEIGHT.get(0, height);
  AGE.get(0, age);
  USBSerial.print("height: "); USBSerial.println(height);
  USBSerial.print("age: ");    USBSerial.println(age);
  USBSerial.println("------------------------------------\n");

  const char* name = "Teo Swee Ann";
  char rname[32];

  if (age != 47) {
    height = 5.8;
    age = 47;

    // Write: Variables ---> EEPROM stores
    NAMES.writeString(0, name);
    HEIGHT.put(0, height);
    AGE.put(0, age);
    USBSerial.println("Values are garbage, writing them to EEPROM:\n");
    USBSerial.print("name: ");   USBSerial.println(name);
    USBSerial.print("height: "); USBSerial.println(height);
    USBSerial.print("age: ");    USBSerial.println(age);
    USBSerial.println("------------------------------------\n");
  }

  // Clear variables
  rname[0] = '\0';
  height = 0;
  age = 0;
  USBSerial.print("height: "); USBSerial.println(height);
  USBSerial.print("age: ");    USBSerial.println(age);
  USBSerial.println("------------------------------------\n");

  // Read: Variables <--- EEPROM stores
  NAMES.get(0, rname);
  HEIGHT.get(0, height);
  AGE.get(0, age);
  USBSerial.println("Newly saved values:\n");
  USBSerial.print("name: ");   USBSerial.println(rname);
  USBSerial.print("height: "); USBSerial.println(height);
  USBSerial.print("age: ");    USBSerial.println(age);
  

  USBSerial.println("Done!");
  ```
  
  The output is always the following, no matter how many times I reboot:
  ```
  Testing EEPROMClass

height: nan
age: 4294967295
------------------------------------

Values are garbage, writing them to EEPROM:

name: Teo Swee Ann
height: 5.80
age: 47
------------------------------------

height: 0.00
age: 0
------------------------------------

Newly saved values:

name: Teo Swee Ann
height: 5.80
age: 47
Done!
  ```

xtrinch, 2021-05-07:
Erased the flash many times previously, and did not forget the EEPROM.commit() in the above example.

Interestingly enough, your last example works for me - when writing directly to address.
The following, however, does not persist between reboots (Tried on two different ESP32S2 devices, just to be sure it's not one faulty board):

```
EEPROMClass  AGE("eeprom2", 0x100);

void setup() {
  USBSerial.begin();
  AGE.begin(AGE.length());

  uint32_t age = 0;
  AGE.get(0, age);
  USBSerial.printf("At boot: %d\n", age);
  age = 47;
  AGE.put(0, age);
  AGE.get(0, age);
  USBSerial.printf("After write: %d\n", age);

  EEPROM.commit(); 
  USBSerial.println("Done!");
}
```

Output is always:
```
At boot: -1
After write: 47
Done!
```

And every once in a while I get this weird crash:

```
␀␀␀␀name: ␀␀Newly saved values:
␀␀␀␀Done!␀␀␀eeprom0␀eeprom1␀eeprom2␀␀␀␀␀␀␀␀␀l�  @�@P@�@(�       @�@ @�@x@P@0�   @x@d�   @Espressif␀␀␀ESP32S2 arduino device␀␀1234-5678␀␀␀CDC class␀␀␀DFU class␀␀␀HID class␀␀␀MIDI class␀␀MSC class␀␀␀Vendor class (webUSB)␀␀␀espUSB␀␀�@�@�@�@�@�@␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀x␎    @��     @(�     @x␎     @x␎     @x␎     @x␎     @P@0�   @x@x␎   @eeprom␀␀/home/xtrinch/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/EEPROM/src/EEPROM.cpp␀␀␀␀[%6u][E][%s:%u] %s(): Unable to open NVS namespace: %d
␀␀␀␀[%6u][E][%s:%u] %s(): Unable to read NVS key: %d
␀␀[%6u][E][%s:%u] %s(): Not enough memory to truncate EEPROM!
␀␀␀[%6u][E][%s:%u] %s(): Not enough memory to expand EEPROM!
␀expand␀␀[%6u][E][%s:%u] %s(): Not enough space to expand EEPROM from %d to %d
␀[%6u][E][%s:%u] %s(): Not enough memory for %d bytes in EEPROM
␀␀␀␀[%6u][E][%s:%u] %s(): error in write
␀␀commit␀␀begin␀␀␀␀␀␀␀␀␀␀␀��    @�@t@nan␀inf␀ovf␀/home/xtrinch/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/esp32-hal-misc.c␀␀␀[%6u][E][%s:%u] %s(): Failed to add loop task to WDT
␀␀*␀␀␀[%6u][E][%s:%u] %s(): Failed to format the broken NVS partition!
␀␀[%6u][E][%s:%u] %s(): Failed to initialize NVS! Error: %u
␀initArduino␀%s␀␀loopTask␀␀␀␀/home/xtrinch/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/esp32-hal-cpu.c␀␀␀␀[%6u][E][%s:%u] %s(): Callback Object Malloc Failed
␀␀␀[%6u][E][%s:%u] %s(): duplicate func=%08X arg=%08X
␀␀␀␀[%6u][E][%s:%u] %s(): Bad frequency: %u MHz! Options are: 240, 160, 80, %u, %u and %u MHz
␀[%6u][E][%s:%u] %s(): Bad frequency: %u MHz! Options are: 240, 160, 80, %u and %u MHz
␀[%6u][E][%s:%u] %s(): CPU clock could not be set to %u MHz
␀␀␀␀setCpuFrequencyMhz␀␀ota_app_count < 16 && "must erase the partition before writing to it"␀␀␀../esp-idf/components/app_update/esp_ota_ops.c␀␀phys_offs != SPI_FLASH_CACHE2PHYS_FAIL␀␀it != NULL␀␀esp_ota_get_running_partition␀␀␀partition␀␀␀E (%u) %s: load_partitions returned 0x%x
␀␀␀it␀␀../esp-idf/components/spi_flash/partition.c␀iterator != NULL␀␀␀␀partition != NULL␀␀␀esp_partition_mmap␀␀esp_partition_erase_range␀␀␀esp_partition_write_raw␀esp_partition_read_raw␀␀esp_partition_write␀esp_partition_read␀␀esp_partition_get␀␀␀esp_partition_next␀␀s_mmap_page_refcnt[i] == 0 || (entry_pro == SOC_MMU_PAGE_IN_FLASH(pages[pageno]) )␀␀../esp-idf/components/spi_flash/flash_mmap.c␀␀␀␀s_mmap_page_refcnt[i] > 0␀␀␀0 && "invalid handle, or handle already unmapped"␀␀␀spi_flash_munmap␀␀␀␀spi_flash_mmap_pages␀␀␀␀(dest_addr % 16) == 0␀␀␀../esp-idf/components/spi_flash/esp32s2/flash_ops_esp32s2.c␀(size % 16) == 0␀␀␀␀spi_flash_write_encrypted_chip␀␀E (%u) %s: unexpected spi flash error code: 0x%x
␀␀␀E (%u) %s: failed to get chip size
␀spi_flash␀␀␀E (%u) %s: Detected size(%dk) smaller than the size in the binary image header(%dk). Probe failed.
␀spi_flash␀␀␀chip_status == 0␀␀␀␀../esp-idf/components/spi_flash/spi_flash_os_func_app.c␀spi1_flash_os_check_yield␀␀␀nvs␀chunkStart == item.blobIndex.chunkStart␀../esp-idf/components/nvs_flash/src/nvs_storage.cpp␀itemTypeOf(dst) == datatype␀../esp-idf/components/nvs_flash/src/nvs_types.hpp␀␀␀err != ESP_ERR_NVS_PAGE_FULL␀␀␀␀dataSize == item.blobIndex.dataSize␀static_cast<uint8_t> (chunkStart) + chunkNum == item.chunkIndex␀offset == dataSize␀␀findItem(nsIndex, datatype, key, findPage, item)␀␀␀␀prevStart == VerOffset::VER_0_OFFSET || prevStart == VerOffset::VER_1_OFFSET␀␀␀␀esp_err_t nvs::Storage::eraseMultiPageBlob(uint8_t, const char*, nvs::VerOffset)␀␀␀␀esp_err_t nvs::Storage::cmpMultiPageBlob(uint8_t, const char*, const void*, size_t)␀esp_err_t nvs::Storage::readMultiPageBlob(uint8_t, const char*, void*, size_t)␀␀esp_err_t nvs::Storage::writeItem(uint8_t, nvs::ItemType, const char*, const void*, size_t)␀esp_err_t nvs::Storage::writeMultiPageBlob(uint8_t, const char*, const void*, size_t, nvs::VerOffset)␀␀␀void nvs::Item::getValue(T&) [with T = unsigned char]␀␀␀␀␀␀␀␀␀␀␀�@�@�@␐@`@�@�@D@�@�@␜@�@<@␀␀␀␀␀␀␀␀P� @�@X�   @l@�@�@�@�@d�   @p�     @␀␀␀␀␀␀␀␀|�     @false && "item should have been present in cache"␀␀␀../esp-idf/components/nvs_flash/src/nvs_item_hash_list.cpp␀␀void nvs::HashList::erase(size_t, bool)␀index == mFirstUsedEntry␀␀␀␀../esp-idf/components/nvs_flash/src/nvs_page.cpp␀␀␀␀mState == PageState::UNINITIALIZED␀␀index < ENTRY_COUNT␀entry < ENTRY_COUNT␀../esp-idf/components/nvs_flash/src/nvs_page.hpp␀␀␀␀end <= ENTRY_COUNT␀␀end > begin␀index < Nitems␀␀../esp-idf/components/nvs_flash/src/compressed_enum_table.hpp␀␀␀size % ENTRY_SIZE == 0␀␀mNextFreeEntry != INVALID_ENTRY␀mFirstUsedEntry != INVALID_ENTRY␀␀␀␀state == EntryState::WRITTEN || state == Ent [... cut ...] ��␀�␄-␑*���␃␝�␀6a␀!R���␌�"␀␌␂)1%~␃�1f␘␄�␂%
␅��␂e�␄��␃�␅�␃␝�6A␀�␂��␄␝�␀␀6A␀!"�␌␙(␂ ��␝�␀6�␀��09␑��=␁␌HI!Y1iAyQ9����r␌�x␑��f␗␂�~␁␝�␀␀␀A␀␂�␃␀�␓Ea␀␀0␀␀E␒�␀␅
␁␒��␒�p␂A␀�␃␀�␃␐!␀�␃    ␑␀�␃���a1␁��␂a␀␐␁ ␀�␓�[␁␁��␀�␓␐ ␀ �␃0�␃A��0"␐@"␐␖�␄␀�␃800�I8@0�I␀�␃A>�@␀ ␀␄�␋B@"␐Vr�␛DL␃@�␓G␓␙␜�@�@@3�At�@3�H␃-␆h␓�␄␀␆��F��␀␀�Y!␀�␑␀�1␘A␐ ␀␐3␀6A␀An�@� eo␃�␂̸�␃e�␁�␚��␅�b␀@� �o␃�␀␀␀␀6a␀�␂�e␇␃f␚␄␌␂�!␀ � @� ���""␀�"�^��^��^�������␀-
␐␑ e␄␃&␚π�␃��4��fD␅␐␑ ��␅B�␀ � ��␌�␄I1�'␂-
G�␂G��2!␃␖␃␂��␀����␅␀0�  � fD   e!␂�* F␁␀␀e␅␂-
```

lbernstone, 2021-05-07:
EEPROM is the default instance.  `AGE.commit();`

xtrinch, 2021-05-07:
Thanks, my bad!

In between doing all this testing trying to figure out what's wrong, I must have deleted the arduino-esp32 package, cleaned everything, rebuilt and reinstalled everything and somehow even my original project (the original version with EEPROM and the preferences version) works now. I guess something corrupted got stuck in some cache somewhere.

Sorry for wasting your time and thanks for your help.

closed 2021-05-07 by xtrinch

brianmichalk1, 2022-02-22:
I have this same exact problem using the same example program.  Here's my platformio.ini, everything else I copied from this thread.
```
[env:esp32-s2-saola-1]
platform = espressif32
board = esp32-s2-saola-1
framework = arduino
lib_deps = EEPROM
build_flags = 
  -DCORE_DEBUG_LEVEL=5
  -DLOG_LOCAL_LEVEL=ESP_LOG_DEBUG
```

lbernstone, 2022-02-23:
EEPROM is included as part of the arduino-esp32 core, so you shouldn't need it in lib_deps, and this is likely to pull in a library for avr.  Note that [eeprom](https://github.com/espressif/arduino-esp32/tree/master/libraries/EEPROM) is only provided for code portability.  If you are writing code specifically for esp32, you should use [Preferences](https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/examples/StartCounter/StartCounter.ino).  OP's issue was solved, and appears to have been an installation issue.