espressif/arduino-esp32 #5164: Windows 10-esp32 bluetooth connection

DavidePasero, 2021-05-08:
### Hardware:
Board:							DOIT esp32 DEVKIT V1
Core Installation version:			don't know
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
Hello everyone, I'm trying to connect via bluetooth my DOIT esp32 DEVKIT V1 with my windows 10 pc. I have a desktop pc, and in order to implement bluetooth I have a "Hommie bluetooth 5.0 adapter"; the problem is that my pc sees Esp32test, but when I try to pair them (after pressing the EN button on the esp32 board) a weird error message appears: 
ASSERT_WARN(103 23), in lc_task.c at line 8775ASSERT_WARN(103 24), in lc_task.c at line 8775.
The sketch I'm using is the SerialToSerialBT example, what do I do?

### Sketch:
#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```cpp

### Debug Messages:
No debug messages

lbernstone, 2021-05-08:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

DavidePasero, 2021-05-08:
> https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

What does it mean?

DavidePasero, 2021-05-11:
To everyone who will have the same problem: use esp32 core installation 1.0.4 instead of 1.0.6 (which, right now in 2021 is the latest). To do that you go on Arduino IDE - Board manager - uninstall esp32 libraries - reinstall esp32 library but use version 1.0.4

closed 2021-05-11 by DavidePasero

Lukaswnd, 2022-04-02:
I got the same Problem. Any Windows 10 system (i tried 4 different) have huge issues to connect to a esp32.
I also tried to use all of my esp32's including esp32s, esp32-woover-e, esp32-wroom,  esp32-wroom-e,  esp32-wroom-u.

There was one chip among 10, that could make a connection to win10, but it was by far not the first tries.
10 Minutes ago i found this issue and BAM with version 1.0.4 it works perfectly fine

LIUHanyang96, 2022-08-31:
> I got the same Problem. Any Windows 10 system (i tried 4 different) have huge issues to connect to a esp32. I also tried to use all of my esp32's including esp32s, esp32-woover-e, esp32-wroom, [...]

The same issue. I cannot pair Win 11 or Android with ESP32-WROOM, but it can pair with Macbook.

erikaemma, 2022-10-06:
But I am using 1.0.6, Win10 and Android works fine, but Bluetooth LE Explorer can't scan it.