espressif/arduino-esp32 #5165: Memory consumption of bluetooth is enormous

HelloDB, 2021-05-09:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|node32|ttgo_lora|Custom w/ ESP32-S2-WROVER 16MB|
|<B>Version/Date</B>|1.0.4|
|<B>IDE name</B>|VSCode + Platform.io|
|<B>Flash Frequency</B>|40Mhz
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
BT occupies about 200kbRAM, I only use BLE（notify）, how to reduce BLE memory consumption ?
[reference：https://esp32.com/viewtopic.php?t=3139](https://esp32.com/viewtopic.php?t=3139)
What should I do in arduino environment ?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
// BLE_notify.ino

#include <Arduino.h>

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer *pServer = NULL;
BLECharacteristic *pCharacteristic = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks : public BLEServerCallbacks
{
    void onConnect(BLEServer *pServer)
    {
        deviceConnected = true;
    };

    void onDisconnect(BLEServer *pServer)
    {
        deviceConnected = false;
    }
};

void setup()
{
    Serial.begin(115200);

    // esp_bt_controller_enable(ESP_BT_MODE_BLE);
    // esp_bt_controller_mem_release(ESP_BT_MODE_BLE);
    // esp_bt_mem_release(ESP_BT_MODE_BTDM);
    // esp_bt_mem_release(ESP_BT_MODE_BLE);

    // Create the BLE Device
    BLEDevice::init("ESP32");

    // Create the BLE Server
    pServer = BLEDevice::createServer();
    pServer->setCallbacks(new MyServerCallbacks());

    // Create the BLE Service
    BLEService *pService = pServer->createService(SERVICE_UUID);

    // Create a BLE Characteristic
    pCharacteristic = pService->createCharacteristic(
        CHARACTERISTIC_UUID,
        BLECharacteristic::PROPERTY_READ |
            BLECharacteristic::PROPERTY_WRITE |
            BLECharacteristic::PROPERTY_NOTIFY |
            BLECharacteristic::PROPERTY_INDICATE);

    // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
    // Create a BLE Descriptor
    pCharacteristic->addDescriptor(new BLE2902());

    // Start the service
    pService->start();

    // Start advertising
    BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
    pAdvertising->addServiceUUID(SERVICE_UUID);
    pAdvertising->setScanResponse(false);
    pAdvertising->setMinPreferred(0x0); // set value to 0x00 to not advertise this parameter
    BLEDevice::startAdvertising();
    Serial.println("Waiting a client connection to notify...");
}
uint32_t last = 0;
void loop()
{
    // notify changed value
    if (deviceConnected)
    {
        pCharacteristic->setValue((uint8_t *)&value, 4);
        pCharacteristic->notify();
        value++;
        delay(3); // bluetooth stack will go into congestion, if too many packets are sent, in 6 hours test i was able to go as low as 3ms
    }
    // disconnecting
    if (!deviceConnected && oldDeviceConnected)
    {
        delay(500);                  // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected)
    {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
    if (millis() - last > 1000)
    {
        Serial.printf("heap:%d, unsed:%d\n", ESP.getFreeHeap(), ESP.getMinFreeHeap());
        last = millis();
        // esp_bluedroid_disable();
        // esp_bluedroid_deinit();
        // esp_bt_controller_disable();
        // esp_bt_controller_deinit();
        // esp_bt_mem_release(ESP_BT_MODE_BTDM);
        // esp_bt_mem_release(ESP_BT_MODE_BLE);
    }
}
```

### Debug Messages:
```
[05:33:30.339]收←◆heap:180928, unsed:176056
[05:33:31.340]收←◆heap:180928, unsed:176056
[05:33:32.342]收←◆heap:180928, unsed:176056
[05:33:33.344]收←◆heap:180928, unsed:176056
[05:33:34.345]收←◆heap:180928, unsed:176056
...
```

chegewara, 2021-05-09:
>  how to reduce BLE memory consumption ?

Use nimble library.

HelloDB, 2021-05-10:
> > how to reduce BLE memory consumption ?
> 
> Use nimble library.
Thanks a million.