espressif/arduino-esp32 #5166: Where did my RAM go?

thezenox, 2021-05-10:
### Hardware:
|||
|:---|---|
|<B>Board</B>|ESP32 Dev Module|
|<B>Version/Date</B>|2.0.0|today's master|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|80Mhz|
|<B>PSRAM enabled</B>|yes|

### Description:
Using NimBLE + WiFi TLS and some bigger custom buffers with a recent update (not sure yet, which version exactly made the change) I don't have enouth heap for WiFiSecure TLS (min ~65kB) left without changing my code. With the prior Version of arduino-esp32 there was no problem.

### Questions
Within the commit history, I didn't find any big changes leading to more RAM / heap usage.
Does anybody know what parts recently increased memory usage or where it is possible to free up some RAM?
Are there any new possibilities to utilize the SPIRAM for BLE, WiFi, ...?

lbernstone, 2021-05-10:
The OS has gotten larger as it supports more features, hardware and software.
In 2.0, memory allocations larger than 16k (4k for S2) are automatically mapped into psram.  Features that will work in psram are mapped into psram. However, as arduino-esp32 is compiled to support a wide set of hardware configurations, it is not optimized for memory space.  With WiFi and NimBLEDevice initialized, I have 177K of memory free (113k alloc) on an esp32-wrover.

h2zero, 2021-05-12:
@thezenox If you're using NimBLE as stated and have PSRAM, you can move the NimBLE core to PSRAM by setting
`CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL 1` in the nimconfig.h file, or using the option in menucofig if using IDF. 
That should reduce the RAM by about 10k from the internal heap, YMMV.

VojtechBartoska, 2021-05-18:
@thezenox Can I consider your question covered and solved?

thezenox, 2021-05-18:
kind of, yes. Setting `CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL ` saved me enougth for getting my application working again. Thanks

closed 2021-05-18 by thezenox

SaitamaTechno, 2023-04-09:
Hey guys, I am using esp32 cam module with micropython firmware. When I capture photos as many as 20 or 30 kb of size, it runs out of ram and the board resets itself. People say it has 500kb of sram but I cannot use it. The similar issue happened when I used esp8266. This time, when I create lists as large as 30kb, it runs out of ram and resets itself.

me-no-dev, 2023-04-09:
@SaitamaTechno this is the Arduino issue tracker, not the micropython one :) Here camera works fine with much larger images.