espressif/arduino-esp32 #5167: CORRUPT HEAP: Bad head at 0x3ffb4a58. Expected 0xabba1234 got 0x3ffb4bc8

glowingkitty, 2021-05-10:
Why is my ESP32 D1 Mini crashing, when I use the leds.fill() and leds.show() function from Adafruit_NeoPixel, after I updated my stripe_config.json with the details about the new LED animation? Friends of mine suggested it sounds like some memory issue / overwriting memory issue. Any ideas where could be the problem? The arduino-esp32 framework? Or my code while using Adafruit_Neopixel? or while using ArduinoJSON (which I use to handle the json)?

### My setup:
- two ESP32 D1 Mini (one is the webserver to provide a web interface for changing LED animations, the other calculates the LED states and sends them via Adafruit_NeoPixel to the WS2812B LED strips), connected via Serial
- Arduino ESP32 framework, v 3.2.0
- coding in VScode with Platform.io (Core 5.1.1, Home 3.3.4)
- the current LED animation is saved in a "stripe_config.json" on both ESP32. If the animation is changed via the web interface, the webserver ESP sends via serial the new stripe_config.json to the LED ESP32. That part works fine.

### Crash message (without hardware debugger):
`
CORRUPT HEAP: Bad head at 0x3ffb4a58. Expected 0xabba1234 got 0x3ffb4bc8
abort() was called at PC 0x40086f11 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088d08:0x3ffc8d10 0x40088f85:0x3ffc8d30 0x40086f11:0x3ffc8d50 0x4008703d:0x3ffc8d80 0x401042f7:0x3ffc8da0 0x401005c9:0x3ffc9060 0x40100561:0x3ffc90b0 0x4008e115:0x3ffc90e0 0x40081d8e:0x3ffc9100 0x40086e09:0x3ffc9120 0x4000bec7:0x3ffc9140 0x400e29fc:0x3ffc9160 0x400d7077:0x3ffc9180 0x400d0e6b:0x3ffc9aa0 0x4008aa4a:0x3ffc9ac0

Rebooting...
`
- at the same time the LEDs are flashing in random colors and brightness (which isn't supposed to happen... all LEDs are supposed to glow in the same color in that example):
[image]

With connected hardware debugger:
**Debugger output when crashing:**
`
Program received signal 
SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1073432812]
invoke_abort () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:154
154	/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c: No such file or directory.
`

This is where the code crashes:
https://github.com/glowingkitty/The-Glowing-Stripes-Project/blob/master/esp32_leds/src/neopixel.cpp#L212

But my guess is that I might handle updating the JSON file wrong?
https://github.com/glowingkitty/The-Glowing-Stripes-Project/blob/master/esp32_leds/src/stripe_config.cpp#L29

Or the way how I do the multi-core-multitasking on the ESP32?
https://github.com/glowingkitty/The-Glowing-Stripes-Project/blob/master/esp32_leds/src/main.cpp

**Debugger automatically opens a "esp_crosscore_int_send_yield.dbgasm" file when the crash happens:**
`0x400855f8: 36 41 00        	entry	a1, 32
0x400855fb: b2 a0 01        	movi	a11, 1
0x400855fe: 20 a2 20        	or	a10, a2, a2
0x40085601: 65 f4 ff        	call8	0x40085548 <esp_crosscore_int_send>
0x40085604: 90 00 00        	retw
`
[image]

### Hardware:
Board:							WeMos D1 MINI ESP32
Core Installation version:			Espressif 32 (3.2.0) 
IDE name:						Platform.io (in VScode)
Flash Frequency:					240MHz
PSRAM enabled:                                    unknown
Upload Speed:					115200
Computer OS:                                       MacOS 11.3 (M1 chip)

lbernstone, 2021-05-10:
I did not look at your code, but this typically results from assigning a variable without allocating memory properly.  If you define a complex variable (struct, array, etc), then you must malloc the memory you will use, and then free() it when you are done.

glowingkitty, 2021-05-10:
solution: seems my stop_animation function caused that issue... but I still don't fully understand why...
https://github.com/glowingkitty/The-Glowing-Stripes-Project/commit/1e7b3e58b3343a17d2321cdeb5db9ed383c98278#diff-d49ce86d20555a4abf26a8eefda8db41fe35175d437e2345287ec4fbbef68f48