espressif/arduino-esp32 #5176: Compile this with ESPIDF v4.2.1 instead of v4.4

calkoe, 2021-05-13:
Hello,

is there a way to compile this Framework with the latest stable ESPIDF 4.2.1 ?
Unfortunately some Espressif Libraries have still some Bugs in the current used ESPIDF 4.4 like the MQTT-Library.

I currently not able use the Arduino Framework, ESPIDF 4.x features and Espressifs MQTT-Library together.

Thanks for the Support!

VojtechBartoska, 2021-05-14:
Hi @calkoe,

@david-cermak will take a look on this. He is develpor of MQTT.

calkoe, 2021-05-14:
> Hi @calkoe,
> 
> @david-cermak will take a look on this. He is develpor of MQTT.

**Some Additional Error Details:**

This is the Error Message i receive:
```
E (3191) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (3192) esp-tls-mbedtls: Failed to set client configurations
E (3197) esp-tls: create_ssl_handle failed
E (3200) esp-tls: Failed to open new connection
E (3205) TRANSPORT_BASE: Failed to open a new connection
E (3212) MQTT_CLIENT: Error transport connect
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

As mentioned in the vanilla 4.2.1 IDF my code works perfectly fine.

This is how i use the MQTT Library:
  ```cpp
config.user_context = this;  
config.uri          = uri.c_str();
config.client_id    = "MyClientID";
client = esp_mqtt_client_init(&config);
esp_mqtt_client_register_event(client, (esp_mqtt_event_id_t)ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
esp_mqtt_client_start(client);
```

david-cermak, 2021-05-14:
@calkoe You have to supply server's certificate (or a cert bundle, ca-store or some other means of verification).

One of the changes in [idf `v4.3`](https://github.com/espressif/esp-idf/releases/tag/v4.3-beta3):
```
esp-tls: esp-tls client connection will now return error by default in the absence of server verification option
```
in the long running effort of discouraging insecure connections or defaulting to insecure if config option missed.

An alternative (for IDF users) is to explicitly configure:
```
CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y
```

calkoe, 2021-05-14:
Thanks for the answer.

Ok thanks bad news for me. I want to use this in my local Network with self-signed Certificates that might be change.
In that case i have to update it on all my ESP32 nodes..

I know the CONFIG_ESP_TLS_INSECURE and CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY but unfortunately they have no effect in the precompiled arduino-esp32-framework.

**It would be awesome to set the skip_server_verify in the esp_mqtt_client_config_t**

calkoe, 2021-05-18:
@david-cermak 
Is there a way to prevent the ESP from crashing if there is something wrong with the Certificate and Provide a error message ?

david-cermak, 2021-05-20:
@calkoe The ESP-IDF should not crash on such errors. If you see this could you please report the use-case with the exact steps to reproduce?

The expected behaviour is raising a `MQTT_EVENT_ERROR` event with appropriate error code(s from TLS/TCP connection lower layers). Here's what I get on the latest `v4.4-dev` with an expired  certificates:

```
I (3636) MQTTS_EXAMPLE: [APP] Free memory: 229868 bytes
I (3646) MQTTS_EXAMPLE: Other event id:7
E (4606) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (4606) esp-tls-mbedtls: Failed to verify peer certificate!
I (4616) esp-tls-mbedtls: verification info:   ! The certificate is not correctly signed by the trusted CA

E (4626) esp-tls: Failed to open new connection
E (4626) TRANSPORT_BASE: Failed to open a new connection
E (4636) MQTT_CLIENT: Error transport connect
I (4636) MQTTS_EXAMPLE: MQTT_EVENT_ERROR
I (4646) MQTTS_EXAMPLE: Last error code reported from esp-tls: 0x8010
I (4646) MQTTS_EXAMPLE: Last tls stack error number: 0x2700
I (4656) MQTTS_EXAMPLE: Last captured errno : 0 (Success)
I (4666) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
```

Please note that, it's possible to check the above errors programmatially and have your app take appropriate reaction.

---
About the alternatives:

>  Certificates that might be change...  expires in some years),

If you're planning on using self signed certificates, would the `CA` certificate change, too? I think you can be perfectly okay with providing a CA-cert here.

There're more alternatives, like PSK method, CA-bundle or CA-store, which could be used also in the local network.

---
About the ` skip_server_verify` in the `esp_mqtt_client_config_t`

I think this is a valid feature request, although I believe that we'd still like to recommend having at least the `CONFIG_ESP_TLS_INSECURE=y` pre-configured, but that could be perhaps arranged by building arduino with this flag enabled.

calkoe, 2021-05-20:
I'm not able to reproduce the Crash with the latest version or #2.0.0-alpha1.
I only receive the expected error messages! :)

I've managed to update the CA Certificate via a MQTT Message and store it in the NVS, i think this is the way to go.

`CONFIG_ESP_TLS_INSECURE=y` pre-configured would be a good solution too.

closed 2021-05-20 by calkoe

zonmen, 2021-07-11:
> @calkoe You have to supply server's certificate (or a cert bundle, ca-store or some other means of verification).
>
> One of the changes in [idf [...]
And (for IDF users)you need to make changes in file "sdkconfig" in root directory of your project
it was really useful information for me because I spent an hour to find this file*