espressif/arduino-esp32 #5179: Power consumption

hzhh110, 2021-05-14:
1. When WiFi.mode (WIFI_MODE_STA or WIFI_MODE_APSTA) is set, and WiFi is not set, or when the WiFi account or password is wrong, the power consumption is 150ma,

2. When setting WiFi.mode (WIFI_MODE_STA or WIFI_MODE_APSTA), when connected to WiFi, power consumption is 100ma

How to reduce the power consumption of question 1?

hzhh110, 2021-05-18:
Hi

hzhh110, 2021-05-18:
void setup() {   WiFi.mode(WIFI_STA);}
[image]

The whole program is just one sentence, but the power consumption has soared to 148ma!How to lower it!When connected to the network, it drops down again

lbernstone, 2021-05-18:
This forum is for issues with the codebase, not general complaints.  WiFi transmit requires more power than receive, and initial connection has a lot of transmitting.  Google knows a great deal about reducing power consumption on the esp32.

closed 2021-08-04 by stale[bot]

hzhh110, 2021-10-08:
好,esp32和esp8266 有linux kernel 内核的驱动吗?想用海思的摄像头芯片,直接用ESP32作为WiFi设备

hzhh110, 2021-10-08:
哪个芯片既能做其他linux嵌入式 如HI3516EV200的wifi外设芯片,又能用arduino实现自己功能的芯片呢?这样在用ESP32实现自己功能的时候,又能提供摄像头的wifi功能