espressif/arduino-esp32 #5181: I'm getting build errors with Arduino IDE 1.8.14.

tanakamasayuki, 2021-05-14:
### Description:
A build-in library option has been added.
As a result, the library for AVR is loaded.

-built-in-libraries C:\Program Files (x86)\Arduino\libraries

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include "SD.h"
void setup() {
}
void loop() {
}
```

### Debug Messages:
```
In file included from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/SdFat.h:29,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:20,
                 from C:\Users\tanaka\Documents\Arduino\buildtest\buildtest.ino:1:
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
```

tanakamasayuki, 2021-05-14:
# built-in-libraries
https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/Compiler.java#L249
```C
    addPathFlagIfPathExists(cmd, "-built-in-libraries", BaseNoGui.getContentFile("libraries"));
```

# BaseNoGui.getContentFile
https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/BaseNoGui.java#L199
```C
  static public File getContentFile(String name) {
    String appDir = System.getProperty("APP_DIR");
    if (appDir == null || appDir.length() == 0) {
      appDir = currentDirectory;
    }
    File installationFolder = new File(appDir);
    return new File(installationFolder, name);
  }
```

# APP_DIR
https://github.com/arduino/Arduino/blob/master/build/windows/launcher/config.xml#L70
```
    <opt>-DAPP_DIR="%EXEDIR%"</opt>
```

It may be a fixed value.

lbernstone, 2021-05-14:
This will be fixed in 2.0.0 (the library versions will be bumped.  I think there is another issue talking about this).  As a workaround, I would propose fixing arduino's version (Arduino/libraries/SD/library.properties and arduino-1.8.14/libraries/SD/library.properties) with:
`architectures=avr,samd,sam,megaavr,teensy`

per1234, 2021-05-14:
Hi all. This has already been fixed and new releases of both Arduino IDEs and Arduino CLI will be out soon. For details, please see this: https://github.com/arduino/arduino-cli/issues/1292

lbernstone, 2021-05-14:
Ok, fixed in the repo SD library also.

lbernstone, 2021-05-14:
Ok, so rather than changing arduino files, change <esp32>/libraries/SD/library.properties:
`name=SD`

tanakamasayuki, 2021-05-15:
https://github.com/arduino/Arduino/releases/tag/1.8.15

It has been fixed.

closed 2021-05-15 by tanakamasayuki