espressif/arduino-esp32 #5183: ESP32-DevKitM-1 Support?

K1rdro, 2021-05-14:
### Hardware:
Board:							ESP32-DevKitM-1 (ESP32-U4WDH)
Core Installation version:			2.0.0-alpha1
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
Is ESP32-DevkitM-1 (ESP32-U4WDH) supported? It's single core, so it gets the same error that the solo's were getting:
```
13:33:33.253 -> E (118) cpu_start: Running on single core variant of a chip, but app is built with multi-core support.
13:33:33.253 -> E (118) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig
```

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(21, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(21, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(21, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Debug Messages:
```
13:33:33.159 -> ELF file SHA256: 0000000000000000
13:33:33.159 -> 
13:33:33.159 -> Rebooting...
13:33:33.159 -> ets Jul 29 2019 12:21:46
13:33:33.159 -> 
13:33:33.159 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:33:33.159 -> configsip: 188777542, SPIWP:0xee
13:33:33.159 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:33:33.159 -> mode:DIO, clock div:2
13:33:33.159 -> load:0x3fff0030,len:1252
13:33:33.159 -> load:0x40078000,len:12692
13:33:33.159 -> load:0x40080400,len:3100
13:33:33.159 -> entry 0x400805ec
13:33:33.253 -> E (118) cpu_start: Running on single core variant of a chip, but app is built with multi-core support.
13:33:33.253 -> E (118) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig
13:33:33.300 -> 
13:33:33.300 -> abort() was called at PC 0x400825e9 on core 0
13:33:33.300 -> 
13:33:33.300 -> 
13:33:33.300 -> Backtrace:0x400d4c55:0x3ffe3b800x40087045:0x3ffe3ba0 0x4008c269:0x3ffe3bc0 0x400825e9:0x3ffe3c40 0x40078f6b:0x3ffe3c90  |<-CORRUPTED
```

lbernstone, 2021-05-14:
https://github.com/lbernstone/arduino-esp32-solo

K1rdro, 2021-05-14:
Yep, tried that, but I get a message back that it's an ESP32, not a single core.

lbernstone, 2021-05-14:
What is the error?  If it is from esptool, you may be able to just update the tool.

K1rdro, 2021-05-14:
Sorry, my mistake. I was getting the 'ESP32, not single core' message when I was trying it with the C3 and S2 board options. I'd followed the instructions on yours, but I didn't see an option for the Solo in the board menu, and choosing the dev module gave the same issue as above. I may be doing something wrong, or or just didn't select the intended board?

zhostik88, 2021-05-28:
> ### Hardware:
>
> Board:							ESP32-DevKitM-1 (ESP32-U4WDH)
>
> Core Installation version:			2.0.0-alpha1
>
> IDE name:						Arduino IDE
>
> Flash Frequency:					40Mhz
>
> PSRAM [...]

RunningDroid, 2021-07-22:
@K1rdro I had the same issue on Linux, then I merged arduino-esp32-solo with `esp32/hardware/esp32/1.0.6/tools` and now I have a watchdog rebooting my ESP32-DevKitM-1 because nothing's resetting the watchdog:

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
E (10147) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10147) task_wdt:  - IDLE0 (CPU 0)
E (10147) task_wdt: Tasks currently running:
E (10147) task_wdt: CPU 0: loopTask
E (10147) task_wdt: Aborting.
abort() was called at PC 0x400e0e47 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008443b:0x3ffbe490 0x40084699:0x3ffbe4b0 0x400e0e47:0x3ffbe4d0 0x400830fe:0x3ffbe4f0 0x400d1c9d:0x3ffb1fb0 0x40085651:0x3ffb1fd0
```

K1rdro, 2021-09-22:
Tested again a few days ago with the current release. Still running into the same issues.

me-no-dev, 2021-09-22:
We might add support for single-core ESP32, but it is not on the roadmap yet.

bfaliszek, 2021-11-06:
any new information on support for ESP32-SOLO-1?

markingle, 2022-04-24:
Following this issue

sixtyfive, 2022-04-25:
Sneaky

ma-lalonde, 2022-05-17:
I'm also interested in single core support, for single-core ESP32-MINI-1 chips

lbernstone, 2022-05-17:
There are automated platform releases by the tasmota team for platformio.  https://github.com/Jason2866/esp32-arduino-lib-builder/releases/.
Look through the assets for the latest solo build and use that as the platform= in your env.

ma-lalonde, 2022-05-17:
Wow! Thanks @lbernstone! That's a really nice patch.
I downloaded the Assets from the repository's release 841 (latest available), replaced my arduino-esp32 git folder (I use Arduino IDE on Ubuntu 20.04 and followed the [manual installation guide](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#linux)), and that was it.

ma-lalonde, 2022-05-19:
Another option that works is to follow the [build guide](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html#), which is very short and straightforward. For the ESP32-MINI-1 it was just a matter of adding `CONFIG_FREERTOS_UNICORE=y` to `configs/defconfig.esp32`. I haven't performed all unit tests

Jason2866, 2022-05-20:
@ma-lalonde correct, thats the only needed different entry in sdkconfig

VojtechBartoska, 2022-06-15:
Hello all users,

we discussed ESP32-DevKitm-1 Support internally and I'm sharing current state.

---

Be aware that this resolution can change in the future¨, mainly depending on Community requests.

---

From our perspective, it doesn't make a lot of sense to add official support for single core ESP32 SoCs now as it's the same board just like regular ESP32 and only difference is that the second core is "disabled". Single Core ESP32s are rarely used comparing to regular ESP32s (our view).

### For now there are 2 options how to use Single Core ESP32s:

#### 1) Adding it into building process of [Arduino-Lib-Builder](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html#)
- it was mentioned above by @ma-lalonde in this [comment](https://github.com/espressif/arduino-esp32/issues/5183#issuecomment-1132219416)

#### 2) Use Arduino as an ESP-IDF component 
- [related documentation about Arduino as Component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html)
- SingleCore ESP32 is supported in ESP-IDF

---

Let's us know if you use Single Core ESP32s more widely and you will really appreciate adding the official support. We are open to reconsider our resolution. 👍

Jason2866, 2022-06-15:
We (Tasmota project team) provide a Solo1 version (builded with Arduino Lib Builder.
Based on core 2.0.3 [Link to the framework](https://github.com/tasmota/arduino-esp32/releases/download/2.0.3/framework-arduinoespressif32-solo1-203.tar.gz) and for Platformio a [platform package](https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-solo1-v.2.0.3.zip)
Only needed Platformio entry:
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-solo1-v.2.0.3.zip
```

ma-lalonde, 2022-06-15:
I do the same as Jason at Connaxio, and the platformIO entry is just as simple. Our fork of arduino-esp32 supports single core on a separate branch. Could that be an option? A single core branch that is updated on each release?

It seems silly to have multiple external third parties maintain what is basically the same repository and documentation. And, since I'm not sure that the differences are between the MINI and the SOLO, and I don't want to depend on a third party, here we are, both maintaining the same thing. Though I suspect there are many more build options that could be subject to debate. For example, task preemptivity. 

My stance on the matter is that Arduino exists primarily for accessibility, not performance. If it were up to me, I'd make the Arduino default build support as many ICs as possible, while users seeking more performance, who may be more knowledgeable, should go through doing the custom builds. I don't think that's a popular opinion, however. ¯\_(ツ)_/¯

untergeek, 2022-06-15:
Please, official support will be greatly appreciated.

lbernstone, 2022-06-15:
The configuration capacity to have multiple board types only came about in v2.0.  That was just over a year ago.  The need was critical for the new devices, and so they were implemented first. Now I guess there is enough time to circle back and set things up for solo/pico.  When I first started building those libraries, it was just a couple files, now it is pretty much the entire set of archive libraries. Patience, please.

me-no-dev, 2022-06-16:
ESP32-Solo chips are not that easy to support in this Arduino repo (for now), precisely because the difference is one sdkconfig item. We are trying to figure out the best way to have the IDF libs portion separate elsewhere. Then we can easily add support for Solo or other weird variants/chips.

I do not agree that we need to support every chip that Espressif makes. Some chips do not make sense for Arduino, others are not widely used (like the Solo) to warrant hundreds more megabytes of libs in the repo. Still we are doing our best to make it easy for you to create your own build for the chip you want (using the lib-builder and abstracting as much as possible).

ArnieO, 2022-06-16:
Just to be sure we understand the terms used in this discussion: What is covered by the term "Solo"? 

As far as I know, there are no chips using term "Solo" in their name: https://www.espressif.com/en/products/socs
And on the modules, the term "Solo" seems to be used for a certain form factor, not really identifying the SoC/chip: https://www.espressif.com/en/products/modules

As per above link, Espressif distinguishes between the following series for their SoCs:

- ESP32: All dual-core, except one single-core used in ESP32-SOLO-1 module (but SoC and module are NRND, so will likely disappear)
- ESP32-S, with two "sub-series": 
  - S2 (single-core, no BLE)
  -  S3 (dual-core, with BLE)
- ESP32-C, with two "sub-series":
  - C2 (RISC-V single core, up to 120 MHz)
  - C3 (RISC-V single core, up to 160 MHz, added features vs C2)
- (And ESP8266, which is outside the scope here)

The modules using the SoCs have the same taxonomy.

**When the term "Solo" is used in this thread, are we then in fact talking about ESP32-S and ESP32-C? 
Or is the term used on all single-core SoCs (and the modules using them)?**

Our company is using two types of S2 modules in our projects (S2-SOLO and S2-MINI), and their programming is "the same". S2 was chosen as it seems to be the lowest consuming of the ESP32 modules available, and form factor (SOLO vs MINI) was chosen based on physical implementation restrictions. FWIW, we are currently using the "Tasmota build" (thank you to their team!).

me-no-dev, 2022-06-16:
@ArnieO `Solo` is the old name for the single core ESP32. It is a binned version of otherwise dual-core chip. To support that variant of the same chip is the issue, not supporting other single-core chips with different names/features.

lbernstone, 2022-06-16:
The ESP32-U4WDH is the re-release of an SoC originally called ESP32-SOLO (circa 2018?).  The Solo is no longer available, but the U4WDH and modules based on it can still be purchased new from Mouser.  These are ESP32 devices. -S* and -C* are fully supported already.
[image]

ArnieO, 2022-06-16:
@me-no-dev and @lbernstone : Thank you for your rapid clarification! 

I understand then that the discussion concerns one SoC type used in one Espressif module type (ESP32-SOLO-1), both no longer manufactured. So the subject is very narrowed down compared to my initial impression due to the term "Solo". No worries then from my side if this chip/module is not officially supported.

Jason2866, 2022-06-16:
The ESP32 single core version is still available and used in commercial products.
For example new "Shelly" devices. Thats why we (Tasmota) provide the framework, because Shelly devices are often bought to run with Tasmota as alternative cloud free firmware .

mlpplm, 2022-10-16:
As there is now very confusing information available in internet regarding ESP32-U4WD(H) support in Arduino IDE, I would appreciate if someone could describe step-by-step instructions, how to get support for unicore ESP32 in Arduino IDE. 
I use the latest A IDE2.0.0 and have been trying to use the board manager to include the 'official' ESP cores (with no luck = no unicore support), as well as manual installation with git. Have been trying to replace the git esp32 folder with the intructions seen here (Tasmota  version framework-arduinoespressif32-solo1.zip) with no luck. There is no board ESP32-1-solo or whatsover available in board selection (not official nor sketch esp32 boards). Which board should I select? DOIT ESP32 DEVKIT V1? (sorry if this is a stupid request, but with esp8266 everything has been very clear and consistent, but with esp32 world nothing is clear)

Jason2866, 2022-10-16:
@mlpplm **Arduino does not official support Unicore Esp32.**
You have to compile the needed Arduino libs yourself or use a fork which does support.
My actual Tasmota Platformio Platform does support the solo1 and all other boards. I will not release a Arduino IDE version only (i dont use...) or doing documentation for.
You find the actual one [here](https://github.com/tasmota/platform-espressif32) and it is described how to use with Platformio.

VojtechBartoska, 2022-10-17:
@mlpplm It's good point and we are aware of this. We will include this in our Docs.

thorathome, 2023-10-10:
My vote is to see these ESP32-U4WD single core devices supported in Arduino IDE. I want to flash the Shelly Plus 1 devices with custom code. Thanks.

rftafas, 2025-01-10:
ESP32-Solo won't be added. If users want to work on this chip, they must use the Arduino Component for ESP-IDF.

After tracking this discussion, besides using the Arduino Component for ESP-IDF, users can get Tasmota, or are already using the dual core replacement (which should work, but seems we are seeing some difficulties - yet, this is a different conversation). 

This seems to cover the majority of the users.

closed 2025-01-10 by rftafas