espressif/arduino-esp32 #5185: interrupt GPIO36 and GPIO37 signal mixing/crosstaling

smarq8, 2021-05-15:
### Hardware:
Board:							TTGO T-DISPLAY
Core Installation version:			espressif32@3.2.0 arduino framework
IDE name:						platformio

### Description:
I have 2 interrupts from sensors, MPU6500_INT=37 and ADXL355_INT=36, the only thing I do is increment some counter for each sensor, everything were fine as I use them separately, but as I run them both at one then interrupts seems to "crosstalking" and my counters do not show me expected value.
Removing wire ADXL355_INT=36 cause firing both interrupts at the exact same time.
However changing ADXL355_INT to 33 or 38 fixing that trouble, and it seems something happen between GPIO36 and GPIO37.
Both signal are push-pull and both pins are configured as INPUT, resistance between them is 10Mohm so its probably not "obvious" shortcircuit.

```
static IRAM_ATTR volatile uint32_t _interruptsN[4] = {0,0,0,0};

void IRAM_ATTR _onInterrupt0(){
	_interruptsN[0]++;
	// if(_onInterruptExt[0]) _onInterruptExt[0]();
}
void IRAM_ATTR _onInterrupt1(){
	_interruptsN[1]++;
	// if(_onInterruptExt[1]) _onInterruptExt[1]();
}
```

atanisoft, 2021-05-24:
@smarq8 In most ESP32 devices that use a WROOM-32 or WROVER module GPIO 36 is connected to GPIO37 using a 270pF capacitor. GPIO 38 and 39 are the same way. You can also find the schematic for the T-Display here: https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/schematic/ESP32-TFT(6-26).pdf

Very few ESP32 DevKit boards expose GPIO 37 and 38 for that reason and those that do shouldn't!

jason-kong, 2021-05-26:
Running into something similar right now on a very simple circuit.

Using external pullup resistors and some keyswitches, the `digitalRead` values from GPIO36 cross talk with GPIO37, so pressing one switch claims that the other is being pressed as well (externally checking my project with a multimeter suggests it's gotta be internal to the board itself and not my parts)

I saw the capacitors @atanisoft  in the schematic, could they be removed perhaps? I'm assuming they're SMT mounted on some boards (mine is a Sparkfun ESP32 thing)

atanisoft, 2021-05-26:
They are internal to the esp32 module in most cases and can not easily be removed. In the case of the T-Display they might be removable. But I wouldn't recommend using GPIO 37 or 38 for any purpose.

jason-kong, 2021-05-26:
> They are internal to the esp32 module in most cases and can not easily be removed. In the case of the T-Display they might be removable. But I wouldn't recommend using GPIO 37 or 38 for any purpose.

Thanks for the tip, I'm running out of GPIOs on my keyboard project but I hope (pray) there's two I can swap 37 and 38 over to

atanisoft, 2021-05-26:
You might consider moving over to the ESP32-S2 which has more IO pins and a dedicated GPIO driver that can be used for a keypad input.

jason-kong, 2021-05-26:
Been looking into that a bit though I recall it doesn't have the BLE functionality of the plain ESP32, though the USB host could would be a great upgrade.

closed 2021-08-21 by stale[bot]