espressif/arduino-esp32 #5187: HTTPClient only authentification is working but no possible to post somehting else

kikisens, 2021-05-16:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>HELTEC WIFI LORA 32V2
|<B>Version/Date</B>||
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|240Mhz|
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
purpose of my project is to do on the esp32 the exact behaviour of the extraordinary project here [](https://github.com/iMicknl/python-tahoma-api) based on Python but unable to reproduce it. It is a script taht allow raspberry to connect tahoma form somfy wesite and execute some scenario. On the ESP, The login is working but i'm not able to execute scenario. I have the feeling the keep alive function is not working but my code should be the issue as I have a lack of knowledge about HTTP.If anyone could help, it will be very appreciated.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
/*
.
*/

#include <WiFi.h>
#include <WiFiClient.h>
#include <HTTPClient.h>

String USERNAME = "HIDDEN";
String PASSWORDSOMFY = "HIDDEN";
const char* ssid = "HIDDEN";
const char* password = "HIDDEN";

//Your Domain name with URL path or IP address with path
String serverName = "https://tahomalink.com/enduser-mobile-web/enduserAPI/";

// the following variables are unsigned longs because the time, measured in
// milliseconds, will quickly become a bigger number than can be stored in an int.
unsigned long lastTime = 0;
// Timer set to 10 minutes (600000)
//unsigned long timerDelay = 600000;
// Set timer to 5 seconds (5000)
unsigned long timerDelay = 10000;
int i;
void setup() {
  i=0;
  Serial.begin(115200); 

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());
 
  Serial.println("Timer set to 5 seconds (timerDelay variable), it will take 5 seconds before publishing the first reading.");
}

void sendHeader(const char* aHeaderName, const char* aHeaderValue);

void loop() {
  if (1==1){
    i =1;
    delay(5000);
  //Send an HTTP POST request every 10 minutes
  if ((millis() - lastTime) > timerDelay) {
    //Check WiFi connection status
    if(WiFi.status()== WL_CONNECTED){

      WiFiClient client2;
      HTTPClient http;
      http.setReuse(true);
      http.setUserAgent("test");
      String serverPath = serverName + "login"; 
      String serverPathscene = serverName + "exec/"scenarioid"";      
      Serial.println(serverPath);
      Serial.println(serverPathscene);
      // Your Domain name with URL path or IP address with path
      http.begin( serverPath.c_str());
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      http.addHeader("Connection", "keep-alive");
      
      String payload = "userId=hidden&userPassword=hidden";
      int httpResponseCode = http.POST(payload); 
      if (httpResponseCode>0) {
        Serial.print("HTTP Response code: ");
        Serial.println(httpResponseCode);
        String payload = http.getString();
        Serial.println(payload);
          }
        else {
          Serial.print("Error codesecond: ");
          Serial.println(httpResponseCode);
        }
        delay(500);
      http.end();
       
      //********************************************
      delay(50);
      http.setReuse(true);
      http.setUserAgent("test");
      http.begin( serverPathscene.c_str());
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      String payload2 = "";
      int httpResponseCode2 = http.sendRequest("POST", payload2);
      //int httpResponseCode2 = http.POST(payload2); 
      if (httpResponseCode2>0) {
        Serial.print("HTTP Response code2: ");
        Serial.println(httpResponseCode2);
        String payload = http.getString();
        Serial.println(payload);
          }
        else {
          Serial.print("Error codesecond2: ");
          Serial.println(httpResponseCode);
        }
        delay(500);
      http.end();
      
      //*********************************************
    }
    else {
      Serial.println("WiFi Disconnected");
    }
    delay(20000);
    lastTime = millis();
  }
  }
}

### Debug Messages:
```
09:36:28.971 -> https://tahomalink.com/enduser-mobile-web/enduserAPI/login
09:36:28.971 -> https://tahomalink.com/enduser-mobile-web/enduserAPI/exec/625461d9-b2ec-4536-a077-3d8f599d79c6
09:36:28.971 -> [W][HTTPClient.cpp:257] beginInternal(): unexpected protocol: https, expected http
09:36:29.012 -> [D][HTTPClient.cpp:293] beginInternal(): protocol: https, host: tahomalink.com port: 443 url: /enduser-mobile-web/enduserAPI/login
09:36:29.012 -> [D][HTTPClient.cpp:579] sendRequest(): request type: 'POST' redirCount: 0
09:36:29.012 -> 
09:36:29.038 -> [I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
09:36:29.857 -> [D][HTTPClient.cpp:1125] connect():  connected to tahomalink.com:443
09:36:29.993 -> [D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 200
09:36:29.993 -> [D][HTTPClient.cpp:1264] handleHeaderResponse(): Transfer-Encoding: chunked
09:36:29.993 -> [D][HTTPClient.cpp:603] sendRequest(): sendRequest code=200
09:36:29.993 -> 
09:36:29.993 -> [D][HTTPClient.cpp:891] writeToStream():  read chunk len: 45
09:36:30.028 -> [D][HTTPClient.cpp:1394] writeToStreamDataBlock(): connection closed or file end (written: 45).
09:36:30.028 -> [D][HTTPClient.cpp:891] writeToStream():  read chunk len: 0
09:36:30.028 -> [D][HTTPClient.cpp:378] disconnect(): still data in buffer (2), clean up.
09:36:30.028 -> 
09:36:30.028 -> [D][HTTPClient.cpp:385] disconnect(): tcp keep open for reuse
09:36:30.540 -> [D][HTTPClient.cpp:385] disconnect(): tcp keep open for reuse
09:36:30.574 -> [W][HTTPClient.cpp:257] beginInternal(): unexpected protocol: https, expected http
09:36:30.574 -> [D][HTTPClient.cpp:293] beginInternal(): protocol: https, host: tahomalink.com port: 443 url: /enduser-mobile-web/enduserAPI/exec/625461d9-b2ec-4536-a077-3d8f599d79c6
09:36:30.608 -> [D][HTTPClient.cpp:579] sendRequest(): request type: 'POST' redirCount: 0
09:36:30.608 -> 
09:36:30.608 -> [D][HTTPClient.cpp:1085] connect(): already connected, reusing connection
09:36:30.608 -> [D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 401
09:36:30.643 -> [D][HTTPClient.cpp:1260] handleHeaderResponse(): size: 66
09:36:30.643 -> [D][HTTPClient.cpp:603] sendRequest(): sendRequest code=401
09:36:30.643 -> 
09:36:30.643 -> [D][HTTPClient.cpp:1394] writeToStreamDataBlock(): connection closed or file end (written: 66).
09:36:30.643 -> [D][HTTPClient.cpp:385] disconnect(): tcp keep open for reuse
09:36:31.157 -> [D][HTTPClient.cpp:385] disconnect(): tcp keep open for reuse

```

TheRealDJ, 2021-05-22:
I believe the `http.setReuse(true)` should be called before the first `http.begin()` (the login). But you should be using HTTPS. You are sending login credentials over the internet unencrypted.

kikisens, 2021-05-22:
First of all , many thanks for your help on this subject. 

By checking the library I was thinking the connection is automatically detected as HTTPS due to the link. What I need to change?
regrding the auth token, I don't know how to get it , it is not given in the request response. 

once again, thanks for your help!

lbernstone, 2021-05-22:
https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/WWW-Authenticate
https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/Authorization/Authorization.ino

TheRealDJ, 2021-05-23:
To use TLS (HTTPS), change
```
WiFiClient client2;
```
to
```
setClock();
WiFiClientSecure client2;
client2.setCACert(cacert);
```
You will need to define `cacert` and `setClock` ...
```
// Subject: C=FR, ST=Paris, L=Paris, O=Gandi, CN=Gandi Standard SSL CA 2
// Issuer: C=US, ST=New Jersey, L=Jersey City, O=The USERTRUST Network, CN=USERTrust RSA Certification Authority
const char* cacert = "-----BEGIN CERTIFICATE-----\n" \
"MIIF6TCCA9GgAwIBAgIQBeTcO5Q4qzuFl8umoZhQ4zANBgkqhkiG9w0BAQwFADCB\n" \
"iDELMAkGA1UEBhMCVVMxEzARBgNVBAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0pl\n" \
"cnNleSBDaXR5MR4wHAYDVQQKExVUaGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNV\n" \
"BAMTJVVTRVJUcnVzdCBSU0EgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMTQw\n" \
"OTEyMDAwMDAwWhcNMjQwOTExMjM1OTU5WjBfMQswCQYDVQQGEwJGUjEOMAwGA1UE\n" \
"CBMFUGFyaXMxDjAMBgNVBAcTBVBhcmlzMQ4wDAYDVQQKEwVHYW5kaTEgMB4GA1UE\n" \
"AxMXR2FuZGkgU3RhbmRhcmQgU1NMIENBIDIwggEiMA0GCSqGSIb3DQEBAQUAA4IB\n" \
"DwAwggEKAoIBAQCUBC2meZV0/9UAPPWu2JSxKXzAjwsLibmCg5duNyj1ohrP0pIL\n" \
"m6jTh5RzhBCf3DXLwi2SrCG5yzv8QMHBgyHwv/j2nPqcghDA0I5O5Q1MsJFckLSk\n" \
"QFEW2uSEEi0FXKEfFxkkUap66uEHG4aNAXLy59SDIzme4OFMH2sio7QQZrDtgpbX\n" \
"bmq08j+1QvzdirWrui0dOnWbMdw+naxb00ENbLAb9Tr1eeohovj0M1JLJC0epJmx\n" \
"bUi8uBL+cnB89/sCdfSN3tbawKAyGlLfOGsuRTg/PwSWAP2h9KK71RfWJ3wbWFmV\n" \
"XooS/ZyrgT5SKEhRhWvzkbKGPym1bgNi7tYFAgMBAAGjggF1MIIBcTAfBgNVHSME\n" \
"GDAWgBRTeb9aqitKz1SA4dibwJ3ysgNmyzAdBgNVHQ4EFgQUs5Cn2MmvTs1hPJ98\n" \
"rV1/Qf1pMOowDgYDVR0PAQH/BAQDAgGGMBIGA1UdEwEB/wQIMAYBAf8CAQAwHQYD\n" \
"VR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMCIGA1UdIAQbMBkwDQYLKwYBBAGy\n" \
"MQECAhowCAYGZ4EMAQIBMFAGA1UdHwRJMEcwRaBDoEGGP2h0dHA6Ly9jcmwudXNl\n" \
"cnRydXN0LmNvbS9VU0VSVHJ1c3RSU0FDZXJ0aWZpY2F0aW9uQXV0aG9yaXR5LmNy\n" \
"bDB2BggrBgEFBQcBAQRqMGgwPwYIKwYBBQUHMAKGM2h0dHA6Ly9jcnQudXNlcnRy\n" \
"dXN0LmNvbS9VU0VSVHJ1c3RSU0FBZGRUcnVzdENBLmNydDAlBggrBgEFBQcwAYYZ\n" \
"aHR0cDovL29jc3AudXNlcnRydXN0LmNvbTANBgkqhkiG9w0BAQwFAAOCAgEAWGf9\n" \
"crJq13xhlhl+2UNG0SZ9yFP6ZrBrLafTqlb3OojQO3LJUP33WbKqaPWMcwO7lWUX\n" \
"zi8c3ZgTopHJ7qFAbjyY1lzzsiI8Le4bpOHeICQW8owRc5E69vrOJAKHypPstLbI\n" \
"FhfFcvwnQPYT/pOmnVHvPCvYd1ebjGU6NSU2t7WKY28HJ5OxYI2A25bUeo8tqxyI\n" \
"yW5+1mUfr13KFj8oRtygNeX56eXVlogMT8a3d2dIhCe2H7Bo26y/d7CQuKLJHDJd\n" \
"ArolQ4FCR7vY4Y8MDEZf7kYzawMUgtN+zY+vkNaOJH1AQrRqahfGlZfh8jjNp+20\n" \
"J0CT33KpuMZmYzc4ZCIwojvxuch7yPspOqsactIGEk72gtQjbz7Dk+XYtsDe3CMW\n" \
"1hMwt6CaDixVBgBwAc/qOR2A24j3pSC4W/0xJmmPLQphgzpHphNULB7j7UTKvGof\n" \
"KA5R2d4On3XNDgOVyvnFqSot/kGkoUeuDcL5OWYzSlvhhChZbH2UF3bkRYKtcCD9\n" \
"0m9jqNf6oDP6N8v3smWe2lBvP+Sn845dWDKXcCMu5/3EFZucJ48y7RetWIExKREa\n" \
"m9T8bJUox04FB6b9HbwZ4ui3uRGKLXASUoWNjDNKD/yZkuBjcNqllEdjB+dYxzFf\n" \
"BT02Vf6Dsuimrdfp5gJ0iHRc2jTbkNJtUQoj1iM=\n" \
"-----END CERTIFICATE-----\n";

void setClock() {
  configTime(0, 0, "pool.ntp.org");

  Serial.print(F("Waiting for NTP time sync: "));
  time_t nowSecs = time(nullptr);
  while (nowSecs < 8 * 3600 * 2) {
    delay(500);
    Serial.print(F("."));
    yield();
    nowSecs = time(nullptr);
  }

  Serial.println();
  struct tm timeinfo;
  gmtime_r(&nowSecs, &timeinfo);
  Serial.print(F("Current time: "));
  Serial.print(asctime(&timeinfo));
}
```
The cacert I included above is the CA that issued the SSL cert to https://tahomalink.com/

kikisens, 2021-05-24:
Hello to all of you, according to your remarks, I 've succeeded in securing the connection and get back the header set-cookie with JSESSIONID as a token. Unfortunately by following the link proposed above with setauthorization(), i don't really know where to use this token and still get 401 issue. Any other Idea?

kikisens, 2021-05-25:
Hello again, finally I make it work by adding a header "cookie". Many Many Many thanks for the time you spent on my subject.
And have a great day!

closed 2021-05-25 by kikisens