espressif/arduino-esp32 #5188: Got analogRead() value always 0

dev-rijan, 2021-05-17:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200
Computer OS:                                         Ubuntu

### Description:
I tried to read analog value using all ADC1 GPIO pins but always got 0.

### Sketch:  
```cpp

char text[50] = {'\0'};

void setup() {
  pinMode(GPIO_NUM_36, INPUT);                                                // set input mode
  analogSetPinAttenuation(GPIO_NUM_36, ADC_11db);                             // ADC setting pin ADC1_CH0

  pinMode(GPIO_NUM_39, INPUT);                                                // set input mode
  analogSetPinAttenuation(GPIO_NUM_39, ADC_11db);                             // ADC setting pin ADC1_CH3

  pinMode(GPIO_NUM_34, INPUT);                                                // set input mode
  analogSetPinAttenuation(GPIO_NUM_34, ADC_11db);                             // ADC setting pin ADC1_CH6

  pinMode(GPIO_NUM_35, INPUT);                                                // set input mode
  analogSetPinAttenuation(GPIO_NUM_35, ADC_11db);                             // ADC setting pin ADC1_CH7

  pinMode(GPIO_NUM_32, INPUT);                                                // set input mode
  analogSetPinAttenuation(GPIO_NUM_32, ADC_11db);                             // ADC setting pin ADC1_CH4

  pinMode(GPIO_NUM_33, INPUT);                                                // set input mode
  analogSetPinAttenuation(GPIO_NUM_33, ADC_11db);                             // ADC setting pin ADC1_CH5

  Serial.begin(115200);                                                       // Arduino serial port for debugging @ 115.2 kBaud
  while (!Serial) continue;                                                   // wait for serial port to connect. Needed for native USB

}

void loop() {

  Serial.println("analogRead of pins...");
  adcStart(36);
  Serial.println(adcStart(GPIO_NUM_36));
  snprintf(text, 40, "Analog value pin %u (ADC1_CH0) = %u", GPIO_NUM_36, analogRead(36));
  Serial.println(text);

  snprintf(text, 40, "Analog value pin %u (ADC1_CH3) = %u", GPIO_NUM_39, analogRead(GPIO_NUM_39));
  Serial.println(text);

  snprintf(text, 40, "Analog value pin %u (ADC1_CH6) = %u", GPIO_NUM_34, analogRead(GPIO_NUM_34));
  Serial.println(text);

  snprintf(text, 40, "Analog value pin %u (ADC1_CH7) = %u", GPIO_NUM_35, analogRead(GPIO_NUM_35));
  Serial.println(text);
  delay(1000);                      // wait a second
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2021-05-17:
It is not necessary to call adcStart to call analogRead, and it will conflict until you call adcEnd.  The adcStart function has been removed in 1.0.5+ in order to use adc calibration based on a factory-tested vref.

dev-rijan, 2021-05-17:
@lbernstone I tested removing "adcStart"  but still get value 0.  I can get correct value using GPIO pin 25 but i can't use this pin because  i need both wifi and analog read.

nmadan692, 2021-05-27:
i am using espr one 32 and unable to read analog value from A0. Is there is any solutions for this?

lbernstone, 2021-05-27:
@nmadan692 A0 has no real meaning on esp32.  Use gpio numbers.
@dev-rijan This code works just fine for me.  It returns 0 when unattached, gives me 105-120 when connected to a floating pin (23 in test).
```
int adc_pin = 36;
void setup() {
  Serial.begin(115200);
  pinMode(adc_pin, INPUT);
}
void loop() {
  Serial.printf("Pin %u = %u\n", adc_pin, analogRead(adc_pin));
  delay(1000);
}

closed 2021-08-21 by stale[bot]