espressif/arduino-esp32 #5194: ESP32C3 deepsleep demo

Rita858, 2021-05-19:
IDE environment：
arduino 1.8.12
esp32 version：2.0.0-alpha1

issue：
A series of error messages occurred while I was compiling deepSleep 3 demo.
`C:\Users\AI-THI~1\AppData\Local\Temp\arduino_modified_sketch_12170\ExternalWakeUp.ino: In function 'void setup()':
ExternalWakeUp:69:3: error: 'esp_sleep_enable_ext0_wakeup' was not declared in this scope
   esp_sleep_enable_ext0_wakeup(GPIO_NUM_2,1); //1 = High, 0 = Low
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\AI-THI~1\AppData\Local\Temp\arduino_modified_sketch_12170\ExternalWakeUp.ino:69:3: note: suggested alternative: 'esp_sleep_enable_wifi_wakeup'
   esp_sleep_enable_ext0_wakeup(GPIO_NUM_2,1); //1 = High, 0 = Low
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
   esp_sleep_enable_wifi_wakeup
exit status 1
'esp_sleep_enable_ext0_wakeup' was not declared in this scope` 

Dose ESP32C3 deepsleep function is not supported in this environment?
How can I fix it?

Looking forward to your reply,thank you.

lbernstone, 2021-05-19:
external wakeup is not supported in C3.  Deep sleep (eg TimerWakeUp) should work.
https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_modes.c#L171

mad462, 2021-06-18:
> external wakeup is not supported in C3. Deep sleep (eg TimerWakeUp) should work.
> https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_modes.c#L171

请问在未来C3会支持外部唤醒吗

lbernstone, 2021-06-18:
english only, please

mad462, 2021-06-18:
> english only, please

Sorry, will ESP32C3 support external Ext0 wakeup in arduino in the future?THX

lbernstone, 2021-06-18:
esp32-c3 has no processor running during deep sleep, so I don't believe it will be possible.  However, the riscv cpu used in the c3 can be clocked down quite low, and is lower in power consumption than esp32-s2's ulp when in light sleep.

mad462, 2021-06-18:
> esp32-c3 has no processor running during deep sleep, so I don't believe it will be possible. However, the riscv cpu used in the c3 can be clocked down quite low, and is lower in power consumption than esp32-s2's ulp when in light sleep.

Thank you for your reply(*￣︶￣)

closed 2021-09-19 by stale[bot]

mopaho, 2022-03-11:
I do not uderstand why it is not supported since it is described in the datasheet as per ESP32. It should work !!!

kriswiner, 2022-03-27:
What is the lowest ESP32C3Mini1 clock speed that still supports BLE?

lbernstone, 2022-03-28:
AFAIK, the radio cannot operate at a lower speed than the crystal frequency, which is almost certainly going to be 40MHz on any Espressif manufactured modules.

mcorrigan, 2023-01-10:
I haven't figured this out yet either, but I'm under the impression that deep sleep wake up sources are supported for ESP32C3 per this document: https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/api-reference/system/sleep_modes.html

kriswiner, 2023-01-10:
I am pretty sure that for C3 deep sleep is limited to a timer for wakeup.

darcythomas, 2024-06-28:
Try this instead:

`  esp_deep_sleep_enable_gpio_wakeup(GPIO_NUM_4, ESP_GPIO_WAKEUP_GPIO_HIGH);`