espressif/arduino-esp32 #5195: SD_MMC.begin() without SDcard present causes abort()

thezenox, 2021-05-19:
With no card present esp32 crashes instead of error

Code:
```
#include "FS.h"
#include "SD_MMC.h"

void setup(){
    Serial.begin(115200);

    Serial.println("Starting");

    if(!SD_MMC.begin()) {
        Serial.println("sd Init Error");
    }
}
void loop(){}
```

Serial output:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1412
load:0x40078000,len:13400
load:0x40080400,len:3672
entry 0x400805f8
⸮Starting
E (2406) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (2407) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
assertion "xQueue" failed: file "../esp-idf/components/freertos/queue.c", line 1940, function: uxQueueMessagesWaiting

abort() was called at PC 0x400eae10 on core 1
```

Stacktrace:
```
0x400d7739: panic_abort at ../esp-idf/components/esp_system/panic.c line 365
0x4008a49d: esp_system_abort at ../esp-idf/components/esp_system/esp_system.c line 126
0x40090295: abort at ../esp-idf/components/newlib/abort.c line 46
0x400eae10: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 62
0x4008b67d: uxQueueMessagesWaiting at ../esp-idf/components/freertos/queue.c line 1940
0x400dcbab: spi_bus_remove_device at ../esp-idf/components/driver/spi_master.c line 442
0x400db1c5: deinit_slot at ../esp-idf/components/driver/sdspi_host.c line 231
0x400db32e: sdspi_host_remove_device at ../esp-idf/components/driver/sdspi_host.c line 275
0x400e4b65: esp_vfs_fat_sdmmc_mount at ../esp-idf/components/fatfs/vfs/vfs_fat_sdmmc.c line 361
0x400d27d1: fs::SDMMCFS::begin(char const*, bool, bool) at C:\Users\x\Documents\Arduino\hardware\espressif\esp32\libraries\SD_MMC\src\SD_MMC.cpp line 9
0x400d17f5: setup() at 
```

TienHuyIoT, 2021-07-09:
I have same issue,

module: ESP32 WROOM
version: install latest 2.0.0

SDMMC begin and work succeed within case the SD-card have present in slot socket SD.
If the SD-card is removed out slot socket SD. Esp32 shall crash and reset log print as below.

with version: release v1.0.6
ESP32 example SD_MMC work normal with this case.

```
rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1412
load:0x40078000,len:13416
load:0x40080400,len:3672
entry 0x400805f8
E (79) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (80) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
assertion "xQueue" failed: file "../esp-idf/components/freertos/queue.c", line 1946, function: uxQueueMessagesWaiting

abort() was called at PC 0x400eafe7 on core 1

Backtrace:0x400d8051:0x3ffb25a00x40088d41:0x3ffb25c0 0x4008e74d:0x3ffb25e0 0x400eafe7:0x3ffb2660 0x40089e89:0x3ffb2690 0x400dd4b7:0x3ffb26b0 0x400dbac1:0x3ffb26e0 0x400dbc2a:0x3ffb2720 0x400e4f95:0x3ffb2740 0x400d30e1:0x3ffb2770 0x400d2278:0x3ffb27f0 0x400d402f:0x3ffb2820 
```

ecobdoor, 2021-07-21:
I confirm the bug :

E (1724) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1725) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
assertion "xQueue" failed: file "../esp-idf/components/freertos/queue.c", line 1940, function: uxQueueMessagesWaiting

abort() was called at PC 0x40195424 on core 1

Backtrace:0x400f5715:0x3ffb25300x40091871:0x3ffb2550 0x40097f3d:0x3ffb2570 0x40195424:0x3ffb25f0 0x40092d11:0x3ffb2620 0x40127d39:0x3ffb2640 0x40126217:0x3ffb2670 0x4012638e:0x3ffb26b0 0x4013113d:0x3ffb26d0 0x400dca86:0x3ffb2700 0x400db7c1:0x3ffb2780 0x400d28fc:0x3ffb27a0 0x400e56cf:0x3ffb2820 

Decoding stack results
0x400f5715: panic_abort at ../esp-idf/components/esp_system/panic.c line 365
0x40091871: esp_system_abort at ../esp-idf/components/esp_system/esp_system.c line 126
0x40097f3d: abort at ../esp-idf/components/newlib/abort.c line 46
0x40195424: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 62
0x40092d11: uxQueueMessagesWaiting at ../esp-idf/components/freertos/queue.c line 1940
0x40127d39: spi_bus_remove_device at ../esp-idf/components/driver/spi_master.c line 442
0x40126217: deinit_slot at ../esp-idf/components/driver/sdspi_host.c line 231
0x4012638e: sdspi_host_remove_device at ../esp-idf/components/driver/sdspi_host.c line 275
0x4013113d: esp_vfs_fat_sdmmc_mount at ../esp-idf/components/fatfs/vfs/vfs_fat_sdmmc.c line 361
0x400dca86: fs::SDMMCFS::begin(char const*, bool, bool) at /home/eric/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/libraries/SD_MMC/src/SD_MMC.cpp line 75
0x400db7c1: SETUP_FILSYS() at /media/DEV/ROB_000/rover/libraries/CAM_FILSYS/CAM_FILSYS.cpp line 21
0x400d28fc: setup() at /media/DEV/ROB_000/rover/Motion_esp_http_server/Motion_esp_http_server.ino line 69
0x400e56cf: loopTask(void*) at /home/eric/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/main.cpp line 31

I'm using  arduino-esp32  version 2.0.0-alpha1 on  Esp-CAM (1 bit mode is the same behaviour).

Anybody has an IDEA ???

ecobdoor, 2021-07-21:
The backtrace looks right, I believe it's a true bug.

**esp_vfs_fat_sdmmc_mount probably fails because no sd_card...**
 - Then sdspi_host_remove_device is called
- Then deinit_slot is called
- Then spi_bus_remove_device is called and call uxQueueMessagesWaiting (see source below)

esp_err_t spi_bus_remove_device(spi_device_handle_t handle){
    SPI_CHECK(handle!=NULL, "invalid handle", ESP_ERR_INVALID_ARG);
    //These checks aren't exhaustive; another thread could sneak in a transaction inbetween. These are only here to
    //catch design errors and aren't meant to be triggered during normal operation.
    SPI_CHECK(uxQueueMessagesWaiting(handle->trans_queue)==0, "Have unfinished transactions", ESP_ERR_INVALID_STATE);
    SPI_CHECK(handle->host->cur_cs == DEV_NUM_MAX || handle->host->device[handle->host->cur_cs] != handle, "Have unfinished transactions", ESP_ERR_INVALID_STATE);
    SPI_CHECK(uxQueueMessagesWaiting(handle->ret_queue)==0, "Have unfinished transactions", ESP_ERR_INVALID_STATE);
    //return
    int spics_io_num = handle->cfg.spics_io_num;
    if (spics_io_num >= 0) spicommon_cs_free_io(spics_io_num);
    //Kill queues
    vQueueDelete(handle->trans_queue);
    vQueueDelete(handle->ret_queue);
    spi_bus_lock_unregister_dev(handle->dev_lock);
    assert(handle->host->device[handle->id] == handle);
    handle->host->device[handle->id] = NULL;
    free(handle);
    return ESP_OK;
}

me-no-dev, 2021-07-22:
As a temporary solution you can try to use the card in SPI mode :)

dotMorten, 2021-09-19:
Experiencing the exact same with the `Thing Plus C - ESP32 WROOM`. Did anyone find a workaround?