espressif/arduino-esp32 #5196: if Software or Hardware Serial .available() is used, seldom garbage is returned at .read()

pabloandresm, 2021-05-19:
### Hardware:
Board:							ESP32 Dev Module and LOLIN D32
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Flash Frequency:					tested 40Mhz and 80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600, but tested with 115200 too
Computer OS:                                        Windows 10, fedora34
---

I have an issue using the .available() when reading from UART.
If I use it I simply receive garbage characters from time to time.

I've tested the 3 hardware UARTs of the ESP32 (using HardwareSerial.h) and a couple of SoftwareSerial s too. Same problem on both types. of UARTs.
I've tested a couple of ESP32s.
I've tested different sensors that talk via UART, and same issue.
I've tested the same code in an STM32 instead, and no error.

Let me explain with code.
CASE 1: If I do the following, I receive garbage from time to time in 'ch'
if (MySerial2.available()>0) {
      ch = MySerial2.read();
      ......
      }

but
CASE 2: If I replace the previous code by the following, everything works correctly:
ch = MySerial2.read();
if (ch>-1) {
   .....
   }

What is happening that if I test .read() against -1 works fine, and using .available() makes this error show up? 

Thank you

lbernstone, 2021-05-19:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L161
::read calls available.  The only way this can happen is if something else is taking elements off the queue in a different process.

pabloandresm, 2021-05-19:
> 
> 
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L161
> ::read calls available. The only way this can happen is if something else is taking elements off the queue in a different process.

I have no other processes. It is a basic arduino sketch. setup() to initialize the port, and loop() to read.

As I mentioned, if I avoid the ".available()" and only use the ".read()!=-1" it works fine. I did this workaround in my code, but it would be nice for me to understand why.

TheXtremeKing, 2021-06-26:
I might have a related/exactly the same problem. Using available() there will be a mangled byte at around every 100.000 bytes. The baudrate doesn't matter much, there will be rotten bytes. The way the data gets corrupted is very peculiar. There seems to get 1 byte inserted somewhere and further down the road a value of 0x01 gets added to a byte (which happened to always be the checksum byte of the serial packet i was receiving from a gps-receiver). 

However, after i've changed the data receive interrupt priority in esp32-hal-uart.c things look good now. If you want to try this as well, replace the library function with this code: 

```
void uartEnableInterrupt(uart_t* uart)
{
    UART_MUTEX_LOCK();
    uart->dev->conf1.rxfifo_full_thrhd = 112;
    uart->dev->conf1.rx_tout_thrhd = 2;
    uart->dev->conf1.rx_tout_en = 1;
    uart->dev->int_ena.rxfifo_full = 1;
    uart->dev->int_ena.frm_err = 1;
    uart->dev->int_ena.rxfifo_tout = 1;
    uart->dev->int_clr.val = 0xffffffff;

    esp_intr_alloc(UART_INTR_SOURCE(uart->num), (int)ESP_INTR_FLAG_IRAM|ESP_INTR_FLAG_HIGH, _uart_isr, NULL, &uart->intr_handle); //Added: ESP_INTR_FLAG_HIGH
    UART_MUTEX_UNLOCK();
}
```

VojtechBartoska, 2022-04-07:
Hello, can you please retest your sketch with v2.0.3-RC1? Thanks!

pabloandresm, 2022-04-07:
Sure. I ll do that this weekend.

VojtechBartoska, 2022-05-06:
I'm closing this issue as expired.

If needed, please reopen it.

Thanks for understanding.

closed 2022-05-06 by VojtechBartoska