espressif/arduino-esp32 #5201: Problem in routing on ESP's Board

Mohammadreza-Sorouriyan, 2021-05-21:
Hi, I'm working on a project where I need to set an esp32 as an access point, and some esp8266-01 as a client.

  Then i running a web server on each of the esp822 chips to get the sensor data in json format.  Now I have done this, when I connect to this network with my mobile and enter the IP of one of the esp8266s, i get response but with two special esp8266.

 And other esp8266's  there's get ip from access point but the web server doesn't send any response and timed out.  I even tried 7 type esp8266 models but it didn't work, but when I replace the access point with the home modem instead of esp32. All of esp8266 works perfectly!!

I think the esp32 can't handle and route the IP's. But here is a contradiction, why those two models of esp8266 working perfectly amd other no??

Thanks if anyone can guide me...

TheRealDJ, 2021-05-22:
SoftAP has a limit of 4 connections ... Is it possible that 4 devices were already connected?

FYI ... for code sections to look like this ...
```
code
code
code
```
Use the following syntax:

    ```
    code
    code
    code
    ```

Mohammadreza-Sorouriyan, 2021-05-22:
> SoftAP has a limit of 4 connections ... Is it possible that 4 devices were already connected?
> 
> FYI ... for code sections to look like this ...
> 
> ```
> code
> code
> code
> ```
> 
> Use the following syntax:
> 
> ```
> ```
> code
> code
> code
> ```
> ```

No. The devices connected to the AP are less than 4.

TheRealDJ, 2021-06-04:
Are you cleanly disconnecting from WiFi on the client side? I did some testing ... If I connect my phone to the WiFi SoftAP, then walk away from the ESP32, it seems to take some time (about 5 minutes if I remember correctly) for the ESP32 to indicate the client count back down to zero.

If I connect my to a different WiFi network or shut off WiFi, the ESP32 indicates a zero client count right away.

closed 2021-08-21 by stale[bot]