espressif/arduino-esp32 #5202: Cannot get ESP32-S2 SPIFFS Mounted with PlatformIO

pattobrien, 2021-05-22:
Hi everyone -

I've been trying to create and read from a SPIFFS partition using the 2.0.0-alpha release of Arduino-ESP32 using PlatformIO (MacOS BigSur) and an **ESP32-S2**-DevKitM-1 development kit from Espressif, but have had no luck after 24 hours of trying different configurations and sketches.

For additional context, our original firmware worked perfectly with an ESP32, and we've also used example sketches (e.g. LittleFS) with no avail on the esp32s2. I've been able to get our firmware compiling and flashing, and all Filesystem flashing seems to be working properly, and SerialMonitor outputs info so I know the rest of our program is being read from the flash IC properly. It simply seems to be unable to initialize / mount the SPIFFS partition (although we've also run into the same problem with FatFS).

However, when trying the same examples on Arduino IDE (2.0.0 beta), it works perfectly on the first try. Therefore, I'm fairly confident it has to do with either the FS toolchain I'm using in PlatformIO or the board configuration within PlatformIO itself.

### Platformio.ini Configuration:
```platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0-alpha1
platform = espressif32
board = esp32dev
framework = arduino
board_build.mcu = esp32s2
monitor_speed = 115200
```

### Debug Messages:
```
E (15984) SPIFFS: mount failed, -10025
[ 16439][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
[ 16439][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1 
```

And occasionally, I'll also see this WDT error with our own firmware (not using an example script) - I think it may be unrelated/insignificant to the bigger problem, but throwing it in here just in case:

```
E (357) SPIFFS: mount failed, -10025
[ 41081][E][esp32-hal-misc.c:125] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
E (16218) SPIFFS: mount failed, -10025
[ 56941][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
[ 56942][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
```

Has anyone actually had success with using SPIFFS on an ESP32-S2 using PlatformIO and Arduino framework? There are some trickles of comments here on the topic, but no concrete answers / solutions that have worked for us.

Any help would be greatly appreciated, as I'm getting close to balding with how much hair I'm ripping out from this singular issue :)

Patt

chegewara, 2021-05-22:
Yesterday i discovered it is platformio issue. The same problem is with FFat on S2 with pio.
The problem is with arduino platform, even if you will use plain esp-idf API to mount SPIFFS, thats why i am saying it is platformio problem:
- works with arduino ide,
- works with esp-idf,
- failed with arduino in platformio.

lbernstone, 2021-05-22:
Please link your issues. https://github.com/platformio/platform-espressif32/issues/546

pattobrien, 2021-05-22:
[](url)

pattobrien, 2021-05-22:
Duplicate of platformio/platform-espressif32#546

closed 2021-08-21 by stale[bot]

aliesultan7, 2024-08-05:
hy i have same problem too over here, I already change the SPIFFS.begin(true) but FS still not readed. pls help me.

`Error deserialize to WiFi config  Json
EmptyInput
null
SSID1:null
PASS1:null
SSID2:null
PASS2:null
`

this my enviroment
`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 921600

build_flags =
    -DARDUINO_RUNNING_CORE=1
    -DARDUINO_EVENT_RUNNING_CORE=1

board_build.f_cpu = 240000000L
board_build.flash_req = 80m
board_build.flash_mode = qio
board_build.flash_size = 4MB
board_build.partitions = default.csv
board_build.filesystem = littleFS
board_build.esp32_psram = disabled

build_type = release
debug_level = None

monitor_speed = 115200
monitor_filters = esp32_exception_decoder

platform_packages =
    toolchain-xtensa-esp32@12.2.0+20230208
    tool-mkspiffs@1.200.0
    

lib_deps = 
    plerup/EspSoftwareSerial@^8.2.0
    knolleary/PubSubClient @ ^2.8
    sebmillet/RF433any @ ^0.7.7
    bblanchon/ArduinoJson @ 6.17.3`
    
  pls anyone help me