espressif/arduino-esp32 #5204: Cant read Multi channel from internal ADC using I2S, no adc_set_i2s_data_pattern function

donnersm, 2021-05-22:
HARDWARE
Board: esp32 devkit
Version: 1.04 (10C4)
IDE Name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM Enabled: Yes
Upload Speed: 921600
Computer OS: Windows 10

DESCRIPTION

I am using the I2S interface to read out the internal ADC but it only works for 1 channel.
The function to initialize several channels at once is missing.
In the manufacturers documentation, it refers to this function fir using 2 channels:

```
 static const adc_i2s_pattern_t adc_i2s_pattern[] = {
        {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_0}
        {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_3}
};
i2s_set_adc_mode(ADC_UNIT_1, adc_i2s_pattern, sizeof(adc_i2s_pattern));
```
However this function is not implemented in the ARDUINO ESP library.
At this time, it is not possible to readout more then 1 channel in the same sketch.

SKETCH: (partly)

```
void setupI2S() {
  Serial.println("Configuring I2S...");
  esp_err_t err;

  // The I2S config as per the example
  const i2s_config_t i2s_config = { 
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
      .sample_rate = samplingFrequency,                        
      .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // could only get it to work with 32bits
      .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT, // although the SEL config should be left, it seems to transmit on right
      .communication_format = I2S_COMM_FORMAT_I2S_MSB,
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,     // Interrupt level 1
      .dma_buf_count = 2,                           // number of buffers
      .dma_buf_len = SAMPLEBLOCK,                     // samples per buffer
      .use_apll = false,
      .tx_desc_auto_clear = false,
      .fixed_mclk = 0
  };

err = adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_0); //step 1
  if (err != ESP_OK) {
    Serial.printf("Failed setting up adc channel: %d\n", err);
    while (true);
  }
//****
 err = adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_3); //step 1
  if (err != ESP_OK) {
    Serial.printf("Failed setting up adc channel: %d\n", err);
    while (true);
  }

  err = i2s_driver_install(I2S_NUM_0, &i2s_config,  0, NULL);  //step 2
  if (err != ESP_OK) {
    Serial.printf("Failed installing driver: %d\n", err);
    while (true);
  }
  err = i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_0);
    if (err != ESP_OK) {
    Serial.printf("Failed setting up adc mode: %d\n", err);
    while (true);
  }

  Serial.println("I2S driver installed.");
}

loop(){
i2s_adc_enable(I2S_NUM_0);
//int test=0;
//test=analogRead(4);
 //############ Step 1: read samples from the I2S Buffer ##################
 i2s_read(I2S_PORT, 
         (void*)samples, 
         sizeof(samples),  
         &bytesRead,   // workaround This is the actual buffer size last half will be empty but why?
         portMAX_DELAY); // no timeout

   if (bytesRead != sizeof(samples))
        {
            Serial.printf("Could only read %u bytes of %u in FillBufferI2S()\n", bytesRead, sizeof(samples));
           // return;
        }
// now all data is in array samples[]
// but only for channel 0 , data from other channels is empty
}

```
Debug messages:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Setting up Audio Input I2S
Configuring I2S...
I2S driver installed.
Audio input setup completed

lbernstone, 2021-05-22:
You mean this documentation?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#_CPPv416i2s_set_adc_mode10adc_unit_t14adc1_channel_t

donnersm, 2021-05-22:
Yes,
They also did an update in there own driver to enable multi channel I2S.
See this post:
https://github.com/espressif/esp-idf/pull/1991/files

lbernstone, 2021-05-22:
I believe that only pull requests that have been merged into the code will be included in arduino.

donnersm, 2021-05-22:
Any idea on a timeframe? When can i expect it to be included in the arduino version?

lbernstone, 2021-05-22:
Anything is possible if you write your own code.

closed 2021-08-21 by stale[bot]

a442509097, 2022-11-23:
```
size_t bytesRead = 0;
i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_6);
i2s_read(I2S_NUM_0, (void*)samples, sizeof(samples), &bytesRead, portMAX_DELAY); 
FFT01 handle

-------------------------------------------------------

bytesRead = 0;
i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_7);
i2s_read(I2S_NUM_0, (void*)samples02, sizeof(samples02), &bytesRead, portMAX_DELAY);
FFT02 handle
```
The difference is 3us, but two FFT processing will take a lot of time.