espressif/arduino-esp32 #5205: Non-blocking ADC functions gone

Triangulix, 2021-05-22:
Hi

I just updated from 1.0.4 to 1.0.6 and realized that the non-blocking ADC functions adcStart, adcBusy and adcEnd have been removed.

How should we now use the ADC without blocking the program flow while the ADC is working?

Triangulix, 2021-07-23:
I am now stuck on version 1.0.4, the last version with non-broken non-blocking ADC. Great.

Triangulix, 2021-09-22:
Any news on this topic?

VojtechBartoska, 2022-04-07:
Hello @Triangulix , are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? ADC was refactored lately.

Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

Triangulix, 2022-04-07:
@VojtechBartoska Since no-one replied for so long I changed the design and the ADCs are not used anymore.

sumitmamoria, 2022-11-18:
@VojtechBartoska I can confirm that the issue is there in the latest version as well.

VojtechBartoska, 2022-11-21:
@sumitmamoria Can you please provide a sketch for reproducing the issue? Thanks!

positron96, 2023-02-14:
The functions are still gone and they would really be useful to me.

@VojtechBartoska the issue is with missing functionality and functions, I don't think a reproducing sketch is possible. You can follow the history here: https://github.com/espressif/arduino-esp32/issues/220

VojtechBartoska, 2023-02-15:
@P-R-O-C-H-Y Can you please triage this request? Thanks!

lbernstone, 2023-07-26:
ADC was refactored to allow the functions to work on all variants (which have different registers for these peripherals).  If you still want this functionality on an esp32, you can add the functions directly to your code [from 1.0.4](https://github.com/espressif/arduino-esp32/blob/1.0.4/cores/esp32/esp32-hal-adc.c#L159-L217)

P-R-O-C-H-Y, 2023-08-16:
If there is anyone who is already trying the new development version 3.0 using IDF-5.1 which is under `esp-idf-v5.1-libs` branch, you can give a try to the new ADC Continuous mode in PR #8490. Any feedback would be welcome. Thanks

VojtechBartoska, 2023-12-21:
Closing this as solved by above mentioned PR.

closed 2023-12-21 by VojtechBartoska