espressif/arduino-esp32 #5209: First factory/ota partition is hard coded at 0x10000

wijnsema, 2021-05-23:
When you use a partition table with a larger nvs (or other) partition at the beginning and thereby exceed the 0x10000 for the first factory/ota partition, the ESP32 will run into a reboot loop. 

More information can be found at this issue at esp32.com:
[Unable to enlarge nvs partition](https://esp32.com/viewtopic.php?f=13&t=20966)

If needed I can makes steps to reproduce, for now I hope the information at esp32.com will do.

Ronald Wijnsema

lbernstone, 2021-05-26:
https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L164
```
tools.esptool_py.upload.pattern_args=--chip {build.mcu} --port "{serial.port}" --baud {upload.speed} {upload.flags} --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" {build.bootloader_addr} "{build.path}/{build.project_name}.bootloader.bin" 0x10000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin" {upload.extra_flags}
```

wijnsema, 2021-05-26:
Maybe the title suggested that I'm more knowledgeable then I am, because I have no idea what this means...

bertmelis, 2021-05-27:
ON platformio, you might also want to take a look here: https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables

bertmelis, 2021-05-27:
In the upload pattern the 0x10000 is hardcoded. You can manually upload using esptool though.

closed 2021-08-21 by stale[bot]