espressif/arduino-esp32 #5215: Trying to log all vprintf() to a file instead of serial monitor including crash logs

zekageri, 2021-05-27:
### Hi! 
I'am trying to redirect all system logs to my own function and write it to a file but it does not redirect everything, just a small number of logs.

```C
int _log_vprintf(const char *fmt, va_list args) {
  File f = LITTLEFS.open(FSLOG, "a");
  if(f){
    //f.println(fmt); // It isn't formatted, maybe with function vprintf() it will be
    f.println( vprintf(fmt, args) );
    f.close();
  }
  return vprintf(fmt, args);
}

void setup(){
esp_log_set_vprintf(_log_vprintf);
}

void loop(){
}
```

WereCatf, 2021-05-31:
`f.println( vprintf(fmt, args) );`

vprintf() doesn't return a string, but instead the number of characters it printed, so that's not doing what you intend. Look into e.g. vasprintf() and f.print() the buffer it allocated into your file (and don't forget to free() the allocated buffer)

Besides which, this issue-tracker is for bugs and issues with ESP-IDF, not for general "how do I do this" questions.

zekageri, 2021-06-01:
### Thanks for the clarification about the `vprintf()`.

**Do you mean something like this?**

```C
int _log_vprintf(const char *fmt, va_list args) {
  size_t len;
  File f = LITTLEFS.open(FSLOG, "a");
  if(f){
    // Format the string
    char *log_buffer; 
    len = vasprintf(&log_buffer, fmt, args);

    f.println("*************************************");
    f.print("       ");
    f.println(Time.GetCurrentTime()); // It's a custom Time class. Fetches the time from an ntp server and returns a String.
    f.println(log_buffer);
    f.println("*************************************");
    f.println(" ");
    f.close();
  }
  return len;
}
```

The thing is that this function does not log the crash dump and things like that.

closed 2021-08-21 by stale[bot]