espressif/arduino-esp32 #5220: WiFiMulti connection fails with 2.0.0 / IDF 4.4

mike-s123, 2021-05-28:
WiFiMulti fails to connect with 2.0.0 / IDF 4.4.

Broken with (5502879a v2.0.0..., Apr 5, 2021)
Works with (2ee66b54 IDF release/v3.3, Mar 20, 2021) 

ESP32 WROVER

Pretty straightforward, recreated using examples:
SimpleWiFiServer.ino (just uses basic WiFi.h) example connects and works.
WiFiMulti.ino example fails to connect.

Seems newer tries to set static IP to 0.0.0.0 instead of using DHCP?

Broken verbose core debug output:

```
[  6400][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: xxxx SSID: xxxx Channel: 6 (-43) 
[  6425][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 11438][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
[ 11438][D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
[ 11440][D][WiFiMulti.cpp:198] run(): [WIFI] start scan
WiFi not connected!
```

Working verbose core debug output:

```
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: xxxx SSID: xxxx Channel: 6 (-42)
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.168.155, MASK: 255.255.255.0, GW: 192.168.168.2
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: xxxx
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.168.155
```

mike-s123, 2021-08-21:
Anti-stale.

brahma-dev, 2021-09-16:
@mike-s123 Did you manage to make it work ?

brahma-dev, 2021-09-22:
I have tried several combinations of idfv4.4 and arduino-es32v2.0.0 and this problem persists. Can someone else please confirm it's happening for them too, and it's not something specific to my setup.

VojtechBartoska, 2022-04-07:
Hello, can you please retest this with v2.0.3-RC1?

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

mike-s123, 2022-04-22:
Seriously???

After almost a year of total inaction, you expect a response in two weeks?

VojtechBartoska, 2022-04-22:
@mike-s123 Sorry for not answering this earlier but we weren't able to handle all of the issues before due to the Team size limitation. 

To keep Issue tracker manageable we are now doing some cleanup with closing outdated issues, this is because we did a lot of changes recently and older issues needs to be retested.

With that goes hand in hand that we are not able to test every single issue, from you comment I understand that issue still occurs so I am reopening it.

I hope it make sense for you and we can cooperate together to solve your problem.

mike-s123, 2022-04-22:
I can test when I return from vacation.

SuGlider, 2022-04-23:
@mike-s123 @VojtechBartoska 

I just tested WiFiMulti.ino example using 2 SSIDs and Arduino Core 2.0.2 and 2.0.3-RC1.
In both cases it worked fine and connected to the strongest signal WiFi AP.

I tested it by unplugging the AP where ESP32 was connected and it switched to the other AP right away, connecting with no problems.

@mike-s123  - please let us know if you also are able to run it with no problems.

mike-s123, 2022-04-25:
@VojtechBartoska 

I can confirm that the WiFiMulti library, using v2.0.3-RC1, is now working with the code it previously failed with. The example now works, too, as stated by @SuGlider 

As far as I'm concerned, this issue may now be closed. Thank you.

closed 2022-04-25 by mike-s123

VojtechBartoska, 2022-04-26:
Thanks for clarification @mike-s123 and also @SuGlider!