espressif/arduino-esp32 #5221: BluetoothSerial fails to accept connection from phone (but not from Linux PC) after using, then shutting down WiFi

avillacis, 2021-05-28:
### Hardware:
Board:							YUBOX ESP32 WROOM (compatible with ESP32 Dev Module)
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Fedora Linux x86_64

### Description:
On a project my team is developing, the ESP32 is intended to work in two modes. In one of the modes, the ESP32 enables WiFi and provides a webserver for configuration and firmware updates. In the other mode, the ESP32 shuts down WiFi and enables the Bluetooth support (using BluetoothSerial) in order to send out sensor measurements to whoever connects. The idea was to switch between the two modes as necessary, with Bluetooth being the default mode, and using the WiFi webserver for calibration, then switching back to Bluetooth mode (through a HTTP POST request) to resume normal operation. The current mode (WiFi or Bluetooth) is persisted through NVRAM so that next boot runs in the last mode.

The project uses a fork of ESPAsyncWebServer and AsyncTCPSock (my own library) for the webserver, plus YUBOX Now for the available operations including firmware upload.

If the ESP32 has been paired with either a Linux PC (Bluetooth dongle and bluez software stack), or an Android phone with Bluetooth support, and the ESP32 has booted in Bluetooth mode (with WiFi off and not initialized), the BluetoothSerial support works as intended. Both the phone (using "Serial Bluetooth Terminal" by "Kai Morich") and the Linux PC can connect to the Bluetooth serial port exposed by the ESP32, in any order, and can freely disconnect.

The problem is that, after booting with WiFi enabled, or if switching to WiFi from Bluetooth mode, and then switching back, the phone (and ONLY the phone) is no longer able to connect to the Bluetooth serial port until the next ESP32 reboot. Even if the device is already paired, the serial terminal app on the phone just reports a failure to connect. If the ESP32 is rebooted, the phone is once again able to connect (booting in Bluetooth mode). So far this is the only workaround that works for our project - rebooting immediately after switching from WiFi to Bluetooth.

The weirdest thing is that the Fedora Linux PC is *not* affected by this. Even when the phone fails to connect to the ESP32, the PC remains able to, and can receive the sensor measurement data as normal. We have seen this behavior in two different Android phones (both fail to connect after wifi->bluetooth transition), and two different Linux PCs (both remain able to connect after wifi->bluetooth transition).

I have tried erasing the pairing (based on the bt_remove_paired_devices example from BluetoothSerial) after the wifi->bluetooth transition, but it has no effect - the phone remains unable to connect, even after re-pairing, and the Linux PC remains able to connect after re-pairing.

I have compiled the project with the verbose setting, and when trying to connect to the phone in the failing scenario, there is simply no activity seen, unlike the Linux connection attempt, which shows messages as normal:
```
[I][yubox-acuacultura-temp-ph.ino:144] yuboxActualizarEstadoBT(): YUBOX Framework cede control de WiFi
[I][YuboxWiFiClass.cpp:109] releaseControlOfWiFi(): Cediendo control del WiFi (WiFi OFF)...
[I][YuboxWiFiClass.cpp:115] releaseControlOfWiFi(): Desconectando del WiFi (STA)...
[I][YuboxWiFiClass.cpp:122] releaseControlOfWiFi(): Desconectando del WiFi (AP)...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 15 - AP_STOP
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 15 - AP_STOP
[I][yubox-acuacultura-temp-ph.ino:148] yuboxActualizarEstadoBT(): Iniciando Bluetooth...
[I][BluetoothSerial.cpp:571] _init_bt(): device name set
[I][yubox-acuacultura-temp-ph.ino:153] yuboxActualizarEstadoBT(): Transición YUBOX Framework-->BlueTooth terminada.
[I][BluetoothSerial.cpp:243] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:246] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
[I][BluetoothSerial.cpp:347] esp_spp_cb(): ESP_SPP_START_EVT
[I][BluetoothSerial.cpp:254] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT: 0
[V][BluetoothSerial.cpp:168] _spp_send_buffer(): SPP Write 43
[V][BluetoothSerial.cpp:299] esp_spp_cb(): ESP_SPP_WRITE_EVT: 43 
[V][BluetoothSerial.cpp:168] _spp_send_buffer(): SPP Write 43
[V][BluetoothSerial.cpp:299] esp_spp_cb(): ESP_SPP_WRITE_EVT: 43 
```

The following might or might not be relevant: I am the author of the AsyncTCPSock library, intended to be a drop-in replacement for the AsyncTCP library. Just for a test, I switched the project to use AsyncTCP instead of my library. When booting in Bluetooth mode, the project behaves the same as before. However, after switching to WiFi mode and back, either the phone or the Linux PC triggers the following crash and reboot:

```
0x4008f6ac: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008f929: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400908dc: vQueueDelete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1796
0x40142e71: osi_sem_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/semaphore.c line 75
0x40172abb: fixed_queue_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/fixed_queue.c line 88
0x40172b24: fixed_queue_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/fixed_queue.c line 67
0x40165fe8: l2cu_allocate_ccb at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_utils.c line 1539
0x40163f94: process_l2cap_cmd at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c line 451
0x40164919: l2c_rcv_acl_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c line 227
0x40159d88: btu_hci_msg_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c line 137
0x40159ff2: btu_task_thread_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c line 229
0x4009099e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
When using my library instead, the crash does not happen.

Wild speculation ahead:
I am starting to suspect some interaction between an incomplete shutdown of the WiFi stack, and the Bluetooth stack, that is only triggered by a phone trying to connect. My code ends up doing WiFi.mode(WIFI_OFF) as part of the handover but the problem still persists.

avillacis, 2021-08-21:
No change in 1.0.6, so I assume still present.

VojtechBartoska, 2022-03-30:
Hello @avillacis , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-08:
Thanks for testing, marking as closed.

closed 2022-04-08 by VojtechBartoska