espressif/arduino-esp32 #5225: ESP32 WebServer on Ethernet, communicate between devices

BoYi0126, 2021-05-31:
### **Hardware:**
Board: ESP32-Ethernet-Kit_A_V1.1
PHY: IP101 (On the board)
Core Installation version: esp32 1.0.6 (installed from Board Manager)
IDE name: Arduino IDE 1.8.13
Upload Speed: 921600
Computer OS: Windows 10

### **Hardware wried:**
    ----------                  ------------
    |        | <--------------- | Computer |
    |        |                  ------------
    | Router |
    |        |                  ---------------
    |        | <--------------- | ESP32 Board |
    ----------                  ---------------

### **Purpose:**
Build a web server on the esp32 board, and communicate to "other device"(computer) by **ethernet**.

### **Description:**
Hello everyone,
I want to send data between devices and esp32 board by ethernet, and I can setup the IP of web server by the library - **ETH.h**.
But when i wired the cable to computer and boards, I cannot connect to the web server after key in IP in the browser.

### **Sketch:** 
```
#include <ETH.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <ESPmDNS.h>

#define SOFT_AP

#define ETH_ADDR        1
#define ETH_POWER_PIN   5
#define ETH_MDC_PIN     23
#define ETH_MDIO_PIN    18
#define ETH_TYPE        ETH_PHY_IP101

// ETH variables
IPAddress ip(192, 168, 1, 11);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns1(192, 168, 1, 10);
IPAddress dns2(192, 168, 1, 20);

// WiFi variables
const char *apSsid = "ETH_AP";

#define PORT 80
AsyncWebServer asyncWebServer(PORT);

static bool eth_connected = false;
    
void setupETH()
{
  // Start the ethernet
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE);
  ETH.config(ip, gateway, subnet);

  Serial.print("MAC: ");
  Serial.println(ETH.macAddress());
}

void notFound(AsyncWebServerRequest *request)
{
  request->send(404, "text/plain", "Not found");
}

void setHtmlCallback()
{
  asyncWebServer.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    Serial.println("Run call back");
    request->send(200, "text/plain", "Hello world");
  });
  
  asyncWebServer.onNotFound(notFound);
}

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void setup()
{
  Serial.begin(115200);

  WiFi.onEvent(WiFiEvent);
  setupETH();

#ifdef SOFT_AP
  WiFi.mode(WIFI_AP);
  WiFi.softAP(apSsid);
  WiFi.softAPConfig(ip, gateway, subnet);
#endif
  
  // Start the Ethernet web server
  setHtmlCallback();
  asyncWebServer.begin();
}

void loop()
{
}
```

me-no-dev, 2021-06-03:
Hi @BoYi0126 :) What is the actual issue that you are having? Code looks OK

BoYi0126, 2021-06-03:
Hi @me-no-dev, I want to connect the web server by Ethernet, but I cannot connect to the IP on the browser, and I cannot ping it by "cmd", so I do not know the function "ETH.config(ip, gateway, subnet);" is really work. 
If my hardware wired was wrong, or the code cannot to achieve the purpose, please tell me what should I need to do. 
Thank you very very much! :)

me-no-dev, 2021-06-03:
I will suggest a couple of things:
1. Enable debug for the board in the Arduino IDE menu
2. Provide logs after that
3. If you have DHCP server there, try without ETH.config
4. Give the 2.0.0-beta a shot as well (maybe roll back to 1.0.5 as a test also). There were some reports about issues with ETH on 1.0.6

BoYi0126, 2021-06-04:
Hi @me-no-dev 
Thanks for your advise, I try it and get some result
    1. & 2. I do not know what it means, can you tell me more detail step, thanks.
    3. The router likes a DHCP server, so I remove the "ETH.config()", the ip becomes (0, 0, 0, 
        0), but if connect the "Internet" to the router, esp32 get real ip, but it not my goal.
    4. I try, it's not work.
    
I have to say it again, thank you very very much, although I still do not find the way to solve this problem.

me-no-dev, 2021-06-04:
I will try to find a board with the ethernet chip and give it a go :) 
On 1 and 2, how do you compile the sketch? Are you using Arduino IDE or PlatformIO or what?
Arduino IDE has a configurable option in it's board menu to set Debug to particular level (Verbose -> Error). There are other options to do it for PIO, but I need to know how you are compiling

chegewara, 2021-06-04:
> 3. The router likes a DHCP server, so I remove the "ETH.config()", the ip becomes (0, 0, 0,
> 0), but if connect the "Internet" to the router, esp32 get real ip, but it not my goal.

I am assuming by "real IP" you mean WAN IP, but it is not possible, because it would be a case with all devices, including PC, being connected to that router, and this is not how DHCP works.

BoYi0126, 2021-06-04:
@me-no-dev 
I compile the sketch by Arduino IDE
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\002\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\002\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\002\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=0403_6010 -ide-version=10813 -build-path C:\Users\002\AppData\Local\Temp\arduino_build_6474 -warnings=none -build-cache C:\Users\002\AppData\Local\Temp\arduino_cache_976635 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose D:\Project\Arduino\Practice\ESP32_EthernetWebServer_2\ESP32_EthernetWebServer_2.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\002\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\002\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\002\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=0403_6010 -ide-version=10813 -build-path C:\Users\002\AppData\Local\Temp\arduino_build_6474 -warnings=none -build-cache C:\Users\002\AppData\Local\Temp\arduino_cache_976635 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\002\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose D:\Project\Arduino\Practice\ESP32_EthernetWebServer_2\ESP32_EthernetWebServer_2.ino
Using board 'esp32wrover' from platform in folder: C:\Users\002\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5
Using core 'esp32' from platform in folder: C:\Users\002\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5
cmd /c if exist "D:\\Project\\Arduino\\Practice\\ESP32_EthernetWebServer_2\\partitions.csv" copy /y "D:\\Project\\Arduino\\Practice\\ESP32_EthernetWebServer_2\\partitions.csv" "C:\\Users\\002\\AppData\\Local\\Temp\\arduino_build_6474\\partitions.csv"
cmd /c if not exist "C:\\Users\\002\\AppData\\Local\\Temp\\arduino_build_6474\\partitions.csv" copy "C:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5\\tools\\partitions\\default.csv" "C:\\Users\\002\\AppData\\Local\\Temp\\arduino_build_6474\\partitions.csv"
Detecting libraries used...
"C:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/config" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/app_trace" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/app_update" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/asio" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/bootloader_support" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/bt" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/coap" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/console" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/driver" "-IC:\\Users\\002\\AppData\\Local\\Arduino15\ [... cut ...]
Writing at 0x0006c000... (85 %)
Writing at 0x00070000... (89 %)
Writing at 0x00074000... (92 %)
Writing at 0x00078000... (96 %)
Writing at 0x0007c000... (100 %)
Wrote 827664 bytes (451087 compressed) at 0x00010000 in 7.1 seconds (effective 932.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1445.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

BoYi0126, 2021-06-04:
@chegewara  I thought you are right, I think the router like a DHCP server because it was auto assign IP to all device which connect to the router

me-no-dev, 2021-06-04:
[image]

@BoYi0126 set the debug level to verbose and provide the logs from serial after the device is flashed.

BoYi0126, 2021-06-04:
@me-no-dev
Thanks for point my wrong action...

 [D][esp32-hal-psram.c:60] psramInit(): PSRAM enabled
MAC: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 21 - ETH_START
ETH Started
84:0D:8E:D3:64:6F
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START

me-no-dev, 2021-06-04:
I do not see ETH Connected. You sure that the cable is OK?

BoYi0126, 2021-06-04:
@me-no-dev 
[D][esp32-hal-psram.c:60] psramInit(): PSRAM enabled
MAC: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 21 - ETH_START
ETH Started
84:0D:8E:D3:64:6F
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 23 - ETH_CONNECTED
ETH Connected

me-no-dev, 2021-06-04:
everything looks good

BoYi0126, 2021-06-04:
@me-no-dev 
yap, but if not use "ETH.config", the ip is (0, 0, 0, 0)
if I use "ETH.config" to configure IP, I cannot ping the ip by cmd...
that was my confuse.

me-no-dev, 2021-06-04:
ca you print the IP one second after "ETH Connected"? As I said, all look fine based on the logs. Could maybe you router be blocking ICMP or something else to not be OK in your network setup?

BoYi0126, 2021-06-04:
@me-no-dev

--------------------------------- 
This is comment the line of ETH.config

[D][esp32-hal-psram.c:60] psramInit(): PSRAM enabled
MAC: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 21 - ETH_START
ETH Started
84:0D:8E:D3:64:6F
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 23 - ETH_CONNECTED
ETH Connected
ETH IP: 0.0.0.0

--------------------------------- 
This is use ETH.config

[D][esp32-hal-psram.c:60] psramInit(): PSRAM enabled
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 21 - ETH_START
ETH Started
MAC: 84:0D:8E:D3:64:6F
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 25 - ETH_GOT_IP
[D][WiFiGeneric.cpp:455] _eventCallback(): ETH IP: 192.168.1.11, MASK: 255.255.255.0, GW: 192.168.1.1
84:0D:8E:D3:64:6F, IPv4: 192.168.1.11, FULL_DUPLEX, 100Mbps
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 25 - ETH_GOT_IP
[D][WiFiGeneric.cpp:455] _eventCallback(): ETH IP: 192.168.1.11, MASK: 255.255.255.0, GW: 192.168.1.1
84:0D:8E:D3:64:6F, IPv4: 192.168.1.11, FULL_DUPLEX, 100Mbps

me-no-dev, 2021-06-04:
everything looks as it should. I can not see a problem on the ESP side.

lbernstone, 2021-06-04:
You are certain that IP addressing is correct?  You can ping 192.168.1.1 (the router) from your PC?

BoYi0126, 2021-06-07:
> You are certain that IP addressing is correct? You can ping 192.168.1.1 (the router) from your PC?

My router's IP is 192.168.1.13, and I can ping it.
I use the jump-wired-cable, connect device to device, and it also cannot ping the 192.168.1.11(ESP32 board).

me-no-dev, 2021-06-08:
You should have your computer and the esp connected to the router. You might need a special cable if you want to connect your esp to your computer directly. If your router is .11, why is it set to .1 in ETH.config()? Did you try instead of pinging to directly open the web server on the esp?

BoYi0126, 2021-06-09:
@me-no-dev @lbernstone 
Thanks guys, I solve the problem, because my hardware setting.
I really thanks for your advice, very thanks.
But I have other question, I will pull a new issue, thanks!

closed 2021-06-09 by me-no-dev