espressif/arduino-esp32 #5233: "Guru Meditation Er" - related to http request?

Magerquark, 2021-06-01:
Hello folks I am currently trying to develop a wifi RFID module based on an ESP32. 
Currently I am using a ESP32 Devkit to test the software. I only make a http request like every 200ms to a wifi relais to check the status if its powered on or not.
After some time like 50 minutes the ESP32 just crashes (LED's on the GIPO pins go off) and the programm just stops.

The problem is I dont get a error message which I could decompile, the last stuff I can read in the Serial Monitor is:
"Guru Meditation" or somtimes "Guru Meditation Er"

I first thought that its a problem of the power supply of the ESP32 but I cant reproduce this failure on high power spikes (for example when I put a Chipcard on the RFID module) it just randomly crashes after some time.
The programm probably crashes at the line "if (http.connected()) http.end();"

Does anybody of you has an idea what I can do to locate the exact problem?

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			I dont know
IDE name:					       Arduino IDE? 
Flash Frequency:				       80Mhz
PSRAM enabled:                                     no 
Upload Speed:				       512000
Computer OS:                                        Windows 10

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <SPI.h>
#include <MFRC522.h>
#include <WiFi.h>
#include <HTTPClient.h>

//Basicly its this part of the code that is from interest I think:

void ShellyAbfragen(){
     if(WiFi.status() != WL_CONNECTED)connectWifi();
      
      // Send HTTP GET request
      http.begin(shelly);
      int httpResponseCode = http.GET();
      
      if (httpResponseCode>0) {
        Serial.print("HTTP Response code: ");
        Serial.println(httpResponseCode);
        String payload = http.getString();
        Serial.println(payload);
        Serial.println("Verbindnung mit Shelly erfolgreich!");
      }
      else {
        Serial.print("Error code: ");
        Serial.println(httpResponseCode);
      }
      // Free resources
      if (http.connected()) http.end();
      //http.end();
}

lbernstone, 2021-06-01:
Most likely you have a memory leak.  Put this at the top of your code and then call `MEMCK;` in your loop function.
```
const char* fmtMemCk = "Free: %d\tMaxAlloc: %d\t PSFree: %d\n";
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())
```

Magerquark, 2021-06-01:
> Most likely you have a memory leak. Put this at the top of your code and then call `MEMCK;` in your loop function.
> 
> ```
> const char* fmtMemCk = "Free: %d\tMaxAlloc: %d\t PSFree: %d\n";
> #define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())
> ```

20:40:31.294 -> Free: 280424	MaxAlloc: 113792	 PSFree: 0
20:40:31.341 -> running 
20:40:31.480 -> HTTP Response code: 200
20:40:31.528 -> {"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"source":"input"}
20:40:31.622 -> Verbindnung mit Shelly erfolgreich!
20:40:31.669 -> Guru Meditation E

Doesnt look like a memory leak since the free memory was constant on 280424.

do you have some other idea?

lbernstone, 2021-06-01:
Put the MEMCK wherever you are passing around String objects between functions or allocating large arrays.  This is a problem in your code.  You are going to have to troubleshoot it.

Magerquark, 2021-06-02:
> Put the MEMCK wherever you are passing around String objects between functions or allocating large arrays. This is a problem in your code. You are going to have to troubleshoot it.

 I allready did as I mentionend in the last comment or do I understand you wrong?
The free memory was constant and then it suddenly crashed.

I uploaded the Code with "Core Debug level: Debug" activated and now the Code is running fine for like 10 hours, even tho I cant read the http request in the serial monitor. 

best regards 
Alexander

Magerquark, 2021-06-02:
Okay so I digged a little bit deeper:
(Im not that good in programming and debugging so I hope I understand this error right)

The ESP32 crashes when trying to check:
if (http.connected()) http.end();

which refers to the HTTPClient.cpp line 406 which is:

ppescher, 2021-07-20:
It looks like it could be related...
Did you try the patch I submitted [there](https://github.com/espressif/arduino-esp32/issues/3659#issuecomment-844181902)?
Or see the merge commit #5197

VojtechBartoska, 2022-03-30:
Hello @Magerquark, still relevant? Thanks for your answer!

VojtechBartoska, 2022-04-13:
Hello, 

as there was no answer in last 14 days, I'm **closing the issue as expired** to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-13 by VojtechBartoska