espressif/arduino-esp32 #5235: ESP_Clinet can't write more than 20 bytes to ESP_Server_multiconnect but Renesas GATT Browser can

KorochanOne, 2021-06-02:
I want to writeValue up to 512 but it turncat by 20 bytes.
but Renesas GATT Browser can write more than 20 bytes. I confirmed up to 50 bytes.
So I think it has some problem in esp_err_t esp_ble_gattc_write_char()

I tried   BLEDevice::setMTU(517) both server and client but no effect at all.

chegewara, 2021-06-02:
I am using BLE library in dozens projects and never had any issue with write longer values, including OTA over BLE which i am doing with 500-512 bytes for each packet.

KorochanOne, 2021-06-03:
Okay it is a good information. Thanks.
I work this more to solve my problem.

KorochanOne, 2021-06-03:
I got workaround.
to send more than 20 byte, it is required response parameter false to true.
Since response is implicit to false, it is required when send more than 20 byte.
I found eventually I write writeValue((byte*)pointer,len); for writeValue(uint8_t* data, size_t length, bool response = false);
but it consider as 
void        writeValue(std::string newValue, bool response = false);
 and respnse became true by len

now writeValue((uint8_t*)pointer,len,true);   works fine.
Thank you.

If you know the meaning of response, please let me know.

Thanks

KorochanOne, 2021-06-03:
So, when I send 20 byte, it takes 0.03ms, 10 times 0.3 ms
but I send 200 bytes, it takes 0.8 ms
it apparently send 20 bytes each by MTU limitation.
it looks   BLEDevice::setMTU(517); doesn't work.
Please let me know how to change MTU and effect it.

chegewara, 2021-06-03:
To change MTU both BLE devices has to change it. If you do it only in esp32 then it wont work.
There is another thing and maybe it is confusing. Limit to 20 bytes is only for notify/indicate. Read and write does not care about length, because it will be performed in multiple packets. Of course with higher MTU there is less overhead and transmission is faster.

> So, when I send 20 byte, it takes 0.03ms, 10 times 0.3 ms

I have no idea how you achieved transaction faster than 1 ms. I could achieve sending notifications, which does not require ACK, every 3-5ms. Read/write request require ACK from per device (usually) which takes twice as long or longer.

KorochanOne, 2021-06-03:
Hello chegewara san,
Thanks
Yes I'm confuse. your comment is always help me.
As you point out, ms is my typo. It is just s.
Now I understood that only notify/indicate is limited.
I'm using write.

>So, when I send 20 byte, it takes 0.03s, 10 times 0.3 s, 200 byte takes 0.8 s
writeValue((uint8_t*)buf,20);  0.03s
for(i=0;i<10;i++){
  writeValue((uint8_t*)buf,20); 0.03s
}
0.3s
writeValue((uint8_t*)buf,200);  0.8s 

I expected write 200 is faster than write 20X10 but not.
so I guess it still use 20 byte payload to send.
Is there any way to use large MTU.  Is there requested MTU value each other when connect?

KorochanOne, 2021-06-04:
ＭTU number didn't effect maximum payload size. so, it seems major effect for transfer speed.
I thought it is similar to payload size of ethernet but BLE use almost fix size for physically transfer.
I need to find another way to speed up transfer rate near to 10Kbyte/s.
Am I right?

chegewara, 2021-06-04:
I really dont know how to help you more, because i dont know where you can have error in code.
I know library works good, at least in that part. Like i said, i am using write to characteristic extensively in OTA over BLE and i am having very good results/performance. Always with MTU 517 and custom android apps.

h2zero, 2021-06-05:
The data transfer rate depends on the MTU size and the connection interval. The maximum MTU is 517 which provides a maximum 512 bytes of data (5 bytes are for the header) and the minimum interval is 7.5ms. 

The fastest you will be able to transfer data is 512 bytes per 7.5ms, but that is only possible if both devices are capable of a 517 byte MTU. The client device also needs to set the connection interval/window to 7.5ms to ensure maximum data transfer rate.

If writing the value without response you should be able to achieve over 100KB/s (theoretically), with response it would be half the rate.

KorochanOne, 2021-06-07:
Thanks both of you.
I just want to create working example which communicate 100KB/s speed.
(communicate two ESP32 with BLE peripheral and central)

I got it from nFR52840 chip Arduino sample.
reviewing these with sniffer, I try to learn more about BLE and find a clue.
If you have some advice, please let me know.

sandmen, 2021-06-13:
Check the version 1.0.4, in board manager, a simple 
BLEDevice::setMTU(510); is working for me, 
After 1.0.4 is not longer working.

KorochanOne, 2021-06-14:
Hello Sandmen,
This is very helpful information, I will check this.

this is using ESP32-arduino 1.0.6
I tested this matter with nRF52840 arduino (Adafruite).   (nRFp eriferal, nFRc entral)
Both are using nFR52840 feather express and using central_throughput and periferal_throughput which provided as example.
Its can change MTU 23 to 247. When central connected, peripheral requests MTU which should be use.

at 23. it send 20 and receive 20.
at 247. it send 244 and receive 244.
at 517. it send 20 and receive 20.
Notification to central,  12KB/s at MTU 20, 157KB/s at MTU 244.
Write to peripheral,  1KB/s at MTU 20, 12KB/s at MTU 244.

change central from nFRc to ESP32c.
at 23. it send 20 and receive 20.
at 247. it send 244 and receive 244.
at 517. it send 20 and receive 20.
Notification to central,  14KB/s at MTU 20, 64KB/s at MTU 244.
Write to peripheral,  14KB/s at MTU 20, 64KB/s at MTU 244.

change periferal from nFRp to ESP32p, central is nFRc
 add follows when central connect 
pServer->updatePeerMTU(pServer->getConnId(), 517);
int mtu = pServer->getPeerMTU(pServer->getConnId());
Serial.printf("connected %d\r\n", mtu);
this returns proper number but seems no effect.

at 23. it send 20 and receive 20.
at 247. it send 20 and receive 20.
at 517. it send 20 and receive 20.
Notification to central,  17KB/s
Write to peripheral,  1KB/s

Final ESP32p and ESP32c
at 23. it send 20 and receive 20.
at 247. it send 20 and receive 20.
at 517. it send 20 and receive 20.
Notification to central,  17KB/s
Write to peripheral,  15KB/s

This result suggests some problem in peripheral side in ESP32.

I'm going to do the same with 1.0.4 and report but need a time. 

My goal is send and receive data near 30KB/s using nFRc and ESP32p

ESP32p  peripheral
ESP32c  central

KorochanOne, 2021-06-14:
Hello Sandmen,
You are right. When I down to 1.0.4 and it works.
the same source works properly.
now
ESP32p and ESP32c
at 23. it send 20 and receive 20.
at 247. it send 244 and receive 244.
at 517. it send 512 and receive 512.
Notification to central, 48KB/s
Write to peripheral, 48KB/s

also
ESP32p and nFR52c
at 23. it send 20 and receive 20.
at 247. it send 244 and receive 244.
at 517. it send 20 and receive 20.  (nFR limit)
Notification to central, 80KB/s
Write to peripheral, 14KB/s

My conclusion is has some bugs in 1.0.6

next step is where is it.   Library source or IDF.

sandmen, 2021-06-14:
I think, something happen hier in
#include <esp_gatt_common_api.h>// ESP32 BLE

KorochanOne, 2021-06-15:
Hello sandmen
I checked "esp_gatt_common_api.h" in 1.0.6 and 1.0.4.
The only difference is 1.0.6 has additiona 4 lines.
#if (BLE_INCLUDED == TRUE)
extern uint16_t esp_ble_get_sendable_packets_num (void);
extern uint16_t esp_ble_get_cur_sendable_packets_num (uint16_t connid);
#endif

So I don't think this is the point. but I'm not sure.
Thanks

KorochanOne, 2021-06-15:
Hello
I found work around for 1.0.6
In BLEClient.cpp  line 267
BLEClient::gattClientEventHandler()
		case ESP_GATTC_CONNECT_EVT: {
			if (evtParam->connect.conn_id != getConnId()) break;      // 1.0.4 not exist 
			BLEDevice::updatePeerDevice(this, true, m_appId);
			esp_err_t errRc = esp_ble_gattc_send_mtu_req(gattc_if, evtParam->connect.conn_id);
			if (errRc != ESP_OK) {
				log_e("esp_ble_gattc_send_mtu_req: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			}
			BLEDevice::updatePeerDevice(this, true, m_appId);
			esp_err_t errRc = esp_ble_gattc_send_mtu_req(gattc_if, evtParam->connect.conn_id);
 evtParam->connect.conn_id!=getConnId() becomes true and never excute esp_ble_gattc_send_mtu_req

then it use 23 only
this is the reason of my problem.
I currently comment out and add around 10 millis delay to each notify.
Now it works.

I don't know why vtParam->connect.conn_id != getConnId()
I don't know why delay(10) is required each notify and writeValue

chegewara, 2021-06-15:
I dont know about code changes, didnt test it.
The delay is required to not get BLE congested. You can try lower delay thou. I some tests i performed long time ago i could make it work with delay 3-5 ms.

sandmen, 2021-06-15:
Maybe something with multiconnect and BLE.
They want to send the MTU-request only to the "correct" device !?
I don't know....
But thanks for the workaround, I had no time to look deeper into that topic.
You put a delay into read/write ?
I think, into your script, not into BLEClient-file ?

chegewara, 2021-06-15:
The delay is required only when notify is send.

KorochanOne, 2021-06-15:
Thanks all
I found a reason.
ESP_GATTC_OPEN_EVT
			m_conn_id = evtParam->open.conn_id;
ESP_GATTC_CONNECT_EVT
			if (evtParam->connect.conn_id != getConnId()) break;

after ESP_GATTC_CONNECT_EVT , ESP_GATTC_OPEN_EVT is occur.
so m_conn_id (getConnId()) is not yet set.

it is better 
ESP_GATTC_CONNECT_EVT
                        if(getConnId()==ESP_GATT_IF_NONE){
                         m_conn_id = evtParam->connect.conn_id;
                        }
			if (evtParam->connect.conn_id != getConnId()) break;

>I think, into your script, not into BLEClient-file ?
Yes.

for my case delay is required notify (peripheral to central) and write (central to peripheral)
and it is 8ms and 10ms in case of 512 and 256 times without error

closed 2021-06-26 by KorochanOne