espressif/arduino-esp32 #5238: ESP32蓝牙关闭在重启之后，就不能正常扫描，提示参数错误。

xiaodingding, 2021-06-03:
使用方式：
第一步：
        BLEDevice::deinit(true);
        delay(100);
        
        BLEDevice::init("cover");

第二步：创建一个蓝牙的Task
BaseType_t result = xTaskCreate(
        BleTask,   /* Function to implement the task */
        "BleTask", /* Name of the task */
        81920,     /* Stack size in words */
        NULL,      /* Task input parameter */
        3,         /* Priority of the task */
        &bleHandle);

第三步：
pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
pBLEScan->setActiveScan(true);
BLEScanResults foundDevices = pBLEScan->start(5, false);

开机上电的时候运行没有问题，但是当蓝牙断开之后，我就将这个蓝天的Task结束了，然后重新创建Task，然后在扫描蓝牙的时候，就会报错，异常重启。

一下是日志
[E][BLEScan.cpp:221] start(): esp_ble_gap_set_scan_params: err: 259, text: Unknown ESP_ERR error
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c2af  PS      : 0x00060330  A0      : 0x800d5920  A1      : 0x3fff8230  
A2      : 0x3fff824e  A3      : 0x00000000  A4      : 0x00000006  A5      : 0x3fff824e  
A6      : 0x3ffc4c34  A7      : 0x3ffba6cc  A8      : 0x800f0632  A9      : 0x3fff8200  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000010  A13     : 0x3fff82b0  
A14     : 0x3ffc4684  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4000c2af:0x3fff8230 0x400d591d:0x3fff8240 0x400d37e0:0x3fff8280 0x40090d96:0x3fff82f0

Rebooting...
ets Jun  8 2016 00:22:57

lbernstone, 2021-06-03:
This is an English forum.  https://esp32.com/viewforum.php?f=24&sid=9f63184878d30d15abd18ce4b2c79aaf

VojtechBartoska, 2021-06-04:
@xiaodingding Please if you want to report an issue translate it in English.

closed 2021-08-21 by stale[bot]